weewx/weewx #898: include my patch to support time_offset within diagrams

olwaldi, 2023-11-21:
To draw forecasts within diagrams from standard weewx skin Seasons I need to extend the graph beyond the current time into a configurable future. For that goal I did path imagegenerator.py and utilities.py modifying the function scaletime(). By that change the new option time_offset is introduced to shift the x-axis (time) by the given time in seconds to the right.

imagegenerator.py to forward an additional parameter in line 160 to become:
```        # Calculate a suitable min, max time for the requested time.
        minstamp, maxstamp, timeinc = weeplot.utilities.scaletime(
            plotgen_ts - int(plot_options.get('time_length', 86400)),
            plotgen_ts,
            int(plot_options.get(time_offset, 0))
        )
```
utilities.py to use the new parameter offset within definition and within the function body:
```
def scaletime(tmin_ts, tmax_ts, offset=0):
...
    tmin_dt = datetime.datetime.fromtimestamp(tmin_ts)
    tmax_dt = datetime.datetime.fromtimestamp(tmax_ts + offset)
...
        # up to the next one hour boundary:
        if tmax_dt > stop_dt:
            stop_dt += datetime.timedelta(hours=1)
        n_hours = int((tdelta + offset + 3599) / 3600)
        start_dt = stop_dt - datetime.timedelta(hours=n_hours)
...
        # The start time is one day earlier (respecting offset in 3h steps)
        h = (offset/3600) // 3
        start_dt = stop_dt - datetime.timedelta(hours=3*h, days=1)
```
I've tried to debug my change as good as possible, and I use it since many days without any problem.

See also our discussion at https://groups.google.com/g/weewx-user/c/76qdbSy3bbQ

tkeffer, 2023-11-21:
Sorry. I recall that discussion you linked to, but I'm not seeing the connection.

Could you please submit this as a pull request? It makes it easier to see exactly what you're proposing.

It should also include how `offset` is provided, documentation, and a unit test.

And, why 3h steps? What happens if you're doing a plot for a full year?

Thanks in advance.

olwaldi, 2023-11-22:
Within that discussion you can see a sample of the diagram weekwindvec using my proposed new feature time_offset - here the corresponding excerpt from my skin.conf:
```
    [[week_images]]
        x_label_format = %d
        bottom_label_format = %x %X
        time_length = 604800 # 7 days
        time_offset = 259200 # 3 days
        aggregate_type = avg
        aggregate_interval = hour
        ...
        [[[weekwindvec]]]
            [[[[windvec]]]]
                plot_type = vector
            [[[[forecast_windvec]]]]
                data_type = windvec
                data_binding = dwd_binding
                plot_type = vector
		label = ' '
		color = blue
```
Maybe my comment w.r.t. 3h steps is a bit misleading - that's my interpretation of your function scaletime() which for daily reports tries to select diagram borders at times being in steps of 3h to get a nice scale.

I did not investigate if my change might work for months or years. I want to include forecast data from German DWD within diagrams, and that is "only" 10days into the future. My purpose is to have a nice and quick view into the near future for planning local hiking.

Before going into patching weewx itself I tried to find any hint within your (very good) weewx customization guide to extend the time axis into the future - without success. Of course I'd prefer to go with any standard functionality instead of my fix.

You are right, I should provide a pull request. But I've never done this before, and risk might be high to create something in a wrong fashion. I'll have a look into the howtos ... but that might take some time since currently I'm busy with other tasks.

tkeffer, 2023-11-23:
OK, that's making sense. Let me know what you come up with.

BTW,

```
        time_length = 604800 # 7 days
        time_offset = 259200 # 3 days
``

can be abbreviated as

```
        time_length = 7d
        time_offset = 3d
```

olwaldi, 2023-12-12:
I've tried to create a pull request - https://github.com/weewx/weewx/pull/905

Hope. that it is visible to you. I did NOT test it within the beta version 5.0 of weewx, but I merged my changes into the files based on your master (probabely 5.0). I did test it of course within weewx version 4.10.2

The git commands at shell level were easy, but the web interface was difficult - especially getting a valid authentification.

closed 2024-01-06 by tkeffer