weewx/weewx #899: `wee_database --calc-missing` stuck always on same record

cincuranet, 2023-11-26:
I'm running ` wee_database --calc-missing` and the process always gets stuck on a particular record (without `--from` it's i.e. `Processing record: 463000; Last record: 2020-07-29 21:58:00 CEST (1596052680)`, but with different `--from`s I can find different ones across the database). I left it running for tens of minutes and it never finishes (and the CPU usage is 0). 

I tried running with `--dry-run` - that works fine. I tried dumping the database and creating fresh new one from the dump - no change.

When `Ctrl-C`ing the process, this is what I get (always the same call trace):

```
Traceback (most recent call last):rd: 2020-07-29 21:58:00 CEST (1596052680)
  File "/usr/share/weewx/weedb/sqlite.py", line 39, in guarded_fn
    return fn(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weedb/sqlite.py", line 233, in execute
    return sqlite3.Cursor.execute(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: database is locked

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/weewx/wee_database", line 1168, in <module>
    main()
  File "/usr/share/weewx/wee_database", line 235, in main
    calc_missing(config_dict, db_binding, options)
  File "/usr/share/weewx/wee_database", line 841, in calc_missing
    calc_missing_obj.run()
  File "/usr/share/weewx/weecfg/database.py", line 451, in run
    wxcalculate.do_calculations(record)
  File "/usr/share/weewx/weewx/wxservices.py", line 138, in do_calculations
    val = weewx.xtypes.get_scalar(obs_type, data_dict, self.db_manager)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/xtypes.py", line 77, in get_scalar
    return xtype.get_scalar(obs_type, record, db_manager, **option_dict)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/wxxtypes.py", line 301, in get_scalar
    r = db_manager.getSql("SELECT MAX(outTemp), MIN(outTemp), "
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/manager.py", line 579, in getSql
    _cursor.execute(sql, sqlargs)
  File "/usr/share/weewx/weedb/sqlite.py", line 43, in guarded_fn
    msg = str(e).lower()
          ^^^^^^
KeyboardInterrupt
```

Any help how to pinpoint what's wrong?

tkeffer, 2023-11-26:
It would help if we could see the log, but off the top of my head: are you trying to run `wee_database` at the same time as `weewxd`?

cincuranet, 2023-11-26:
I'm not running `weewxd` (whole `weewx` service is stopped).

Here's the log (all the way up to the point of stucking, nothing more after that point):
```
Nov 26 17:12:12 x2pi-weather python3[13815]: wee_database[13815] INFO __main__: Preparing to calculate missing derived observations...
Nov 26 17:12:12 x2pi-weather python3[13815]: wee_database[13815] INFO __main__: Missing derived observations will be calculated for all records.
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.manager: Daily summary version is 4.0
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO __main__: Calculating missing derived observations...
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.engine.StdTimeSynch
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.engine.StdTimeSynch
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.engine.StdConvert
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.engine: StdConvert target unit is 0x1
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.engine.StdConvert
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.engine.StdCalibrate
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.engine.StdCalibrate
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.engine.StdQC
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.engine.StdQC
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.wxservices.StdWXCalculate
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.wxservices: StdWXCalculate will use data binding wx_binding
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.manager: Daily summary version is 4.0
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.wxservices.StdWXCalculate
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdWXXTypes
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdWXXTypes
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdPressureCooker
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdPressureCooker
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdRainRater
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdRainRater
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.wxxtypes.StdDelta
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.wxxtypes.StdDelta
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.engine.StdArchive
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.engine: Archive will use data binding wx_binding
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.engine: Record generation will be attempted in 'hardware'
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.engine: Using archive interval of 300 seconds (specified by hardware)
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Use LOOP data in hi/low calculations: 1
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.engine.StdArchive
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.restx.StdStationRegistry
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.restx: StationRegistry: Station will be registered.
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.restx.StdStationRegistry
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.restx.StdWunderground
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.restx: Wunderground: Posting not enabled.
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.restx.StdWunderground
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Loading service weewx.restx.StdPWSweather
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.restx: PWSWeather: Data for station ZELESICE will be posted
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.restx.StdPWSweather
[... cut ...]
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.engine: Finished loading service weewx.engine.StdReport
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] INFO weewx.wxservices: StdWXCalculate will use data binding wx_binding
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.manager: Daily summary version is 4.0
Nov 26 17:12:14 x2pi-weather python3[13815]: wee_database[13815] DEBUG weewx.manager: Daily summary version is 4.0
```

cincuranet, 2023-11-27:
This is how `lsof` looks when the app is running fine:
```
COMMAND  PID USER   FD   TYPE DEVICE  SIZE/OFF    NODE NAME
python3 6595 root    4ur  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    5ur  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    6ur  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    7ur  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    8ur  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    9ur  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
```

And this is after it got stuck:
```
COMMAND  PID USER   FD   TYPE DEVICE  SIZE/OFF    NODE NAME
python3 6595 root    4uw  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    5uw  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    6uw  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    7uw  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    8uw  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
python3 6595 root    9uw  REG    8,1 398299136 1046813 /var/lib/weewx/weewx.sdb
```

tkeffer, 2023-11-27:
No clues there.

Try removing the MQTT and windy extensions, then trying again. Something is locking the database, then not letting go.

cincuranet, 2023-11-28:
I installed fresh _weewx_ on a fresh install of Debian 12, took copy of config file (minus MQTT and Windy) and copy of database and the behavior is exactly the same. I can share the database with you, if you want.

In the meantime, this is the "offending" record. Maybe you can spot something that throws the application off.
```sql
INSERT INTO [archive] ([dateTime],[usUnits],[interval],[altimeter],[appTemp],[appTemp1],[barometer],[batteryStatus1],[batteryStatus2],[batteryStatus3],[batteryStatus4],[batteryStatus5],[batteryStatus6],[batteryStatus7],[batteryStatus8],[cloudbase],[co],[co2],[consBatteryVoltage],[dewpoint],[dewpoint1],[ET],[extraHumid1],[extraHumid2],[extraHumid3],[extraHumid4],[extraHumid5],[extraHumid6],[extraHumid7],[extraHumid8],[extraTemp1],[extraTemp2],[extraTemp3],[extraTemp4],[extraTemp5],[extraTemp6],[extraTemp7],[extraTemp8],[forecast],[hail],[hailBatteryStatus],[hailRate],[heatindex],[heatindex1],[heatingTemp],[heatingVoltage],[humidex],[humidex1],[inDewpoint],[inHumidity],[inTemp],[inTempBatteryStatus],[leafTemp1],[leafTemp2],[leafWet1],[leafWet2],[lightning_distance],[lightning_disturber_count],[lightning_energy],[lightning_noise_count],[lightning_strike_count],[luminosity],[maxSolarRad],[nh3],[no2],[noise],[o3],[outHumidity],[outTemp],[outTempBatteryStatus],[pb],[pm10_0],[pm1_0],[pm2_5],[pressure],[radiation],[rain],[rainBatteryStatus],[rainRate],[referenceVoltage],[rxCheckPercent],[signal1],[signal2],[signal3],[signal4],[signal5],[signal6],[signal7],[signal8],[snow],[snowBatteryStatus],[snowDepth],[snowMoisture],[snowRate],[so2],[soilMoist1],[soilMoist2],[soilMoist3],[soilMoist4],[soilTemp1],[soilTemp2],[soilTemp3],[soilTemp4],[supplyVoltage],[txBatteryStatus],[UV],[uvBatteryStatus],[windBatteryStatus],[windchill],[windDir],[windGust],[windGustDir],[windrun],[windSpeed]) VALUES (1596052680,1,2,30.1403754505141,NULL,NULL,30.1363069170139,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,48.556490358971,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,66.2,NULL,NULL,NULL,NULL,NULL,NULL,65,77.9,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,53,66.2,0,NULL,NULL,NULL,NULL,29.2917789341667,NULL,0,NULL,0,NULL,100,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,NULL,66.2,NULL,0.671082555581791,NULL,NULL,0);
```

tkeffer, 2023-11-28:
Just curious: how did you determine that this statement is the problem?

It uses a non-standard syntax. I'll check with Gary on that.

What version of sqlite are you using?

    python3 -c "import sqlite3;print(sqlite3.version)"

cincuranet, 2023-11-28:
I did `select * from archive where dateTime = 1596052680` where `1596052680` is from the console log. I then copied the row as insert in _Database .NET_.

The version is `2.6.0`.

cincuranet, 2023-11-28:
OK. It's probably not this particular record. I deleted it and now the process stops on `Processing record: 463000; Last record: 2020-07-29 22:00:00 CEST (1596052800)` and the `1596052800` is right after the one mentioned previously. And same story repeats if I delete `1596052800`. 

Could it be that some flushing is locking the database and not unlocking?

tkeffer, 2023-11-28:
V2.6.0 is current.

A few things come to mind:
1. The "last record" statement is only a rough guide. It's only printed every 1,000 records, so the offending record could be any one of a thousand.
2. It looks like you're working on a largish database, at least 463,000 records. How big is it? The calculation should use 20 day "tranches" to limit the amount of memory required, but it's possible that's too much. How large a machine are you running it on?
3. Finally, I'm still not understanding where the `INSERT` statement came from. The `--calc-missing` calculation does not use `INSERT`, only `UPDATE`. You said, "I then copied the row as insert in Database .NET," but I don't understand what that means.

Not quite ready to ask for a copy of your database, but it's getting close!

cincuranet, 2023-11-28:
Coincidentally. I've been poking into the code left and right (`manager.py`, `sqlite.py`, ...) trying to cause something that would give me some extra clues and I ended up changing this line https://github.com/weewx/weewx/blob/5c5d9ff2eedd70204e563c2432312c427f43d8b3/bin/weecfg/database.py#L390 in `database.py` hardcoding some values.

And depending on `trans_days` size I get these results:
| trans_days | result |
|------------|--------|
| 1000       | ❌      |
| 20         | ❌      |
| 10         | ❌      |
| 5          | ❌      |
| 4          | ✔️      |
| 3          | ✔️      |
| 2          | ✔️      |
| 1          | ✔️      |

Not sure what the conclusion is. But at least it's something.

tkeffer, 2023-11-28:
That's exactly what I was going to suggest! Well done, and good sleuthing!

So, if you limit the transaction size to 4 days or less, it works. At 2GB, your machine is not a huge one, but that's still a reasonable amount of memory. I'm surprised 50 is too big.

What we need to do is make the number configurable. I'll changed the title of this issue to reflect.

cincuranet, 2023-11-29:
Because I'm testing on a VM I can easily assign more memory. But even with 16GB it ends up with the same `database is locked` error (for `trans_days` 5).

staze, 2024-01-07:
Hey @tkeffer could this be the issue I was seeing where I had to set the tranche?

tkeffer, 2024-01-07:
If you're referring to recalculating ET, yes, it is the same issue. But, you also had some issues with radiation as I recall.

staze, 2024-01-08:
> If you're referring to recalculating ET, yes, it is the same issue. But, you also had some issues with radiation as I recall.

The radiation issue was from old weather software populating garbage into that field... which caused ET to calculate as garbage. But yeah, I mean the calc-missing command hanging because of some sqlite interaction or... something.

tkeffer, 2024-01-10:
Closing, because the user can set the tranche size with option [`--tranche`](https://weewx.com/docs/5.0/utilities/weectl-database/#-tranche) in V5.