weewx/weewx #904: rpm --import https://weewx.com/keys.html fails on RHEL9(ish)

vinceskahan, 2023-12-08:
(we talked email about this in August, but really didn't get to an implemented solution.  Suggest this needs being worked before v5 drops)a current issue with details and suggested path forward).

The RHish docs specify the command above to import the signing keys for the rpms.  The problem here is that the keys.html file contains Tom's legacy SHA1 key, so the suggested command fails on RHEL9(ish) systems as currently written.   I see a number of ways around this that I'll get to.  Stay tuned.

Background - we discussed this via email many months ago and Tom created a new key and added it to keys.html, but that handles only 'newly' signed rpms, not the legacy ones in the repos and download link.

Workaround - users of RHEL(ish) 9 or above can 'temporarily' permit import of SHA1 keys by:

* update-crypto-policies --set DEFAULT:SHA1
* (run the import commands)
* update-crypto-policies --set DEFAULT

Alternate workaround:

* re-sign all the rpms in the repo (rpm --addsign foo.rpm) and rebuild the repo
* then remove Tom's legacy key from keys.html
* and the docs as written should work

vinceskahan, 2023-12-09:
Found a nice article in RH land....
https://access.redhat.com/articles/3359321

This deciphers the installed keys after rpm --import has been run....
rpm -q gpg-pubkey --qf '%{name}-%{version}-%{release} --> %{summary}\n'

````
[root@322d1e7705f7 /]# rpm -q gpg-pubkey --qf '%{name}-%{version}-%{release} --> %{summary}\n'
warning: Signature not supported. Hash algorithm SHA1 not available.
gpg-pubkey-350d275d-6279464b --> Rocky Enterprise Software Foundation - Release key 2022 <releng@rockylinux.org> public key
gpg-pubkey-3228467c-613798eb --> Fedora (epel9) <epel@fedoraproject.org> public key
gpg-pubkey-17fc079e-64e676bb --> Tom Keffer (Author of WeeWX) <tkeffer@gmail.com> public key
gpg-pubkey-f0e2b09e-52e7e1ce --> Matthew Wall (weewx) <mwall@users.sourceforge.net> public key
gpg-pubkey-75bf977c-5466a5e2 --> Thomas Keffer (Author of weewx) <tkeffer@gmail.com> public key
````

Unfortunately it doesn't decipher which key was used to sign the rpm.  I can't find a way to do so without importing the same keys into gpg as indicated above.  Just passing along some rpm trivia I guess...

tkeffer, 2023-12-09:
Thanks for doing the research on this. It sounds like the easiest would be to just resign all the rpm packages `rpm --addsign foo.rpm` using my new key.

I can't run a VM on my Mac M1, but I can run Docker. Do you think you could set me up with something that would go through the necessary steps? 

When we do a new version, we download the old repository, make changes, then upload the now changed repo, then swap the old for the new. So, a docker file that does the first two steps (download, resign). Then we'll figure out the last 2 steps.

Matthew knows way more about this because he set the system up...

vinceskahan, 2023-12-09:
I would agree resigning is what I'd recommend.  Folks are going to (maybe)
have to know to download the newest keys.html file but that's the case now
anyway.  Anybody with the original one from a year ago won't be able to
validate a 'new' rpm without doing so if you're going to use your added
summer-2023 key.

A dockerfile to automate resigning is sure no problem.  That's how I did my
testing.  Stay tuned.

I would lean more toward splitting the steps up.  Have the docker mount a
directory with the previously downloaded copy of the repo, as well as your
.gnupg directory (I'll have to think of that re: how to be secure enough
there).   I'll fiddle a bit and see what I can come up with.

If you see somebody downloading all your repo tomorrow with an address
starting with 24.16.x.x , it's me :-)

--
-----  ***@***.*** ----

vinceskahan, 2023-12-10:
docker setup to do the downloading works and is very snappy since your repos are tiny.  I used reposync which I've used previously in $work.   Having issues with resigning the rpms for some reason in docker so that might take a bit.

dumb question - what computer do you currently use to build the rpms on ?   The current makefile adds a signature to the rpm in the make process down near line 317.  You could likely do this one-time re-signing of the existing rpms in about 10 lines of bash if you ran it on the rpm build host.  Docker might be overkill here perhaps.

vinceskahan, 2023-12-10:
update - you literally can type "rpm --resign *.rpm" while in a directory of rpms and it will resign them all in one bash command.   You can verify it by running "rpm -qi *.rpm | grep Sig" before and after.   All you need to do is make sure your ~/.rpmmacros file points to the key you want to (re)sign the rpms with.

vinceskahan, 2023-12-10:
ok tom - I think https://github.com/vinceskahan/resign-weewx-rpms does what you want.  See if the README and two bash scripts are documented enough.   Seems to work ok here for me on the mac.

tkeffer, 2023-12-11:
Got docker working. Downloaded repositories. But, when I either run `resign-it.sh`, or use a bash command `rpm --resign *.rpm`, I'm not changing the signature. The key ID is still of the old signature.

Somehow, it feels like I'm not connecting the `--resign` command to the proper signature. I didn't understand your comment about `~/.rpmmacros`.

vinceskahan, 2023-12-12:
(coordinated via email - I had some typos and was missing adding a rpm to the image.)
updated repo to fix errors/omissions and added suse15 repo as well so I 'think' the docker stuff is stable now.

tkeffer, 2023-12-12:
Got everything resigned. Thanks to @matthewwall for rebuilding the repos on weewx.com.

I'm not going to do SUSE because I've run out of energy, and they're just not used that often. 

Closing.

vinceskahan, 2023-12-12:
You need to try that again.   I (re)downloaded and signatures are all over
the map and keys.html does not import on rocky 9.2 so the old key is still
there.

My guess is somebody forgot to put the new one into place 'or' forgot to
run createrepo to rebuild the metadata

tkeffer, 2023-12-13:
You may have caught things half-way to completion. Is it still like that?

vinceskahan, 2023-12-13:
Yes - it looks like it was never touched.....

***@***.*** ~]$ docker run --rm -it rockylinux:9.2 bash

>

***@***.*** /]# yum install -y -q yum-utils
> Importing GPG key 0x350D275D:
>  Userid     : "Rocky Enterprise Software Foundation - Release key 2022 <
> ***@***.***>"
>  Fingerprint: 21CB 256A E16F C54C 6E65 2949 702D 426D 350D 275D
>  From       : /etc/pki/rpm-gpg/RPM-GPG-KEY-Rocky-9
> Installed:
>   dbus-libs-1:1.12.20-8.el9.aarch64
>  dnf-plugins-core-4.3.0-11.el9_3.noarch
> python3-dateutil-1:2.8.1-7.el9.noarch
> python3-dbus-1.2.18-2.el9.aarch64
>   python3-dnf-plugins-core-4.3.0-11.el9_3.noarch
> python3-six-1.15.0-9.el9.noarch
>  python3-systemd-234-18.el9.aarch64          yum-utils-4.3.0-11.el9_3.noarch
>

> ***@***.*** /]# rpm --import https://weewx.com/keys.html
> warning: Signature not supported. Hash algorithm SHA1 not available.
> error: https://weewx.com/keys.html: key 1 import failed.
>

> ***@***.*** /]# curl -s https://weewx.com/yum/weewx-el8.repo | tee
> /etc/yum.repos.d/weewx.repo
> [weewx]
> name=weewx
> baseurl=http://weewx.com/yum/weewx/el8
> enabled=1
> gpgcheck=0
> ***@***.*** /]# yumdownloader weewx
> weewx
>
>    182 kB/s |  23 kB     00:00
> weewx-4.10.2-1.el8.noarch.rpm
>
>    7.4 MB/s | 1.5 MB     00:00
>

> ***@***.*** /]# rpm -qi weewx-4.10.2-1.el8.noarch.rpm
> warning: weewx-4.10.2-1.el8.noarch.rpm: Header V4 RSA/SHA256 Signature,
> key ID 75bf977c: NOKEY
> Name        : weewx
> Version     : 4.10.2
> Release     : 1.el8
> Architecture: noarch
> Install Date: (not installed)
> Group       : Applications/Science
> Size        : 5825539
> License     : GPLv3
> Signature   : RSA/SHA256, Thu Feb 23 01:05:53 2023, Key ID a0cb255b75bf977c
> Source RPM  : weewx-4.10.2-1.el8.src.rpm
> Build Date  : Thu Feb 23 01:05:52 2023
> Build Host  : localhost
> URL         : http://www.weewx.com
> Summary     : weather software
> Description :
> weewx interacts with a weather station to produce graphs, reports, and HTML
> pages. weewx can upload data to weather services such as
> WeatherUnderground,
> PWSweather.com, or CWOP.

Keys still has all three keys including your old invalid SHA1 one.
Downloaded weewx and looked at the Signature line which has a variety of
old dates on the rpms.

The keys should import successfully.  The dates on the Signature lines when
you 'rpm -qi foo.rpm | grep Signature' should look the same and from today.

tkeffer, 2023-12-13:
Try the yum-test repo. Matthew is still testing things a bit.

```
wget  https://weewx.com/yum-test/weewx/el8/RPMS/weewx-4.10.2-1.el8.noarch.rpm
rpm -qi weewx-4.10.2-1.el8.noarch.rpm 
```
yields
```
warning: weewx-4.10.2-1.el8.noarch.rpm: Header V4 RSA/SHA256 Signature, key ID 17fc079e: NOKEY
Name        : weewx
Version     : 4.10.2
Release     : 1.el8
Architecture: noarch
Install Date: (not installed)
Group       : Applications/Science
Size        : 5825539
License     : GPLv3
Signature   : RSA/SHA256, Tue Dec 12 21:58:46 2023, Key ID b7d370ec17fc079e
Source RPM  : weewx-4.10.2-1.el8.src.rpm
Build Date  : Thu Feb 23 01:05:52 2023
Build Host  : localhost
Relocations : (not relocatable)
URL         : http://www.weewx.com
Summary     : weather software
Description :
weewx interacts with a weather station to produce graphs, reports, and HTML
pages. weewx can upload data to weather services such as WeatherUnderground,
PWSweather.com, or CWOP.
```

vinceskahan, 2023-12-13:
The test repo is a bit disheveled currently.  I cooked up a bash script to query my mirror from there and see which sigs worked and which don't, as well as a tweaked dockerfile to build over rocky 9.2 so it's picky re: key algorithm for your signing keys.  The following fail.

````
root@ac9ec112ae19 tmp]# bash h
warning: /root/repos/el8test/RPMS/weewx-4.0.0-1.el8.noarch.rpm: Header V4 RSA/SHA1 Signature, key ID 15306a67: NOKEY
Signature   : RSA/SHA1, Thu Apr 30 15:43:44 2020, Key ID 50abae5f15306a67

warning: /root/repos/el8test/RPMS/weewx-4.1.0-1.el8.noarch.rpm: Header V4 RSA/SHA1 Signature, key ID 15306a67: NOKEY
Signature   : RSA/SHA1, Mon May 25 19:22:19 2020, Key ID 50abae5f15306a67

warning: /root/repos/el8test/RPMS/weewx-4.2.0-1.el8.noarch.rpm: Header V4 RSA/SHA256 Signature, key ID 15306a67: NOKEY
Signature   : RSA/SHA256, Tue Oct 27 01:18:02 2020, Key ID 50abae5f15306a67

warning: /root/repos/el8test/RPMS/weewx-4.6.0-1.el8.noarch.rpm: Header V4 RSA/SHA256 Signature, key ID 15306a67: NOKEY
Signature   : RSA/SHA256, Fri Feb  4 14:27:01 2022, Key ID 50abae5f15306a67

warning: /root/repos/el8test/RPMS/weewx-4.6.1-1.el8.noarch.rpm: Header V4 RSA/SHA256 Signature, key ID 15306a67: NOKEY
Signature   : RSA/SHA256, Fri Feb 11 00:41:02 2022, Key ID 50abae5f15306a67

warning: /root/repos/el8test/RPMS/weewx-4.6.2-1.el8.noarch.rpm: Header V4 RSA/SHA256 Signature, key ID 15306a67: NOKEY
Signature   : RSA/SHA256, Fri Feb 11 03:40:19 2022, Key ID 50abae5f15306a67

/root/.gnupg/pubring.kbx
------------------------
pub   rsa2048 2014-11-15 [SC]
      EAC207118C0BD050B8359313A0CB255B75BF977C
uid           [ unknown] Thomas Keffer (Author of weewx) <tkeffer@gmail.com>
sub   rsa2048 2014-11-15 [E]

pub   rsa3072 2023-08-23 [SC]
      3EF8C49F6B923DDCF4B993D0B7D370EC17FC079E
uid           [ unknown] Tom Keffer (Author of WeeWX) <tkeffer@gmail.com>
sub   rsa3072 2023-08-23 [E]

pub   dsa1024 2014-01-28 [SC]
      1E9FF20B86EF78AB3B454944ED444FCCF0E2B09E
uid           [ unknown] Matthew Wall (weewx) <mwall@users.sourceforge.net>
sub   elg1024 2014-01-28 [E]

````

Script to audit is kinda brute force

````

# el8 repo mirror
# REPODIR='/root/repos/el8/RPMS'

# el8test repo mirror
REPODIR='/root/repos/el8test/RPMS'

for f in `ls ${REPODIR}/*.rpm`
do
	for fn in `rpm -K $f | grep NOT | awk -F ":" '{print $1}'`
	do
		rpm -qi $fn | grep Signature
		echo ""
	done
done
````

tkeffer, 2023-12-13:
I only resigned the rpm files that used my old key. I did not resign the ones Matthew had signed.

Is that what you're seeing?

vinceskahan, 2023-12-13:
I would have just re-signed them all rather than thinking too much, but success would look like:

- all keys in keys.html are valid for RH9 and above
- all rpms are signed with one of those two keys

So the handful that I put in my earlier reply this morning are signed with need to be re-signed at a minimum....and the old key deleted from keys.html as well of course.

Basically your test is to run "rpm -K" against all the rpms.  They should all report "digests signatures OK".

Currently....
````
[root@b284e9b45d54 RPMS]# rpm -K *
weewx-4.0.0-1.el8.noarch.rpm: digests SIGNATURES NOT OK
weewx-4.1.0-1.el8.noarch.rpm: digests SIGNATURES NOT OK
weewx-4.1.1-1.el8.noarch.rpm: digests signatures OK
weewx-4.10.0-1.el8.noarch.rpm: digests signatures OK
weewx-4.10.1-1.el8.noarch.rpm: digests signatures OK
weewx-4.10.2-1.el8.noarch.rpm: digests signatures OK
weewx-4.2.0-1.el8.noarch.rpm: digests SIGNATURES NOT OK
weewx-4.3.0-1.el8.noarch.rpm: digests signatures OK
weewx-4.4.0-1.el8.noarch.rpm: digests signatures OK
weewx-4.5.0-1.el8.noarch.rpm: digests signatures OK
weewx-4.5.1-1.el8.noarch.rpm: digests signatures OK
weewx-4.6.0-1.el8.noarch.rpm: digests SIGNATURES NOT OK
weewx-4.6.1-1.el8.noarch.rpm: digests SIGNATURES NOT OK
weewx-4.6.2-1.el8.noarch.rpm: digests SIGNATURES NOT OK
weewx-4.7.0-1.el8.noarch.rpm: digests signatures OK
weewx-4.8.0-1.el8.noarch.rpm: digests signatures OK
weewx-4.8.0-2.el8.noarch.rpm: digests signatures OK
weewx-4.9.0-1.el8.noarch.rpm: digests signatures OK
weewx-4.9.1-1.el8.noarch.rpm: digests signatures OK
````

The old ones showing NOT OK were signed with a gpg key that has been NOT OK for years.  I found an ancient thread mentioning it but can't find whose ancient key it corresponded to.  Probably doesn't matter.  Just resign those offending rpms too.

tkeffer, 2023-12-13:
OK, I resigned _everything_ this time and have asked Matthew to move them into place.

vinceskahan, 2023-12-13:
As I mentioned in email, that spellcheck 'infernal centos' was pretty accurate.  RPM can make you a crazy person.

matthewwall, 2023-12-29:
i just re-synced the test-repo with the rpms from tom.  now all of the el7 and el8 rpms are signed by tom, and the latest v5 and all el9 rpms are signed by me.  verified using 'rpm -K' on rocky8.  when we release yum-test into prod then the non-test repo will get the changes.

i even maintained the file dates :)

closed 2023-12-29 by matthewwall