weewx/weewx #910: Error when crt is installed togheter weewx-weatherlink-live in WeeWX

GW400, 2024-01-03:
Error when crt is installed togheter weewx-weatherlink-live in WeeWX 5.x
Currently tested with: https://weewx.com/downloads/development_versions/python3-weewx_5.0.0rc2-1_all.deb
+ https://github.com/michael-slx/weewx-weatherlink-live

Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]: CRITICAL __main__:     ****  ValueError: invalid version number '5.0.0rc2'
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]: CRITICAL __main__:     ****  Exiting.
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]: Traceback (most recent call last):
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/share/weewx/weewxd.py", line 265, in <module>
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     main()
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/share/weewx/weewxd.py", line 160, in main
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     engine = weewx.engine.StdEngine(config_dict)
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/share/weewx/weewx/engine.py", line 89, in __init__
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     self.loadServices(config_dict)
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/share/weewx/weewx/engine.py", line 157, in loadServices
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     obj = weeutil.weeutil.get_object(svc)(self, config_dict)
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/share/weewx/weeutil/weeutil.py", line 1404, in get_object
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     module = importlib.import_module(module_name)
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     return _bootstrap._gcd_import(name[level:], package, level)
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/etc/weewx/bin/user/crt.py", line 97, in <module>
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     if StrictVersion(weewx.__version__) < StrictVersion(REQUIRED_WEEWX):
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/lib/python3.10/distutils/version.py", line 40, in __init__
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     self.parse(vstring)
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:   File "/usr/lib/python3.10/distutils/version.py", line 137, in parse
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]:     raise ValueError("invalid version number '%s'" % vstring)
Jan  3 21:01:51 fl-ubuntu-01 weewxd[52259]: ValueError: invalid version number '5.0.0rc2'
Jan  3 21:01:51 fl-ubuntu-01 systemd[1]: weewx.service: Main process exited, code=exited, status=1/FAILURE
Jan  3 21:01:51 fl-ubuntu-01 systemd[1]: weewx.service: Failed with result 'exit-code'.

tkeffer, 2024-01-03:
This is because `crt` uses `StrictVersion` from `distutils`, which does not know how to handle "release candidate" (`rc`) version numbers. This is a problem in `crt`, not WeeWX.

The error will go away once V5.0 is released. In the meantime, you can comment out the check (lines 97-99) in `crt.py`.

closed 2024-01-03 by tkeffer

G400, 2024-01-08:
Thank you tkeffer! 
Commenting out the lines works.

I can see that there are missing values in exported Cumulus realtime.txt file.

Any suggestion to how I can fix this, if possible?

crt v0.21 + weatherlink-live v1.1.2:
08/01/24 10:48:41 NULL NULL NULL 0.0 0.0 NULL 0.00 0.00 NULL --- 0 mph F in in 0.0 NULL 0.00 0.00 NULL NULL NULL NULL NULL 20.3 10:45 8.7 00:05 0.0 09:30 0.0 09:30 30.83 09:55 30.79 06:25 5.0.0rc3 0 0.0 NULL NULL NULL 0.00 NULL NULL 0.00 0 1 0 --- NULL ft NULL 2.5 50.8 0

WeatherLink Live data:
{
	"data": {
		"did": "001D0A714331",
		"ts": 1704710999,
		"conditions": [
			{
				"lsid": 400819,
				"data_structure_type": 1,
				"txid": 1,
				"temp": 20.9,
				"hum": 48.0,
				"dew_point": 4.3,
				"wet_bulb": 15.3,
				"heat_index": 19.9,
				"wind_chill": 20.9,
				"thw_index": 19.9,
				"thsw_index": 17.9,
				"wind_speed_last": 0.0,
				"wind_dir_last": 0,
				"wind_speed_avg_last_1_min": 0.0,
				"wind_dir_scalar_avg_last_1_min": 0,
				"wind_speed_avg_last_2_min": 0.0,
				"wind_dir_scalar_avg_last_2_min": 0,
				"wind_speed_hi_last_2_min": 0.0,
				"wind_dir_at_hi_speed_last_2_min": 0,
				"wind_speed_avg_last_10_min": 0.0,
				"wind_dir_scalar_avg_last_10_min": 0,
				"wind_speed_hi_last_10_min": 0.0,
				"wind_dir_at_hi_speed_last_10_min": 0,
				"rain_size": 2,
				"rain_rate_last": 0,
				"rain_rate_hi": 0,
				"rainfall_last_15_min": 0,
				"rain_rate_hi_last_15_min": 0,
				"rainfall_last_60_min": 0,
				"rainfall_last_24_hr": 0,
				"rain_storm": 0,
				"rain_storm_start_at": null,
				"solar_rad": 67,
				"uv_index": 0.6,
				"rx_state": 0,
				"trans_battery_flag": 0,
				"rainfall_daily": 0,
				"rainfall_monthly": 0,
				"rainfall_year": 0,
				"rain_storm_last": 252,
				"rain_storm_last_start_at": 1702624080,
				"rain_storm_last_end_at": 1703034060
			},
			{
				"lsid": 400818,
				"data_structure_type": 4,
				"temp_in": 75.8,
				"hum_in": 26.0,
				"dew_point_in": 38.5,
				"heat_index_in": 73.8
			},
			{
				"lsid": 400817,
				"data_structure_type": 3,
				"bar_sea_level": 30.608,
				"bar_trend": 0.018,
				"bar_absolute": 27.941
			}
		]
	},
	"error": null
}

crt v0.21 + WLLDriver v0.4
08/01/24 11:46:31 24.4 83 20.0 0.0 0.0 NULL 0.00 0.00 30.71 --- 0 mph F in in 6.9 0.00 0.00 0.00 0.00 64.0 34 24.4 5.8 24.1 11:45 13.8 05:00 2.7 01:35 4.0 01:30 30.72 09:30 30.66 00:25 4.10.2 0 0.0 24.4 24.4 NULL NULL NULL NULL 0.00 0 1 0 --- 1173 ft 19.4 2.4 NULL 0

WeatherLink Live data:
{
	"data": {
		"did": "001D0A714475",
		"ts": 1704710672,
		"conditions": [
			{
				"lsid": 620021,
				"data_structure_type": 1,
				"txid": 1,
				"temp": 24.3,
				"hum": 83.1,
				"dew_point": 19.9,
				"wet_bulb": 22.8,
				"heat_index": 24.3,
				"wind_chill": 24.3,
				"thw_index": 24.3,
				"thsw_index": null,
				"wind_speed_last": 0.0,
				"wind_dir_last": 0,
				"wind_speed_avg_last_1_min": 0.0,
				"wind_dir_scalar_avg_last_1_min": 0,
				"wind_speed_avg_last_2_min": 0.0,
				"wind_dir_scalar_avg_last_2_min": 0,
				"wind_speed_hi_last_2_min": 0.0,
				"wind_dir_at_hi_speed_last_2_min": 0,
				"wind_speed_avg_last_10_min": 0.0,
				"wind_dir_scalar_avg_last_10_min": null,
				"wind_speed_hi_last_10_min": 0.0,
				"wind_dir_at_hi_speed_last_10_min": 0,
				"rain_size": 2,
				"rain_rate_last": 0,
				"rain_rate_hi": 0,
				"rainfall_last_15_min": 0,
				"rain_rate_hi_last_15_min": 0,
				"rainfall_last_60_min": 0,
				"rainfall_last_24_hr": 0,
				"rain_storm": 0,
				"rain_storm_start_at": null,
				"solar_rad": null,
				"uv_index": null,
				"rx_state": 0,
				"trans_battery_flag": 0,
				"rainfall_daily": 0,
				"rainfall_monthly": 0,
				"rainfall_year": 0,
				"rain_storm_last": 94,
				"rain_storm_last_start_at": 1703715061,
				"rain_storm_last_end_at": 1704002461
			},
			{
				"lsid": 620020,
				"data_structure_type": 4,
				"temp_in": 64.1,
				"hum_in": 34.6,
				"dew_point_in": 35.6,
				"heat_index_in": 61.1
			},
			{
				"lsid": 620019,
				"data_structure_type": 3,
				"bar_sea_level": 30.712,
				"bar_trend": 0.013,
				"bar_absolute": 30.545
			}
		]
	},
	"error": null
}