weewx/weewx #911: Download package not found

RebeckaJohansson, 2024-01-04:
I tryed to download 

Platform Packages
For Debian, Ubuntu, Mint, including Raspberry Pi, see:
http://weewx.com/docs/debian.htm

but when i do i got this problem

404 Not Found
nginx/1.22.0

Can any tell whats wrong becaus i always used that installation and dont know i to do right now to build my system up again.

gjr80, 2024-01-04:
Rather than just the current release version, on-line docs are now provided for each WeeWX release. Consequently, any old document links will likely not work. Try going to http://weewx.com/docs.html and select the WeeWX version you are using. For WeeWX v4.x you should find the docs in the familiar format. WeeWX v5 documents have been re-formatted with a new look and feel and a slightly different structure.

See how you go.

RebeckaJohansson, 2024-01-04:
> Rather than just the current release version, on-line docs are now provided for each WeeWX release. Consequently, any old document links will likely not work. Try going to http://weewx.com/docs.html [...]

Thank you so much for fast answere :)

closed 2024-01-04 by gjr80