weewx/weewx #915: values for derived variables when there should be none

matthewwall, 2024-01-09:
in a fresh install of v5, LOOP packets contain values for some observations where there should be `None` or simply no observation.  this is illustrated when using hardware that emits only pressure and temperature.

values of `0.0` are present for `maxSolarRad` and `rainRate`.

values of `None` appear for `ET` and `inDewpoint`

why the inconsistency?  i would consider the values of 0 to be a bug, because it makes plots and values appear in the Seasons skin, even though the station has no radiation or rain instruments.

there is a workaround: remove the observations from `[StdWXCalculate][[Calculations]]` - then they do not show up at all.  but users should not have to do this.

matthewwall, 2024-01-09:
it happens with interceptor, gw1000, sdr drivers (at sites i have been running for years) and the 'digiwx' driver i just wrote (for capturing data from an old digiwx METAR system at the vinalhaven 'airport').  

it happens for any driver that does not provide rain or radiation.   

this is a long-standing issue, not new to v5.  

it happens with the out-of-the-box set of calculations:
```
[StdWXCalculate]
    [[Calculations]]
        pressure = prefer_hardware
        altimeter = prefer_hardware
        appTemp = prefer_hardware
        barometer = prefer_hardware
        cloudbase = prefer_hardware
        dewpoint = prefer_hardware
        ET = prefer_hardware
        heatindex = prefer_hardware
        humidex = prefer_hardware
        inDewpoint = prefer_hardware
        maxSolarRad = prefer_hardware
        rainRate = prefer_hardware
        windchill = prefer_hardware
        windrun = prefer_hardware
```

my workaround has been to remove the offending items 'rainRate', 'inDewpoint', and 'ET'
```
[StdWXCalculate]
    [[Calculations]]
        pressure = prefer_hardware
        altimeter = prefer_hardware
        appTemp = prefer_hardware
        barometer = prefer_hardware
        cloudbase = prefer_hardware
        dewpoint = prefer_hardware
        heatindex = prefer_hardware
        humidex = prefer_hardware
        maxSolarRad = prefer_hardware
        windchill = prefer_hardware
        windrun = prefer_hardware
```

tkeffer, 2024-01-09:
`maxSolarRad` is the maximum radiation for a location at a specific time. There is no reason why it can't always be calculated. However, if the sun is below the horizon, its value will be `None`.

`rainRate` will be zero if, well..., it hasn't been raining for awhile.

`ET` requires a bunch of values, including `radiation`, `outTemp`, `windSpeed`, `outHumidity`. If any of them are missing, you get a value of `None`. You must be missing one of them.

`inDewpoint` requires `inTemp` and `inHumidity`. If either of them is missing, you get a value of `None`.

All 4 of these are showing expected behavior.

If a station is unable to provide a value (for example, because of a missing rain gauge), WeeWX has no way of knowing that, unless you use a setting of `hardware` in `[[Calculations]]`. Then it won't try and calculate these quantities.

Or, am I misunderstanding something?

gjr80, 2024-01-12:
My 2c. 

Agree `maxSolarRad` should be zero or a calculated value. Would go a step further and say if no `ephem` then it should be omitted from packets/records (probably moot as `ephem` is a dependency now). 

I think `rainRate` needs some work. If the station is emits `rain` then `rainRate` should be zero or some calculated value. If the station does not emit rain (or has not yet emitted rain - eg partial packet station or Ecowitt station without rain gauge) then `rainRate` should be omitted from packets/records or at the very least until WeeWX sees some rainfall data. At the moment `rainRate` is zero whether there is rainfall data or not. If I remember correctly `RainRater` needs some work so that it may calculate `rainRate` from archive record `rain` (eg when importing data).

Agree `ET` is a calculated value or None. Likewise `inDewpoint`

tkeffer, 2024-01-21:
I set both `rain` and `rainRate` to null in a database, recalculated the daily summary, then calculated `$day.rainRate.max`. It resolved to `N/A'.

What are you seeing, Gary?

tkeffer, 2024-06-07:
@matthewwall  Any more comments on this? OK to close?

closed 2024-07-11 by tkeffer