weewx/weewx #916: Is this the extension or the code?

JonFear, 2024-01-09:
`jon@weewx-v5:~$ sudo weectl extension install https://github.com/chaunceygardiner/weewx-loopdata/releases/download/v3.3.2/weewx-loopdata-3.3.2.zip
sudo: unable to resolve host weewx-v5.local: Name or service not known
Using configuration file /etc/weewx/weewx.conf
Install extension 'https://github.com/chaunceygardiner/weewx-loopdata/releases/download/v3.3.2/weewx-loopdata-3.3.2.zip' (y/n)? y
Extracting from tar archive /tmp/tmp90g4sr43
Traceback (most recent call last):
  File "/usr/share/weewx/weectl.py", line 74, in <module>
    main()
  File "/usr/share/weewx/weectl.py", line 66, in main
    namespace.func(namespace)
  File "/usr/share/weewx/weectllib/__init__.py", line 121, in dispatch
    namespace.action_func(config_dict, namespace)
  File "/usr/share/weewx/weectllib/extension_cmd.py", line 116, in install_extension
    ext.install_extension(namespace.source, no_confirm=namespace.yes)
  File "/usr/share/weewx/weecfg/extension.py", line 122, in install_extension
    extension_name = self._install_from_file(test_fd.name, info.get_content_subtype())
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/extension.py", line 153, in _install_from_file
    member_names = weecfg.extract_tar(filepath, dir_name, self.printer)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weecfg/__init__.py", line 665, in extract_tar
    with tarfile.open(filename, mode='r') as tar_archive:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/tarfile.py", line 1639, in open
    raise ReadError(f"file could not be opened successfully:\n{error_msgs_summary}")
tarfile.ReadError: file could not be opened successfully:
- method gz: ReadError('not a gzip file')
- method bz2: ReadError('not a bzip2 file')
- method xz: ReadError('not an lzma file')
- method tar: ReadError('invalid header')`

tkeffer, 2024-01-09:
Flaw with WeeWX (not the extension). 

It seems that when zip files are retrieved via http, they cannot be identified as such. 

Thanks for the report!

JonFear, 2024-01-09:
Ah, OK thanks. It appears there si some issue with the code but that is up to the writer I guess!

Jon

tkeffer, 2024-01-09:
Fixed in commit https://github.com/weewx/weewx/commit/125caebf779d7c65047f038d6e5b10b67ccbe97c

closed 2024-01-09 by tkeffer