weewx/weewx #917: --compare-versions issue when updating from 4.10.2 to 5.0.0

MidwestTechnologies, 2024-01-15:
I get the following when updating my buster install from 4.10.2 to 5.0.0:
``
Setting up weewx (5.0.0-1) ...
Using root:root as user:group
  File "/usr/share/weewx/weectl.py", line 50
    version=f"weectl {weewx.__version__}")
                                        ^
SyntaxError: invalid syntax
dpkg: error: --compare-versions takes three arguments: <version> <relation> <version>
``

I would appreciate any help.

tkeffer, 2024-01-15:
The problem is that you are using Python 2. Version 5 requires Python 3.6 or greater.

closed 2024-01-15 by tkeffer

MidwestTechnologies, 2024-01-16:
While the original python version on my system was 2.7, I made sure the default was 3.6 or greater before the update.  
`$ python -V` returns 3.7.3.  The symbolic links were also updated and no alternatives are shown with the `update-alternatives` command.  I'm not sure why I'm still getting the error.

tkeffer, 2024-01-16:
See what the "shebang" (the first line in the file) of `/usr/bin/weectl` contains.

MidwestTechnologies, 2024-01-16:
``#!/bin/sh``

tkeffer, 2024-01-16:
And, what are the contents of `/etc/default/weewx`?

MidwestTechnologies, 2024-01-17:
Ah, that did it.
Original `WEEWX_PYTHON=python2`
Changed to `WEEWX_PYTHON=python3`
Re-install went without a hitch.  

Thank you, Tom, and sorry for the hassle.

matthewwall, 2024-01-30:
the debian postinst script invokes `/usr/bin/weectl`, which uses the python defined in `/etc/default/weewx`

the workaround described above will fix it, but we need to fix the postinst so that it uses a self-consistent combination of python and weectl.

this missed our testing because we did not test for upgrades to weewx where not only weewx was upgraded, but also python was changed from python2 to python3 (that requires a change of repository)

MidwestTechnologies, 2024-01-30:
Let me know if you need help testing.  5.0.0 is running smoothly on the Pi 2 but I have other Pi’s at my disposal for testing.  Appreciate the look at it.