weewx/weewx #918: `weectl device` command doesn't inherit the module path for extensions

marriotb, 2024-01-17:
I'm running `weewx-5.0.0rc3` on both MacOS and Linux platforms, installed in both cases with `pip` using virtual environments.

When a device driver is added using `weectl extension install <path>`, it is correctly installed and works as expected, but the command `weectl device` crashes out with the message: "Unable to import driver user.<devicename>: No module named 'user'."

I added a line to the exception code in `device_actions.py`to log the value of `sys.path`. It shows no reference to the directory `<WEEWX_ROOT>/bin` or its subdirectory `user`.

My initial example was an externally-developed driver (`HP1000`) but I've subsequently reproduced the issue using the `fileparse`  0.6 driver from the weewx 4.10 examples folder. In both cases the `files` dict in `install.py` specifies 'bin/user' as the destination directory.

Kind regards - you have a great tool here and it's good to see it evolving with the underlying implementation language.
Brian Marriott

tkeffer, 2024-01-17:
I was able to reproduce this as well. 

Thanks, Brian

tkeffer, 2024-01-17:
Fixed.

If you are up to it, perhaps you can test the fix by running out of the git repository?

closed 2024-01-17 by tkeffer

marriotb, 2024-01-23:
I can confirm it's now fixed in 5.0.1.
Verified by running `python3 ./src/weectl.py device` and also by printing the value of `sys.path`.

Cheers
Brian