weewx/weewx #920: Errors reported in systemctl status weewx (Seasons telemetry.tmpl missing "raw")

staze, 2024-01-19:
Things seem to be working, but seeing this error in systemctl status weewx. Updated to 5.0.0-1 from RC's. 

Jan 19 11:55:18 raspi-server-misc weewxd[11715]: ERROR weewx.cheetahgenerator: Evaluation of template /etc/weewx/skins/Seasons/telemetry.html.tmpl failed.
Jan 19 11:55:18 raspi-server-misc weewxd[11715]: ERROR weewx.cheetahgenerator: **** Ignoring template /etc/weewx/skins/Seasons/telemetry.html.tmpl
Jan 19 11:55:18 raspi-server-misc weewxd[11715]: ERROR weewx.cheetahgenerator: **** Reason: cannot find 'raw' while searching for 'lasttime.raw'
Jan 19 11:55:18 raspi-server-misc weewxd[11715]: ERROR weewx.cheetahgenerator: **** To debug, try inserting '#errorCatcher Echo' at top of template

Looks like it started with 5.0.0-1 downgrade (from rc3-4 (I think it was -4). 

Thanks!

tkeffer, 2024-01-19:
There were no changes to the skins between any of the release candidates and the release of 5.0.0. Indeed, the last change was on 30 Nov 2023, and that had to do with language translations.

A few questions

1. What type of hardware?
2. You say the "left bar in Seasons" is not updating. Which page in Seasons? `telemetry.html`?
3. Have you made any modifications to the templates? I recall you had earlier telemetry issues.
4. Have you made any modifications to the `[DisplayOptions]` section in `skin.conf`?

staze, 2024-01-19:
Yeah, that's what I thought. and it was working with rc3... 

1. Vantage Vue. 
2. I mean the main left bar. when just visiting the index for weewx. It seems stuck with old data. The graphs also got messed up. =(
3. No. I even grabbed a fresh version from github, same issue. 
4. Same. 

The issue I had previously was missing column (data type column) in weewx db that was causing it to try to calculate ET. Put the data type back, just blank, and it's fine now. 

Something else seems off though, because I'm seeing
Jan 19 15:50:17 raspi-server-misc weewxd[15156]: INFO weewx.manager: Added record 2024-01-19 15:50:00 PST (1705708200) to database 'weewx.sdb'
Jan 19 15:50:17 raspi-server-misc weewxd[15156]: INFO weewx.manager: Added record 2024-01-19 15:50:00 PST (1705708200) to daily summary in 'weewx.sdb'
Jan 19 15:50:18 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: Evaluation of template /etc/weewx/skins/Seasons/index.html.tmpl failed.
Jan 19 15:50:18 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: **** Ignoring template /etc/weewx/skins/Seasons/index.html.tmpl
Jan 19 15:50:18 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: **** Reason: cannot find 'raw' while searching for 'lasttime.raw'
Jan 19 15:50:18 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: **** To debug, try inserting '#errorCatcher Echo' at top of template
Jan 19 15:50:19 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: Evaluation of template /etc/weewx/skins/Seasons/telemetry.html.tmpl failed.
Jan 19 15:50:19 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: **** Ignoring template /etc/weewx/skins/Seasons/telemetry.html.tmpl
Jan 19 15:50:19 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: **** Reason: cannot find 'raw' while searching for 'lasttime.raw'
Jan 19 15:50:19 raspi-server-misc weewxd[15156]: ERROR weewx.cheetahgenerator: **** To debug, try inserting '#errorCatcher Echo' at top of template
Jan 19 15:50:19 raspi-server-misc weewxd[15156]: INFO weewx.cheetahgenerator: Generated 6 files for report SeasonsReport in 2.04 seconds
Jan 19 15:50:24 raspi-server-misc weewxd[15156]: INFO weewx.imagegenerator: Generated 42 images for report SeasonsReport in 5.13 seconds
Jan 19 15:50:24 raspi-server-misc weewxd[15156]: INFO weewx.reportengine: Copied 5 files to /var/www/html/weewx
Jan 19 15:50:28 raspi-server-misc weewxd[15156]: INFO weewx.cheetahgenerator: Generated 2 files for report Emkubed in 3.74 seconds
Jan 19 15:50:28 raspi-server-misc weewxd[15156]: INFO weewx.reportengine: Copied 0 files to /var/www/html/weewx/emkubed
Jan 19 15:50:30 raspi-server-misc weewxd[15156]: INFO weeutil.rsyncupload: rsync'd 55 files (703,530 bytes) in 2.18 seconds
Jan 19 15:51:02 raspi-server-misc weewxd[15156]: ERROR weewx.restx: StationRegistry: Failed to publish record 2024-01-19 15:50:00 PST (1705708200): HTTP Error 429: TOO MANY REQUESTS

based on previous discussion thread, I dropped and and re-created the daily summary. No difference. 
So something is seemingly wedged.

staze, 2024-01-19:
not sure if it's relevant, but I did have to "downgrade... crap, found it. the downgrade to weewx from rc3 back to release set my station back to simulator... =(

staze, 2024-01-20:
fixed. had something to do with the "downgrade" from 5.0.0rc3 to 5.0.0-1 (release). The process flipped my station from Vantage to Simulator which had data it was putting into DB causing an issue. 

Ran through process to remove simulator data (for only 5 hours today) and then started weewx back up (after flipping back to Vantage) and it downloaded the data from station and is back to doing it's thing. 

Thanks!

closed 2024-01-20 by staze