weewx/weewx #922: Problem installing v5 on Debian using apt repository

alexdelprete, 2024-01-29:
I'm in the process of installing v5 to Ubuntu via apt repo, as per manual. Then I'll migrate my config/data manually from an old v4 setup.py installation.

I added the key and the repo to sources.list.d, followed the instructions in the install guide.

Files are there, no issues apparently. I have the following error when updating:

```
Err:9 https://weewx.com/apt/python3 buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY ED444FCCF0E2B09E
Reading package lists... Done
W: GPG error: https://weewx.com/apt/python3 buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY ED444FCCF0E2B09E
E: The repository 'https://weewx.com/apt/python3 buster InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

It seems that the pubkey is not verifiable by Ubuntu. Has anoyone else tested this on Ubuntu?

Thanks. For now I'll just download the deb file and install manually.

alexdelprete, 2024-01-29:
Ok, found the answer in google groups, post by @matthewwall: https://groups.google.com/g/weewx-user/c/AJlPGe61xVU/m/AuOh3oiZAAAJ

Also, I couldn't find .deb packages in the release as expected.

Anyway, with the updated key the update completes correctly:

```console
❯ nala update
╭─ Updating Package List ───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╮
│No Change: https://download.docker.com/linux/ubuntu lunar InRelease                                                                                                                                                                        │
│No Change: http://ports.ubuntu.com/ubuntu-ports mantic InRelease                                                                                                                                                                           │
│No Change: http://ports.ubuntu.com/ubuntu-ports mantic-updates InRelease                                                                                                                                                                   │
│No Change: http://ports.ubuntu.com/ubuntu-ports mantic-backports InRelease                                                                                                                                                                 │
│No Change: http://ports.ubuntu.com/ubuntu-ports mantic-security InRelease                                                                                                                                                                  │
│Updated:   https://alpha.floxsdlc.com/downloads/debian-archive unstable/ InRelease [2 KB]                                                                                                                                                  │
│No Change: https://ftp.tu-chemnitz.de/pub/linux/ubuntu-ports mantic InRelease                                                                                                                                                              │
│Updated:   https://weewx.com/apt/python3 buster InRelease [4 KB]                                                                                                                                                                           │
│No Change: https://in.mirror.coganng.com/ubuntu-ports mantic InRelease                                                                                                                                                                     │
│No Change: https://ubuntu-mirror.cloud.mu/ubuntu-ports mantic InRelease                                                                                                                                                                    │
│No Change: https://ubuntu-ports.mirror.net.in mantic InRelease                                                                                                                                                                             │
│Updated:   https://weewx.com/apt/python3 buster/main all Packages [4 KB]                                                                                                                                                                   │
│Fetched 6 KB in 5s (1.2 KB/s)                                                                                                                                                                                                              │
╰───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯
All packages are up to date.
```

tkeffer, 2024-01-29:
If you're upgrading a setup.py installation, a pip install is a lot easier. See [_Migrating setup.py installs to V5_](https://github.com/weewx/weewx/wiki/v5-upgrade)

closed 2024-01-29 by tkeffer

alexdelprete, 2024-01-29:
> If you're upgrading a setup.py installation, a pip install is a lot easier

I read that before migrating. I also have a multi weewx setup. So I took some time before deciding pipx vs debian. In the end I opted for Debian, even though I had to move data from /home/weewx to the specific debian folders, then update my 3 conf files for the 3 stations. Took 1 hour but it worked fine in the end. The only issue I had was permissions: in the end I modified the systemd unit file to run as root.

I'm quite happy now: the only thing I don't like about the debian install is that db/config/html are in different places, so in order to backup/restore I need to create a script. I wish there still was the `setup.py` option, with everything in `/home/weewx`, it was simple to install/use/maintain/backup. :)

Anyway, thanks for v5. I still have to check all the new stuff, but it's working fine.

tkeffer, 2024-01-29:
I liked setup.py as well, but the Python overlords don't like us roaming all over the file system.

alexdelprete, 2024-01-29:
> Python overlords don't like us roaming all over the file system

it was ONE folder. Literally. :)

I guess I'm just an old IT dinosaur nostalgic of another era, where we had control of everything...