weewx/weewx #923: Crash robustness in restx.py

roe-dl, 2024-02-03:
I found that out with some extension that I wrote, but it applies to core WeeWX, too.

The working principle of the restful services in `restx.py` is to separate the output part into a thread. The thread and core WeeWX communicate by a queue. That queue is created by the statement `queue.Queue()`.

The problem is: If the thread dies for some reason or does not read the queue any more, the queue is still filled. So the queue becomes longer and longer and consumes more and more memory. At the end the computer dies from memory exhaustion or the Linux kernel will eventually kill the WeeWX process.

I guess the solution could be to limit the size of the queue like `queue.Queue(5)` and handle the `queue.Full` exception around the `put()` statement. The `put()` statement needs a timeout limit, too.

And yes, I experienced that memory exhaustion with some erroneous service of my own, that accidentally did not read the queue any more. After about half a day the Linux kernel killed WeeWX.

tkeffer, 2024-02-03:
This is something I've thought of, but it's not an easy problem to solve. The problem is that if a thread is misbehaving, there is no way to kill it in Python. You can only ask it to commit suicide. But, if it's defunct, it's not going to listen to a suicide request either.

One could just abandon it and launch another thread, but we don't know what resources it may have tied up, such as database connections.

Better to solve the problem at the root: what is preventing the queue to drain properly?

roe-dl, 2024-02-04:
> Better to solve the problem at the root: what is preventing the queue to drain properly?

As this thread is about robustness, it is about things that should not happen.

I had a teacher who always said: "What is possible, that will happen some time." He encouraged me to never think it is possible to get things safe.

So in this case a thread can die. It is possible. We can try to find all the possibilities, but there are more. From the perspective of the main thread, we have to consider that possibility. What happens if the thread dies? In this case the queue will be filled endlessly, until all the system memory is consumed by the queue. It is simple to prevent that case by limiting the size of the queue. In normal operation there is at maximum one entry in the queue. So limiting the size should not harm.

Another thing is, if the thread freezes, if you say. The only thing I see in the moment that can be done is to realize it and complain it to the log. So the user can do something about it. But in the meantime, readings are still saved to the database. That is the difference to not checking it.

tkeffer, 2024-02-04:
Adding a limit to the queue size is easy enough, but it has to be done by each writer of the RESTful service --- it can't be done through the WeeWX API.

How do you propose realizing that a thread has frozen, and how would you make an entry in the log?

Unless it's the main thread that freezes, values should still be stored in the database. That's one of the design goals of WeeWX. 

I don't see why the threads themselves cannot be made robust. They have two jobs: read from a database, and post to an HTTP connection. Both can have timeouts and retries and, indeed they do.

I'm not sure what problem we're trying to solve. It would help to see the failure mode that you are describing.

roe-dl, 2024-02-04:
> I'm not sure what problem we're trying to solve. It would help to see the failure mode that you are describing.

This was the log:

```
Feb  1 23:45:21 XXX kernel: [184600.440990] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/system.slice/weewx.service,task=python3,pid=37557,uid=0
Feb  1 23:45:21 XXX kernel: [184600.441028] Out of memory: Killed process 37557 (python3) total-vm:9633168kB, anon-rss:7612100kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:16620kB oom_score_adj:0
```

You see, the kernel killed the WeeWX process. No data were saved any more. It was in the middle of the night. So I realized it the next morning, missing the whole night of data. 

I am with you to do all to make the thread themself robust. No doubt, that has to be done first. But I also say, that it is not possible to find all possibilities how the thread can crash. At least not to mankind. So I always take in mind that I could have missed something.

tkeffer, 2024-02-04:
That's a standard oom message. What thread caused it? How do we know it was a thread at RESTful thread at all.

I'm trying to understand what would cause the memory to grow without bounds. Then let's fix that.

roe-dl, 2024-02-04:
No. Look. I wrote a WeeWX service involving a thread and a queue, the same way the RESTful services do. There was a bug within the thread that caused the queue not to be read. So I became aware of the problem. I debugged the thread. But additionally I replaced `queue.Queue()` by `queue.Queue(5)` and surrounded the `put` statement by `try:` and `except queue.Full:`. Now the thread cannot crash any more (I hope) and the queue cannot eat all the memory any more. 

And after all that trouble I thought that may be interesting to you, as the RESTful services use the same communication principle. I thought it may be useful not only to rely on a carefully designed thread, but also on some safety measures for the queue.

But after all, I can only present my thoughts. It is up to you to decide.

tkeffer, 2024-02-04:
If I understand you correctly, the problem was with a thread you wrote that contained a logic error. It sounds like you corrected the logic error, then added an extra layer of protection in case there is another logic error in there. Sounds like a fine solution.

I will look into setting a limit to the size of the queue. As I said upthread, that is easy enough.

I'm a little more reluctant to simply abandon a frozen thread. That has unknown implications for reasons cited above.

roe-dl, 2024-02-04:
> If I understand you correctly, the problem was with a thread you wrote that contained a logic error. It sounds like you corrected the logic error, then added an extra layer of protection in case there is another logic error in there. Sounds like a fine solution.

Yes. That is, what I wanted to say.

tkeffer, 2024-02-12:
I started implementing this, but decided not to go ahead. It adds complexity to the standard WeeWX RESTful services, but would fix a problem that so far has not been exhibited. They have proven to be robust after a decade of service. I can't recall a single instance of OOM due to an overflowing queue.

That would be OK if the implementation prevented new RESTful services from bringing the system down, but it can't do that because only the new author can create and manage the queue. The WeeWX RESTful API can't.

closed 2024-02-12 by tkeffer