weewx/weewx #925: inconsistent behavior with "weectl extension"

matthewwall, 2024-02-04:
the `weectl extension` commands do not use consistent defaults.  it looks like `weectl station create` is a special case that does not use `dispatch` - all of the commands that *do* use dispatch do not have the (correct) defaults that `station create` gets.

if i do `weectl extension install foobar.zip`, the `foobar` extension will be installed into `~/weewx-data` and the config `~/weewx-data/weewx.conf` will be modfied.

if i do `weectl extension install foobar.zip --config bazbum.conf`, i would expect the `foobar` extension to be installed into `~/weewx-data` and the config `~/weewx-data/bazbum.conf` should be modified.

that does not happen.

instead, i get a stack trace with `OSError: bazbum.conf is not a file`

this applies to all of the `extension` options.  here is an example using `list`.

```
[vagrant@localhost ~]$ weectl extension list
Using configuration file /home/vagrant/weewx-data/weewx.conf
Extension Name    Version   Description
exfoliation       0.45      A minimalist layout with lots of data.
interceptor       0.60      Capture weather data from HTTP requests
simple            0.5       A minimalist layout.
```

```
[vagrant@localhost ~]$ weectl extension list --config foobar.conf
Error parsing config file: foobar.conf is not a file
Traceback (most recent call last):
  File "/usr/share/weewx/weectllib/__init__.py", line 88, in dispatch
    config_path, config_dict = weecfg.read_config(namespace.config)
  File "/usr/share/weewx/weecfg/__init__.py", line 145, in read_config
    config_path = find_file(config_path, args,
  File "/usr/share/weewx/weecfg/__init__.py", line 106, in find_file
    raise OSError(f"{file_path} is not a file")
OSError: foobar.conf is not a file
```

```
[vagrant@localhost ~]$ weectl extension list --config weewx-data/foobar.conf
Using configuration file weewx-data/foobar.conf
Extension Name    Version   Description
exfoliation       0.45      A minimalist layout with lots of data.
interceptor       0.60      Capture weather data from HTTP requests
simple            0.5       A minimalist layout.
```

the output above was using weewx 5.0.1 on redhat 9.3 installed using yum.  one could argue that the default `WEEWX_ROOT` should be `/etc/weewx`, not `$HOME/weewx-data`, but that is a separate issue.  that could be solved by using `$HOME/weewx-data` as default only if `/etc/weewx` does not exist.

matthewwall, 2024-02-04:
more inconsistency.  if i do `weectl station create` on a machine where weewx was installed from deb/rpm, it will create a directory `~/weewx-data`

however, if i do `weectl extension install` on a machine where weewx was installed from deb/rpm, and there is no directory `~/weewx-data`, it tries to install into `/etc/weewx` (that actually fails, but that is a separate issue #926)

tkeffer, 2024-02-04:
I guess it depends on how you look at it.

My thinking was that `weectl` has a default location for the configuration file, not the station data area, but I can see it the other way around.

What's the expectation for the following directory structure

```
weewx-data
├── foo.conf
└── weewx.conf
other-data
└── foo.conf
```
with commands
```
cd ~/other-data
weectl extension list --config=foo.conf
```

matthewwall, 2024-02-05:
good example.  which has precedence, the directory `.` inferred from the filename `foo.conf` (resulting in `~/other-data/foo.conf`), or the directory `~/weewx-data` that is default since no weewx-root was specified (resulting in `~/weewx-data/foo.conf`)?

it depends on whether you see the directory or the config file as the primary object.

tkeffer, 2024-02-05:
I've always throught as the config file as the target. 

Recall, in the absence of any other information, it's inferred that `WEEWX_ROOT` is the directory the config file is sitting in, not the other way around.

tkeffer, 2024-02-05:
LOL, originally the default target was the config file `~/weewx-data/weewx.conf`, not the station data area `~/weewx-data`, and you didn't like that!

closed 2024-02-12 by tkeffer