weewx/weewx #931: latest buster deb release has no pub key

flashydave, 2024-02-11:
Failed to fetch http://weewx.com/apt/python3/dists/buster/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY E788768B9CB16E10

evilbunny2008, 2024-02-11:
https://www.weewx.com/docs/5.0/quickstarts/debian/

flashydave, 2024-02-11:
This is on a system that has been working getting updates for years and just stopped in the last few days

tkeffer, 2024-02-11:
The keys have changed. The keys we have used for years are no longer accepted on many OSs.

closed 2024-02-11 by tkeffer

flashydave, 2024-02-11:
So the fix for others experiencing the same issue is to repeat the step
```
wget -qO - https://weewx.com/keys.html | \
    sudo gpg --dearmor --output /etc/apt/trusted.gpg.d/weewx.gpg
```
    and overwite the existing weewx.gpg file.