weewx/weewx #932: 5.0.1 and 5.0.2 upgrades via deb breaking rsync id_rsa key

staze, 2024-02-11:
Looks like the upgrade process is doing a blanket chmod on /var/lib/weewx and causing the .ssh/id_rsa file to get changed to 0644, which causes rsync to fail with

```
Feb 11 11:35:40 raspi-server-misc weewxd[11637]: ERROR weeutil.rsyncupload: **** @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
Feb 11 11:35:40 raspi-server-misc weewxd[11637]: ERROR weeutil.rsyncupload: **** @         WARNING: UNPROTECTED PRIVATE KEY FILE!          @
Feb 11 11:35:40 raspi-server-misc weewxd[11637]: ERROR weeutil.rsyncupload: **** @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
Feb 11 11:35:40 raspi-server-misc weewxd[11637]: ERROR weeutil.rsyncupload: **** Permissions 0664 for '/var/lib/weewx/.ssh/id_rsa' are too open.
Feb 11 11:35:40 raspi-server-misc weewxd[11637]: ERROR weeutil.rsyncupload: **** It is required that your private key files are NOT accessible by others.
Feb 11 11:35:40 raspi-server-misc weewxd[11637]: ERROR weeutil.rsyncupload: **** This private key will be ignored.
Feb 11 11:35:40 raspi-server-misc weewxd[11637]: ERROR weeutil.rsyncupload: **** Load key "/var/lib/weewx/.ssh/id_rsa": bad permissions
```

This happend with the upgrade from 5.0.0 to 5.0.1 and 5.0.1 to 5.0.2. Fixing it just means changing the permissions back to 700. May want to exclude the .ssh directory, or at least the id_rsa file, from the chmod. 

Note, this may be a corner case as during testing the beta's of v5 my install was converted from running under root to running under weewx user.

matthewwall, 2024-02-11:
very sorry about this.  /var/lib/weewx is a perfect place to keep the ssh credentials for the `weewx` user, and the 5.0.1 and 5.0.2 postinst is too aggressive about setting permissions.  this has been dialed back at commit 2f97c0e6d4c91ade58e4fde6a42554bc2a45551b and will be included in the next release.

staze, 2024-02-11:
Hey @matthewwall no worries at all! Thanks for always being so quick getting things like this addressed. Never quite sure if I'm just a corner case for everything or not. =)

closed 2024-02-12 by tkeffer