weewx/weewx #943: weewx v5.0.2 PID file?

silke-99, 2024-03-08:
Hello, this might be a minor issue, but is PID file creation still working for v5?

I recently upgraded from v4 on Ubuntu and have always used the PID file for monitoring the process via monit.

Now, there is an new empty /run/weewx folder and no pid file in sight. I tried to fiddle with the --pidfile option and I think there is no permission issue, but no file gets created. 

weewxd[3156840]: INFO __main__: Command line: /usr/share/weewx/weewxd.py /etc/weewx/weewx.conf --pidfile /run/weewx/weewx.pid

Any hints? Thank you!

tkeffer, 2024-03-08:
Probably a permissions issue.

1. What is the ownership of your process?
2. What are the permissions for `/run/weewx`?

silke-99, 2024-03-08:
probably yes but I don't find it. weewx is run by user weewx

folders are set like this:

```
silke@enterprise:~$ ls -la /run/weewx/
total 0
drwxrwxr-x  2 weewx weewx   40 Mär  8 23:11 .
drwxr-xr-x 50 root  root  1400 Mär  9 00:22 ..
```

bellrichm, 2024-03-09:
My very limited understanding of systemd is that it doesn’t create a pid file. Here is a quick google result.
https://www.freedesktop.org/software/systemd/man/latest/systemd.service.html#PIDFile=
Posting to https://groups.google.com/g/weewx-user will get more ‘eyes’ on this, including people who know alot more about systemd than me.

tkeffer, 2024-03-09:
It looks like `/run/weewx` is owned by `root`. Try changing ownership to `weewx`.

tkeffer, 2024-03-09:
I fear that I have led you astray. The `--pidfile` option only comes into play if you specify that `weewxd` should be run as a daemon (using the `--daemon` flag), which forks the process. Forking is strongly discouraged these days.

Alternatively, if you want to monitor `weewxd`, you can get its PID with

    systemctl show --property MainPID --value weewx

silke-99, 2024-03-09:
Thanks, this is really helpful!

I have looked more into how I can let monit interact with systemd directly for checking services, and it seems there is a way! 

I really appreciate your support, many thanks again! :-)

closed 2024-03-09 by silke-99