weewx/weewx #946: WeeWX starts before clock correct when using ntp

ps-crawford, 2024-03-31:
Now that the WeeWX code runs as a systemd service there is an assumption that it will get the correct time using the 'Requires=time-sync.target' statement. However, that is not necessarily true if ntp is in use instead and that is very apparent on devices like a Raspberry PI that lacks a real-time clock to set time on boot.

A fix I found it to edit the systemd service /lib/systemd/system/weewx.service and adding ntp-wait to the pre-start list, with '-' preceding it so failure is ignored (which might be the case if no network at all, or ntp removed). So you should then have something like:

```
[Service]
ExecStartPre=-ntp-wait -v -s 2 -n 60
ExecStart=weewxd /etc/weewx/weewx.conf
```

tkeffer, 2024-03-31:
But, that won't work if systemd-timesyncd is in use, which is the default on an RPi.

ps-crawford, 2024-03-31:
I think it will just quietly fail and due to the leading '-' continue anyway. It did on my machine when I mistyped the command at first!

tkeffer, 2024-03-31:
Most users are going to use the default, which works.

If a user wants to use ntp, then some modifications should be expected.

closed 2024-03-31 by tkeffer

ps-crawford, 2024-03-31:
I just tried to add a note to https://github.com/weewx/weewx/wiki/Time-services/#ntp to reference this case, but seems I can't edit this. Or I have missed something obvious.

tkeffer, 2024-03-31:
You have to be logged into GitHub. Other than that, I don't think there are any requirements to edit anything in the wiki.

ps-crawford, 2024-03-31:
OK I tried it again and it seems to have worked. Maybe last time I forgot to add a note about the change or similar, as the 'save' button went greyed out as if doing something but it never completed.

duane-nepa, 2024-03-31:
I just installed a RTC. They are cheap and Chrony in place of NTP it will adjust the clock for you.

 

chrony – Comparison of NTP implementations (chrony-project.org) <https://chrony-project.org/comparison.html> 

 

-dk

 

Duane Kerzic
138 Old Kettle Court
Stroudsburg, PA 18360

18TVL748345
570.982.4381

KB3ZXE

 

Notice:  The information contained in the e-mail message is intended only for the confidential use of the recipient(s) named above.  This message is privileged and confidential.  If the reader of this message is not the intended recipient, you are hereby notified that you have received this message in error and that any review, dissemination, distribution, copying or use is strictly prohibited.  If you have received this communication in error, please notify us immediately by e-mail and delete the original message.

 

This e-mail and any files transmitted with it are confidential and intended for the use of the individual or entity to which they are addressed.  If you have received this e-mail in error please contact  ***@***.***> Duane P. Kerzic

ps-crawford, 2024-03-31:
The NTP vs Chrony vs OpenNTPD debate is kind of irrelevant as none of them are properly detected by systemd's built-in time check. I don't know if chrony has an equivalent wait-for-sync command to the `ntp-wait` that I used.

However it all comes back to the same point: if time matters before WeeWX is started, which is the case, you need to take extra steps to make sure it has been synchronised by whatever system you have chosen whenever it is not systemd's built-in choice.

duane-nepa, 2024-04-01:
I think you missed the RTC (REAL TIME CLOCK). HiLetgo 5pcs DS3231 High Precision RTC Real Time Clock Module 3.3V/5V for Arduino Raspberry Pi: Amazon.com: Industrial  <https://www.amazon.com/HiLetgo-DS3231-Precision-Arduino-Raspberry/dp/B01N1LZSK3/ref=sr_1_2?crid=1T6NRU49Y529M&dib=eyJ2IjoiMSJ9.i5MRxeSfNbDKjMFNUkHLWDS1AJmp4Sk1_BwXS7gax69c8abzJ1102-zG70aF6J-pwSPqJij5lfpCeFlezrHIZCFSfmczzndJmE7Zp4Jtvc7pYg3GExsE9P3hW7JoY_J8rTfNYMDJ8yK1CSg-uMbl8lnwLtgh0-e31JAZgmodatGCfQ1CqYOGTMmDeHTAAVvoWA7Jlkk1q28Zzh_CLHxlXldus9OTCh25mTJzHTl0GVg.0QVlMmGVm_20K6IqDIl8Oi6V--ZKyNacGTpFzsfYoqA&dib_tag=se&keywords=rtc+for+raspberry+pi&qid=1711939131&sprefix=rtc+for+%2Caps%2C91&sr=8-2> & Scientific

 

If you have a RTC your system has the time before WeeWX starts. Because it has the time when it boots. No need to wait for sync. 

 

-dk

evilbunny2008, 2024-04-01:
Doesn't the RPi5 come with a RTC? Pretty sure you just need to supply a battery.

duane-nepa, 2024-04-01:
Yes the RPi5 come with a RTC. All you need is a battery holder and battery, Amazon.com: Treedix 2pcs RTC Battery Case Real Time Clock Holder Compatible with Raspberry Pi 5 : Electronics <https://www.amazon.com/Treedix-Battery-Holder-Compatible-Raspberry/dp/B0CWDFN51V/> . 

 

I don’t know which version of RPi he has. I have a 4, it doesn’t come with a RTC. And I believe you can run WeeWx on a RPi2 and certainly on a 3, again no RTC.

 

-dk

ps-crawford, 2024-04-01:
Thanks for the RTC suggestions, it is a Pi 3B I am using. I have one which has a GPS board with RTC though the RTC side never seemed to work.

However, this is missing the point! The WeeWX package **should** be waiting for time-sync before starting so that ought to be a secondary consideration (after all the RTC could be wrong) and if you are using systemd's **own** method that works.

My point was if you use NTP or something else you need to modify the weewx.service file to put in a synchronisation check that actually works with your choice of time-sync method. Mine was NTP hence the use of `ntp-wait`, but for other systems you will need to check what does the same to have the same guarantee of clock correctness on starting WeeWX