weewx/weewx #947: Running WeeWX as executable (for apparmour use)

ps-crawford, 2024-03-31:
I was wondering if there are any specific reasons why /usr/bin/weewxd invokes the python script as an argument to python3, rather than to make the script a python3 executable (i.e. add '#!/usr/bin/python3' to first line and add execute permissions) and run it directly?

The reason why I made that change on my own system is so it appears by process name /usr/share/weewx/weewxd.py and so it can be constrained by an apparmor setting. This might not be a common request, but if anyone is interested then also here is the /etc/apparmor.d/usr.share.weewx.weewxd file related to my installation (it seems not a supported type for attaching, so just posted below).

```
# Last Modified: Wed Aug 17 15:05:08 2022
#include <tunables/global>

# vim:syntax=apparmor
# AppArmor policy for weewxd
# ###AUTHOR###
# ###COPYRIGHT###
# ###COMMENT###
# No template variables specified

/usr/share/weewx/weewxd.py {
  #include <abstractions/base>
  #include <abstractions/lightdm>
  #include <abstractions/totem>

  capability dac_read_search,

  deny @{HOME}/.bashrc rw,
  deny @{HOME}/.config/ rw,
  deny @{HOME}/.ssh/ rw,

  /etc/weewx/** mrk,
  /usr/share/weewx/** mrkix,
  owner /usr/lib/arm-linux-gnueabihf/ld-*.so mr,
  owner /var/lib/weewx/weewx.sdb rwk,
  owner /var/lib/weewx/weewx.sdb-journal rw,
  owner /var/www/html/weewx/** rw,
  owner /{,var/}run/weewx.pid mrwk,

}
```

ps-crawford, 2024-03-31:
Just to add some context as to why I created the AppArmor profile. Firstly I would like to say that WeeWX is quite a good design from a security point of view as it is:
- Written in python, which is a memory-safe language
- Has no need for incoming connections, so usual firewall rules (or even IPv4 NAT) prevent incoming attacks
- No longer runs as a root process, so limited attack options if exploited

However, we see occasional error messages logged when our instance is trying to send the HTML files to an externally hosted web server and that got me wondering. Could an attacker who managed to control the 3rd party machine could abuse something like outgoing transfers using SSH, that are so regular in time, to gain a reverse-shell?

True, this might be an unlikely attack vector but there always seems to be a lot of bad actors out there who are smarter and more cunning in this field than I am. The best option I could think of was to limit the whole WeeWX process by means of AppArmor to what I think it ought to be accessing, and it was trying to get AppArmor to know which process that it should be constraining that prompts this question.

matthewwall, 2024-04-07:
> I was wondering if there are any specific reasons why /usr/bin/weewxd invokes the python script as an argument to python3, rather than to make the script a python3 executable (i.e. add '#!/usr/bin/python3' to first line and add execute permissions) and run it directly?

i would prefer to use the env pattern in the two weewx entry points, `weewxd` and `weectl`:
```
#!/usr/bin/env python3
```
it gives you great flexibility to specify different/specific python interpreters, and makes the entry points obvious.

unfortunately, that is no longer the python way.  a pip install *forces* the shebang to be an explicit, absolute path to the python interpreter.  redhat and debian systems started doing this at least 5-6 years ago as well in their rpm and deb packages.

one reason for this is to make the python interpreter explicit.  another reason is to minimize the chance of breaking an application when you upgrade/modify python - it enforces a tight binding between the python interpreter and the python modules installed to use that interpreter.

weewx has always had minimal dependencies, so it is not as vulnerable to being broken by a change to python or python package.  but in order to be pythonic and stay on the "python way" bandwagon, we have removed the shebangs and added the `.py` to the entry points.

ps-crawford, 2024-04-07:
Agreed, the env pattern for the shebang seems a better approach, but I was copying the debian (ubuntu) style assuming that was normal to put the absolute path in, as for bash and similar.

The use of the .py extension also makes sense so folks see it is a python script, again as often done with .sh for sh/bash script (though thankfully not suffering the Windows file extension issues over the years!).

I guess this is an unusual situation as most folks don't bother with AppArmor, even for packages that have workable profiles already, but using the shebang seemed both a solution to this and a more natural approach to executing script-based programs.

matthewwall, 2024-04-08:
when you configure apparmor, which process gets the protections/restrictions in a profile?

for example, in the case of a deb or rpm install, the shell script /usr/bin/weewxd invokes "python3 /usr/share/weewx/weewxd.py" using `exec`.  if you write an apparmor profile for /usr/bin/weewxd, does this work properly?  does it matter whether the shell script does a direct invocation of python3, or an 'exec' of python3?

note that if you do a pip install of weewx in a python virtual environment, you end up with a 'weewxd' entry point in the `bin` directory of the virtual environment.

deb/rpm: `/usr/bin/weewxd`
```
exec "$WEEWX_PYTHON" $WEEWX_PYTHON_ARGS "$WEEWX_BINDIR/$app" "$@"
```

pip in venv: `~/weewx-venv/bin/weewxd`
```
# -*- coding: utf-8 -*-
import re
import sys
from weewxd import main
if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])
    sys.exit(main())
```

ps-crawford, 2024-04-08:
It is the `/usr/share/weewx/weewxd.py` process that is constrained by the AppArmour profile. But it is a slightly odd situation in that when running I don't really see any difference, but AppArmour does.

So in my case I added the shebang and made weewxd.py executable (by chmod), and then I changed the /usr/bin/weewxd daemon point to run directly and not as the argument to python3:
```
#!/bin/sh
app=weewxd.py

# Get the weewx location and interpreter.  Default to something sane, but
# look for overrides from the system defaults.
WEEWX_BINDIR=/usr/share/weewx
WEEWX_PYTHON=python3
[ -r /etc/default/weewx ] && . /etc/default/weewx
#exec "$WEEWX_PYTHON" $WEEWX_PYTHON_ARGS "$WEEWX_BINDIR/$app" "$@"
exec "$WEEWX_BINDIR/$app" "$@"
```

Basically just changing the last line. If I check with `sudo aa-status` then I get various things but the WeeWX script is listed as one of 7 constrained process:
```
apparmor module is loaded.
37 profiles are loaded.
25 profiles are in enforce mode.
   /usr/bin/man
   /usr/bin/pidgin
   /usr/bin/pidgin//sanitized_helper
   /usr/bin/totem
   /usr/bin/totem-audio-preview
   /usr/bin/totem-video-thumbnailer
   /usr/bin/totem//sanitized_helper
   /usr/lib/arm-linux-gnueabihf/lightdm/lightdm-guest-session
   /usr/lib/arm-linux-gnueabihf/lightdm/lightdm-guest-session//chromium
   /usr/lib/cups/backend/cups-pdf
   /usr/sbin/apt-cacher-ng
   /usr/sbin/cups-browsed
   /usr/sbin/cupsd
   /usr/sbin/cupsd//third_party
   /usr/sbin/dnsmasq
   /usr/sbin/dnsmasq//libvirt_leaseshelper
   /usr/sbin/ntpd
   /usr/sbin/rsyslogd
   /usr/share/weewx/weewxd.py
   avahi-daemon
   man_filter
   man_groff
   nvidia_modprobe
   nvidia_modprobe//kmod
   ping
12 profiles are in complain mode.
   /usr/bin/irssi
   identd
   klogd
   mdnsd
   nmbd
   nscd
   smbd
   smbldap-useradd
   smbldap-useradd///etc/init.d/nscd
   syslog-ng
   syslogd
   traceroute
7 processes have profiles defined.
7 processes are in enforce mode.
   /usr/sbin/cups-browsed (25723) 
   /usr/sbin/cupsd (25721) 
   /usr/sbin/ntpd (568) 
   /usr/sbin/rsyslogd (413) 
   /usr/bin/python3.7 (32430) /usr/share/weewx/weewxd.py
   /usr/sbin/avahi-daemon (417) avahi-daemon
   /usr/sbin/avahi-daemon (439) avahi-daemon
0 processes are in complain mode.
0 processes are unconfined but have a profile defined.
```

If I see what is running then I get:
```
$ ps -Af | grep weewx
weewx    32430     1  7 08:38 ?        00:00:00 python3 /usr/share/weewx/weewxd.py /etc/weewx/weewx.conf
localad+ 32439 32297  0 08:38 pts/0    00:00:00 grep --color=auto weewx
```
Now when the original weewxd style is used with calling python and the weewx script as its argument I get no constraining of the process seen with aa-status as just the 6 others:
```
apparmor module is loaded.
37 profiles are loaded.
25 profiles are in enforce mode.
   /usr/bin/man
   /usr/bin/pidgin
   /usr/bin/pidgin//sanitized_helper
   /usr/bin/totem
   /usr/bin/totem-audio-preview
   /usr/bin/totem-video-thumbnailer
   /usr/bin/totem//sanitized_helper
   /usr/lib/arm-linux-gnueabihf/lightdm/lightdm-guest-session
   /usr/lib/arm-linux-gnueabihf/lightdm/lightdm-guest-session//chromium
   /usr/lib/cups/backend/cups-pdf
   /usr/sbin/apt-cacher-ng
   /usr/sbin/cups-browsed
   /usr/sbin/cupsd
   /usr/sbin/cupsd//third_party
   /usr/sbin/dnsmasq
   /usr/sbin/dnsmasq//libvirt_leaseshelper
   /usr/sbin/ntpd
   /usr/sbin/rsyslogd
   /usr/share/weewx/weewxd.py
   avahi-daemon
   man_filter
   man_groff
   nvidia_modprobe
   nvidia_modprobe//kmod
   ping
12 profiles are in complain mode.
   /usr/bin/irssi
   identd
   klogd
   mdnsd
   nmbd
   nscd
   smbd
   smbldap-useradd
   smbldap-useradd///etc/init.d/nscd
   syslog-ng
   syslogd
   traceroute
6 processes have profiles defined.
6 processes are in enforce mode.
   /usr/sbin/cups-browsed (25723) 
   /usr/sbin/cupsd (25721) 
   /usr/sbin/ntpd (568) 
   /usr/sbin/rsyslogd (413) 
   /usr/sbin/avahi-daemon (417) avahi-daemon
   /usr/sbin/avahi-daemon (439) avahi-daemon
0 processes are in complain mode.
0 processes are unconfined but have a profile defined.
```

What is curious though is when I look at the running process it looks very much the same:
```
$ ps -Af | grep weewx
weewx    32364     1  5 08:37 ?        00:00:00 python3 /usr/share/weewx/weewxd.py /etc/weewx/weewx.conf
localad+ 32371 32297  0 08:37 pts/0    00:00:00 grep --color=auto weewx

```
So in both cases the process listing shows `python3 /usr/share/weewx/weewxd.py` but AppArmour sees them as different.

tkeffer, 2024-07-11:
Is this issue still active?

ps-crawford, 2024-07-15:
Yes, still an issue for running it in AppArmour. Unless anyone knows of a way to tall AppArmour to limit a python-invoked process in another way?

closed 2025-03-14 by github-actions[bot]