weewx/weewx #948: Upgrade setting permissions on weewx.conf

ps-crawford, 2024-03-31:
It appears that the upgrade process for WeeWX re-sets permissions on the /etc/weewx directory including weewx.conf and that makes the config file world-readable. However, that file can include user name and password combinations for interfacing to 3rd party sites for ftp upload, Wunderground, etc, and those should not be readable by world+dog on any multi-user system.

Would it make more sense for that file to be always set to 660 permission and any user that needs non-sudo ability to edit or read the file is then added to the weewx group?

tkeffer, 2024-03-31:
Good point. We'll take a look at it.

matthewwall, 2024-04-07:
which files should have 660 permissions?

each time the config file is modified by weectl, weectl makes a copy.  weectl does not ensure 660 permissions.

during and upgrade, there may be many copies of the config file.  all of those should have the same permissions as the 'active' config file.

if it is weewx-multi configuration, there will be multiple 'active' config files.

so i'm not sure what the right approach is here.

ps-crawford, 2024-04-07:
The only file that bothered me from a security perspective is the /etc/weewx/weewx.conf file as it has login details for 3rd party sites that might be sensitive information. You are right to point out that often has version copies/diffs placed in the same directory with differing names (sometimes appended to the '.conf' side) so all potentially have such information.

In terms of options then your 'option 3' seems the simplest and safest. My initial thought was based on looking at the `pkg/debian/postinst` file and changing it so following the (current) line 427 it removes other read/write permission, so something like this:

```
set_config_permissions() {
    echo "Setting permissions $WEEWX_USER:$WEEWX_GROUP on /etc/weewx"
    set_permissions $WEEWX_USER $WEEWX_GROUP /etc/weewx
    chmod o-rw /etc/weewx/*.conf*
}
```
This might not be perfect as there remains a window of opportunity from setting all files globally readable to removing that on the conf file(s) but given:

- Very short time they would be accessible
- Infrequent nature of such upgrade changes
- Few cases where a WeeWX machine has multiple users of low trust

It would seem quite acceptable to me.

matthewwall, 2024-04-08:
> chmod o-rw /etc/weewx/*.conf*

this is what i was thinking.

ps-crawford, 2024-04-08:
I don't understand much about the package manager process, but that seemed like the sort of change that ought to work.

matthewwall, 2024-07-04:
this will appear in weewx 5.1.0

closed 2024-07-04 by matthewwall