weewx/weewx #949: WMR300 permission error

luis96ah, 2024-04-01:
Hello,

I upgraded from 5.0.0 to 5.0.2 and encountered a permission error with the WMR300 driver:

ERROR weewx.engine: Import of driver failed: [Errno 13] Access denied (insufficient permissions) (<class 'usb.core.USBError'>)
...
weewx.service: Main process exited, code=exited, status=4/NOPERMISSION
weewx.service: Failed with result 'exit-code'.

I realized that there was a change in permissions implemented in the 5.0.1 version, so I checked the rules in /usr/lib/udev/rules.d/60-weewx.rules and had to change this one:

SUBSYSTEM=="usb",ATTRS{idVendor}=="0FDE",ATTRS{idProduct}=="CA08",MODE="0664",GROUP="weewx"

To this one:

SUBSYSTEM=="usb",ATTRS{idVendor}=="0fde",ATTRS{idProduct}=="ca08",MODE="0664",GROUP="weewx"

(identificators in lowercase)

Then, after reloading the rules (udevadm control --reload-rules) and rebooting the raspberry, weewx started working again.

I know this isn't an issue but more of a solution, but I didn't know where to put it.

tkeffer, 2024-04-01:
Thanks for your report! You're not the first to have to change the case in the vendor and product IDs.

tkeffer, 2024-04-01:
Closing is premature. We haven't changed anything yet! Let's leave it open remind ourselves of the issue.

tkeffer, 2024-04-03:
Fixed in commit https://github.com/weewx/weewx/commit/69336c795a647f103051cb94dc514a7cc66b53f6

closed 2024-04-03 by tkeffer