weewx/weewx #950: "Total Rain" is shown now for 1 hour on the next day since daylight saving

ebnerjoh, 2024-04-02:
Hello, 

two days ago we had the change to Daylightsaving (+2 instead +1 hour).

Since then I can see the "Total Rain" from the last day also for the first hour of the next day.

Br,
Johannes

tkeffer, 2024-04-02:
We will need more information than this. At the very least:
1. What kind of station?
2. What do you mean by "Total rain"? Do you mean the value in the Seasons skin under "Statistics" for "Today"? Something else? Or, did you see this in the NOAA reports?
3. Are you using hardware or software record generation?
4. What is in your database for `rain` over this time period?

ebnerjoh, 2024-04-02:
> We will need more information than this. At the very least:
>
> 1. What kind of station?
> 2. What do you mean by "Total rain"? Do you mean the value in the Seasons skin under "Statistics" for "Today"? [...]

Hi,

Ad 1: Davis Vantage Pro, I use also the Belchertown Skin

Ad 2: In the Graphs its shown as "Rain Total" but I think its "dayRain_mm" coming from the WeatherlivelinkUDP

Ad 3: hardware

Ad 4: Good question, I am not the DB Expert. Any hints for the correct query?

Br,
Johannes

tkeffer, 2024-04-02:
Let's see what's in the main archive table.

I'm assuming your archive interval is 5 minutes (300 seconds). Adjust as necessary.

sqlite> select datetime(max(dateTime),'unixepoch','localtime'), sum(rain) from archive group by date(dateTime-300,'unixepoch','localtime') order by dateTime desc limit 10;

ebnerjoh, 2024-04-02:
`2024-04-02 13:00:00|0.9
2024-04-02 00:00:00|0.900000000000001
2024-04-01 00:00:00|0.12
2024-03-31 00:00:00|0.0
2024-03-30 00:00:00|0.0
2024-03-29 00:00:00|0.46
2024-03-28 00:00:00|1.68
2024-03-27 00:00:00|0.0
2024-03-26 00:00:00|0.0
2024-03-25 00:00:00|0.04`

ebnerjoh, 2024-04-02:
2024-04-02 13:00:00|0.9
2024-04-02 00:00:00|0.900000000000001
2024-04-01 00:00:00|0.12
2024-03-31 00:00:00|0.0
2024-03-30 00:00:00|0.0
2024-03-29 00:00:00|0.46
2024-03-28 00:00:00|1.68
2024-03-27 00:00:00|0.0
2024-03-26 00:00:00|0.0
2024-03-25 00:00:00|0.04

tkeffer, 2024-04-02:
I think I messed up that last `SELECT` statement. Try this:

```
sqlite> select date(min(dateTime),'unixepoch','localtime'), sum(rain) from archive group by date(dateTime-300,'unixepoch','localtime') order by dateTime desc limit 10;
```

Compare to what you think is correct.

ebnerjoh, 2024-04-02:
2024-04-02|0.9
2024-04-01|0.900000000000001
2024-03-31|0.12
2024-03-30|0.0
2024-03-29|0.0
2024-03-28|0.46
2024-03-27|1.68
2024-03-26|0.0
2024-03-25|0.0
2024-03-24|0.04

Everything looks good, except the 02.04. as we had no rain today.

tkeffer, 2024-04-02:
OK, then let's look at the raw data for today:

```
sqlite> select dateTime,datetime(dateTime,'unixepoch','localtime') as t,rain from archive where t>="2024-04-02";
```

Thanks for your patience!

ebnerjoh, 2024-04-02:
sqlite> select dateTime,datetime(dateTime,'unixepoch','localtime') as t,rain from archive where t>="2024-04-02";
1712008800|2024-04-02 00:00:00|0.0
1712009100|2024-04-02 00:05:00|0.9
1712009400|2024-04-02 00:10:00|0.0
1712009700|2024-04-02 00:15:00|0.0
1712010000|2024-04-02 00:20:00|0.0
1712010300|2024-04-02 00:25:00|0.0
1712010600|2024-04-02 00:30:00|0.0
1712010900|2024-04-02 00:35:00|0.0
1712011200|2024-04-02 00:40:00|0.0
1712011500|2024-04-02 00:45:00|0.0
1712011800|2024-04-02 00:50:00|0.0
1712012100|2024-04-02 00:55:00|0.0
1712012400|2024-04-02 01:00:00|0.0
1712012700|2024-04-02 01:05:00|
1712013000|2024-04-02 01:10:00|
1712013300|2024-04-02 01:15:00|
1712013600|2024-04-02 01:20:00|
1712013900|2024-04-02 01:25:00|
1712014200|2024-04-02 01:30:00|
1712014500|2024-04-02 01:35:00|
1712014800|2024-04-02 01:40:00|
1712015100|2024-04-02 01:45:00|
1712015400|2024-04-02 01:50:00|
1712015700|2024-04-02 01:55:00|
1712016000|2024-04-02 02:00:00|
1712016300|2024-04-02 02:05:00|
1712016600|2024-04-02 02:10:00|
1712016900|2024-04-02 02:15:00|
1712017200|2024-04-02 02:20:00|
1712017500|2024-04-02 02:25:00|
1712017800|2024-04-02 02:30:00|
1712018100|2024-04-02 02:35:00|
1712018400|2024-04-02 02:40:00|
1712018700|2024-04-02 02:45:00|
1712019000|2024-04-02 02:50:00|
1712019300|2024-04-02 02:55:00|
1712019600|2024-04-02 03:00:00|
1712019900|2024-04-02 03:05:00|
1712020200|2024-04-02 03:10:00|
1712020500|2024-04-02 03:15:00|
1712020800|2024-04-02 03:20:00|
1712021100|2024-04-02 03:25:00|
1712021400|2024-04-02 03:30:00|
1712021700|2024-04-02 03:35:00|
1712022000|2024-04-02 03:40:00|
1712022300|2024-04-02 03:45:00|
1712022600|2024-04-02 03:50:00|
1712022900|2024-04-02 03:55:00|
1712023200|2024-04-02 04:00:00|
1712023500|2024-04-02 04:05:00|
1712023800|2024-04-02 04:10:00|
1712024100|2024-04-02 04:15:00|
1712024400|2024-04-02 04:20:00|
1712024700|2024-04-02 04:25:00|
1712025000|2024-04-02 04:30:00|
1712025300|2024-04-02 04:35:00|
1712025600|2024-04-02 04:40:00|
1712025900|2024-04-02 04:45:00|
1712026200|2024-04-02 04:50:00|
1712026500|2024-04-02 04:55:00|
1712026800|2024-04-02 05:00:00|
1712027100|2024-04-02 05:05:00|
1712027400|2024-04-02 05:10:00|
1712027700|2024-04-02 05:15:00|
1712028000|2024-04-02 05:20:00|
1712028300|2024-04-02 05:25:00|
1712028600|2024-04-02 05:30:00|
1712028900|2024-04-02 05:35:00|
1712029200|2024-04-02 05:40:00|
1712029500|2024-04-02 05:45:00|
1712029800|2024-04-02 05:50:00|
1712030100|2024-04-02 05:55:00|
1712030400|2024-04-02 06:00:00|
1712030700|2024-04-02 06:05:00|0.0
1712031000|2024-04-02 06:10:00|0.0
1712031300|2024-04-02 06:15:00|0.0

everything later looks the same

tkeffer, 2024-04-02:
It appears that, for whatever reason, the logger put a whole day's worth of rain in the archive record at 2024-04-02 00:05:00. This seems to be a Vantage firmware problem. Could you show us the results of

```
weectl device --info
```

To fix, you'll have to correct the value for 1712009100.

```
sqlite> update archive set rain=0.0 where dateTime=1712009100;
```

Make a backup of the database!

Then rebuild the daily summary for that day:

```
weectl database rebuild-daily --date 2024-04-02
```

ebnerjoh, 2024-04-02:
```
pi@Weewx:~ $ weectl device --info
Using configuration file /etc/weewx/weewx.conf
Using driver user.weatherlinkliveudp.
Cannot load configurator for WeatherLinkLiveUDP.
```

ebnerjoh, 2024-04-02:
Thanks the cleanup-commands were working.

As you stated, I also think it is related to the Davis System. Maybe this Device had an hickup after the change to daylight savings. 

I have rebooted the device and will monitor.

Many thanks for your great support here!

tkeffer, 2024-04-02:
Hang on, hang on. Sorry, but I missed that you are using the WeatherLinkLive driver. I have no idea how it calculates `rain`. You should check with the author. That could be where the problem is.

closed 2024-04-02 by tkeffer