weewx/weewx #952: Bug in Debian postinst prevent installing 5.0.2 debs

jgoerzen, 2024-04-04:
Hello,

Previously, I had been running 4.10.2.  In attempting a from-scratch install of 5.0.2, within a controlled Docker environment, I ran into an issue:

The usermod invocation is incorrect.  You can see the build log at https://salsa.debian.org/jgoerzen/docker-weewx/-/jobs/5542335#L646 and following lines.  You'll note from the "Adding user" line that `$inst_user` appears to be empty, resulting in an odd echo line and malformed usermod line.

This no doubt derives from the incorrect usage of `$USER` here.  `$USER` is not guaranteed to be defined for dpkg runs, and even if it is, it is incorrect to make decisions about how to install weewx based on how a particular shell session was invoked, because this makes the installation non-repeatable and different across systems.  It also causes issues on systems where multiple accounts with sudo access are defined, because the systemwide installation becomes dependent on which user account sudo was invoked from.  If you must persist with this approach, `$(id -nu)` would be better than `$USER`.

You'll also notice a weird broken grep at line 646.  I am unsure where that comes from.

matthewwall, 2024-04-07:
thank you for reporting this.  it has been addressed in commit ebdc08af

i say addressed, not fixed, because i am not certain that we are using the best approach here.  the intent is to make it easier to modify and manage the weewx configuration.  we are trying to balance the best practice of having weewx run as a dedicated non-root user with not getting users of weewx too tangled in the intricacies of permissions and privilege escalations.

matthewwall, 2024-04-07:
it looks like there is also a problem with the default log handler when doing a docker install

closed 2024-04-07 by tkeffer