weewx/weewx #954: View from Reports / NOAA (Belchertown Skin)

Sebeck81, 2024-04-12:
Describe the bug
Under Reports / NOAA Reports, the line “YYYY: Jan Feb Mar Apr May Jun Jul Aug Sep Oct Nov Dec” is displayed next to the years and months

Link to your website (IMPORTANT)
https://wetter2.eckelmann-online.de/reports/

Version of the skin you're using
Belchertown Skin 1.3.1

To Reproduce
In /var/www/html/weewx/Belchertown/NOAA/ there are two files: NOAA-YYYY.txt.tmpl and NOAA-YYYY-MM.txt.tmpl. As soon as I delete them the view is “clean”. After the next run the files were created again. Does anyone have an idea why the tmpl. Files to be copied to /var/www/html/weewx/Belchertown/NOAA/?

Screenshots
[image]

Weewx 5.0.2

Kind regards Sebastian

tkeffer, 2024-04-12:
It may be an issue with the `skin.conf` file for the Belchertown skin. Look in the `[CopyGenerator]` section of `skin.conf`. Does it contain an entry that could be interpreted to mean "copy the templates over?"

Sebeck81, 2024-04-12:
Hello,
many thanks for the answer. Unfortunately no, I have pasted the contents of skin.conf here.

It is identical to the content of the skin.conf from Belchtertown, which I run on weewx4.x. I don't have the problem under weewx4.x.

[CopyGenerator]

    # This section is used by the generator CopyGenerator

    # List of files to be copied only the first time the generator runs
    copy_once = favicon.ico, images/*, json/index.html, js/index.html, js/responsive-menu.js, robots.txt

    # List of files to be copied each time the generator runs
    copy_always = *.css, ..,dwd/bwk_bodendruck_weu_ana.png

tkeffer, 2024-04-12:
I'm a little suspicious of the entry "`..`". That would imply copy everything over.

There is also no need to keep copying the CSS files over with every invocation of the generator.

I'd simplify to

```
# List of files to be copied only the first time the generator runs
copy_once = favicon.ico, images/*, json/index.html, js/index.html, js/responsive-menu.js, robots.txt, *.css

# List of files to be copied each time the generator runs
copy_always = dwd/bwk_bodendruck_weu_ana.png
```

See if that makes a difference.

closed 2024-04-12 by Sebeck81