weewx/weewx #955: Problems with "Zambretti" forecast

jens-b, 2024-04-14:
In my log i found:
 Apr 14 16:45:26 raspi-test weewxd.py[2248]: forecast: MainThread: Zambretti: starting thread
Apr 14 16:45:26 raspi-test weewxd[2248]: Exception in thread ZambrettiThread:
Apr 14 16:45:26 raspi-test weewxd[2248]: INFO weewx.engine: Launch of report thread aborted: existing report thread still running
Apr 14 16:45:26 raspi-test weewxd[2248]: Traceback (most recent call last):
Apr 14 16:45:26 raspi-test weewxd[2248]:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
Apr 14 16:45:26 raspi-test weewxd[2248]:     self.run()
Apr 14 16:45:26 raspi-test weewxd[2248]:   File "/etc/weewx/bin/user/forecast.py", line 995, in run
Apr 14 16:45:26 raspi-test weewxd[2248]:     self._target(*self._args)
Apr 14 16:45:26 raspi-test weewxd[2248]: TypeError: 'NoneType' object is not callable

What is going wrong?

tkeffer, 2024-04-14:
Hello,

This is a problem with the `forecast` extension, not WeeWX. You should raise an issue with the author.

closed 2024-04-14 by tkeffer

jens-b, 2024-04-14:
Ok, thank you for the Info. I raised an issue to the forecast extension.