weewx/weewx #956: *BSD rc script

Tig9999, 2024-04-29:
Hi,
Here's a rc script for the BSD's. It fixes quite a few issues I ran into when attempting to use the older script. e.g. Works for reboots, works with MySQL, works with installed ports. 
Tested on FreeBSD 12-14, OpenBSD 7, but should work on NetBSD and Dragonfly too.
Uses native Bourne shell and BSD tools like pkill(1), pgrep(1) etc.

Regards...
[weewx-BSD.rc.zip](https://github.com/weewx/weewx/files/15153930/weewx-BSD.rc.zip)

matthewwall, 2024-05-08:
thank you for this.  when you say "older script", are you referring to the `weewx.bsd` that ships with v5, or something else?

Tig9999, 2024-05-08:
Hi,

I got an rc script off github, or maybe from weewx.com, when I installed v4.9 on FreeBSD-12 last year.
I haven't moved to v5 yet, but I just downloaded 5.0.2 it and the weewx.bsd script also has some issues.

For example, the weewx_stop and weewx_status functions only check for the existence of $weewx_pid, not if the PID in the file is currently resident. 
I would suggest that it's better to use the rc.subr function 'check_pidfile()' because it reads the pid file and confirms that the process is still running. 
Another issue I see is that if someone is using 'rsync' or any 3rd party tools, os.environ['PATH'] won't find it after a reboot of the box since all 3rd party ports are installed in /usr/local on *BSD and that won't be in $PATH yet. Probably an edge case, but this script also plays nice with MySQL, et al.

Regards...

matthewwall, 2024-07-04:
changes adopted in commits 67a0911e and b01e6285.  to appear in v5.1.0

i did not include the mysql dependency, since that is not a default weewx installation configuration

closed 2024-07-04 by matthewwall