weewx/weewx #958: [Question] How to regenerate reports for months and years

paulober, 2024-05-01:
Hi, i recently changed something in the theme for the month-%Y-%m how can i regenerate all months?

closed 2024-05-02 by tkeffer

paulober, 2024-05-02:
> Just delete your old files. WeeWX will regenerate at the next reporting cycle.
> 
> 
> 
> If you don't want to wait that long, you can use [`weectl report run`](https://www.weewx.com/docs/5.0/utilities/weectl-report/#run-reports-on-demand).
> 
> 
> 
> Incidentally, this list is for bug reports and issue tracking. For normal support questions, please use the [weewx-user](https://groups.google.com/g/weewx-user) mailing list.

Ok, so there is no option to tell the cli tool to overwrite existing files?

I know this is for bugs but why not just enable discussions on this repo instead of realying on janky external mailing lists from the pre 2000 era?🤣🤭

tkeffer, 2024-05-02:
There is no such option.

roe-dl, 2024-05-02:
> I know this is for bugs but why not just enable discussions on this repo instead of realying on janky external mailing lists from the pre 2000 era?🤣🤭

And why not accept the decisions of the repository owner?

paulober, 2024-05-02:
> And why not accept the decisions of the repository owner?

1. Cause it's a mailing list 🤯 and I'm not 60+
2. Why use 2 separate platforms if you could have both on the same with convenience features like linking to code lines or issues, and so on....

roe-dl, 2024-05-02:
@paulober There is someone who spends his spare time to make a great weather software and provide it to the public for free. And you are complaining about mailing lists and your age. Go and make something similar great and support it and then see.