weewx/weewx #961: On the first weewx generation cycle the temperatures of the previous day are not updated

areax99, 2024-06-02:
i have created this template 
`#encoding UTF-8
#import datetime
#errorCatcher Echo
{
    "current": {
        "datetime": "$current.dateTime",
        "datetime_raw": "$current.dateTime.raw"
    },
    "day": {
        "outTemp": {
            "max": "$day.outTemp.max",
            "max_formatted": "$day.outTemp.max.formatted",
            "maxtime": "$day.outTemp.maxtime",
            "min": "$day.outTemp.min",
            "min_formatted": "$day.outTemp.min.formatted",
            "mintime": "$day.outTemp.mintime"
            }
        },
    "yesterday": {
        "outTemp": {
            "max": "$day($days_ago=1).outTemp.max",
            "max_formatted": "$day($days_ago=1).outTemp.max.formatted",
            "maxtime": "$day($days_ago=1).outTemp.maxtime",
            "min": "$day($days_ago=1).outTemp.min",
            "min_formatted": "$day($days_ago=1).outTemp.min.formatted",
            "mintime": "$day($days_ago=1).outTemp.mintime"
        }
    }
}`

but when weewx performs the first generation of the new day (00:00), the details of the previous day are not updated. They are updated only in the next cycle.

`{
    "current": {
        "datetime": "02/06/2024 00:00:00",
        "datetime_raw": "1717279200"
    },
    "day": {
        "outTemp": {
            "max": "25,3 &#176;C",
            "max_formatted": "25,3",
            "maxtime": "11:06:02",
            "min": "18,1 &#176;C",
            "min_formatted": "18,1",
            "mintime": "22:59:06"
        }
    },
    "yesterday": {
        "outTemp": {
            "max": "24,3 &#176;C",
            "max_formatted": "24,3",
            "maxtime": "12:59:41",
            "min": "18,9 &#176;C",
            "min_formatted": "18,9",
            "mintime": "03:22:26"
        }
    }
}`

`{
    "current": {
        "datetime": "02/06/2024 00:05:00",
        "datetime_raw": "1717279500"
    },
    "day": {
        "outTemp": {
            "max": "18,7 &#176;C",
            "max_formatted": "18,7",
            "maxtime": "00:01:07",
            "min": "18,6 &#176;C",
            "min_formatted": "18,6",
            "mintime": "00:00:04"
        }
    },
    "yesterday": {
        "outTemp": {
            "max": "25,3 &#176;C",
            "max_formatted": "25,3",
            "maxtime": "11:06:02",
            "min": "18,1 &#176;C",
            "min_formatted": "18,1",
            "mintime": "22:59:06"
        }
    }
}`

I tried to use $yesterday.... but it gives me an error. 
I have 5.0.2 version

tkeffer, 2024-06-02:
I am having a hard time understanding your post. The formatting is difficult to follow, the data you used are unknown, and the I don't know what you expected. Nevertheless, let me try to explain what I think you are asking.

If the generation time is 02-06-2024 00:00, then as far as weewx is concerned, "today" is **01-06-2024**. That's because the midnight record actually belongs to the _previous_ day. The reason is that all of the data in the record came from 01-06-2024 from 23:55 to 00:00. Also, note that the max temperature for "today", that is, 01-06-2024, is 25.3°C.

Because "today" is 01-06-2024, "yesterday" would be 31-05-2024, the last day in May.

Now advance to the next archive record, that is 02-06-2024 00:05. "Today" is now 02-06-2024, and yesterday is 01-06-2024. If you look at the data you provided, the max value for "yesterday" is 25.3°C, which is the same value as "today" of the previous archive record.

The key point is that archive records timestamped at midnight actually belong to the _previous_ day.

Is that what you're asking?

areax99, 2024-06-02:
@tkeffer Exactly. I thought that from precisely midnight weewx would give the data for the new day. Sorry for the unclear English

closed 2024-06-02 by tkeffer

areax99, 2024-06-02:
Maybe it's clearer that way

[image]

[image]