weewx/weewx #962: shim deprecated wee_ commands for a while

vinceskahan, 2024-06-07:
It is remarkable how many "where did wee_xyz go" posts we see.  The move of everything to weectl is great, but users are clearly reading ancient notes and howtos and 'not' reading the upgrade and utilities guides that were so painstakingly written...

Suggest that the old wee_ commands be shimmed with something obvious for at least a while.  A bash example might look like something like the following....

```
#!/bin/sh
echo ""
echo "###################################################################"
echo "#                                                                 #"
echo "# ERROR - command replaced with 'weectl' as of weeWX version 5.0  #"
echo "#         please see the Upgrade and Utilities Guides for details #"
echo "#                                                                 #"
echo "###################################################################"
echo ""
exit 1
```

Suggest the same message be returned for the wee_config, wee_database, wee_debug, wee_device, wee_extension, wee_import, and wee_reports commands.

Unfortunately I cannot follow how exactly your existing weectl command is created through all the makefile and pip twisty little passages, so I'm at a loss regarding how to make this happen in terms of a PR, but basically suggest you add (add back) the old commands with a big honkin' unambigious message for the folks who simply will not RTFM.

duane-nepa, 2024-06-07:
That's a great idea.

So you have an idea how I got there. I was trying to add an extension. The git for the extension said to use wee_extension for the install. This didn't work. I google searched for wee_extension and I got to this guide. https://www.weewx.com/docs/4.10/utilities.htm

Even if there was a note on the top of that guide that said it was deprecated that would have been the clue I needed to look elsewhere. Maybe just the CSS could be edited. 

I don't often need to do anything to my setup. I was switching home automation software to Home Assistant and trying to get station data connected to that. And thought I was going to use MQTT.

But instead, I found an easy way to scrape the rss.xml file.

But now that I've seen the ver. 5 docs I'll know if I'm in the wrong place in the future.

Hope this helps.

roe-dl, 2024-06-07:
May be, it could be an idea, if someone checks all the commonly used extensions and creates an issue with them, if the documentation does not reflect the changed tools.

tkeffer, 2024-06-16:
I'm not sure how to package a shell script using Poetry, and then install it using `pip`. If there's a way to do that, I haven't found it.

It would have to be a bunch of small Python programs.

vinceskahan, 2024-06-16:
Python would be equally fine.  The bash snippet was just an example wording that exits non-zero for discussion purposes.

My guess (untested) is that you can probably do it in bash if so inclined and that poetry will handle it just as it handles adding systemd files and skins and the like.  They're not python either.  But I don't know about whether poetry can/would set the permissions executable or not.  Maybe a 8-liner python equivalent would be simplest to do.

I ignored any .deb or .rpm packaging completely but those should be just adding more stuff to the bin directory hopefully.

vinceskahan, 2024-06-19:
Initial PR is here - https://github.com/weewx/weewx/pull/964

tkeffer, 2024-10-30:
This problem seems to have mostly gone away as people have adapted to `weectl`.

closed 2024-10-30 by tkeffer