weewx/weewx #963: Error when starting weewxd

mjm455, 2024-06-08:
I ran "sudo weectl station reconfigure" and picked Simulator but the error is mostly similar:

Jun 7 04:41:21 raspberrypi systemd[1]: Started WeeWX.
Jun 7 04:41:22 raspberrypi weewxd[17162]: INFO main: Initializing weewxd version 5.0.2
Jun 7 04:41:22 raspberrypi weewxd[17162]: INFO main: Command line: /usr/share/weewx/weewxd.py /etc/weewx/weewx.conf
Jun 7 04:41:22 raspberrypi weewxd[17162]: INFO main: Using Python 3.9.2 (default, Mar 12 2021, 04:06:34) https://github.com/themoosman/weewx-ambientweatherapi-json/issues/12[GCC 10.2.1 20210110]
Jun 7 04:41:22 raspberrypi weewxd[17162]: INFO main: Located at /usr/bin/python3
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO main: Platform Linux-6.1.21+-armv6l-with-glibc2.31
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO main: Locale: 'en_GB.UTF-8'
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO main: Entry path: /usr/share/weewx/weewxd.py
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO main: WEEWX_ROOT: /etc/weewx
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO main: Configuration file: /etc/weewx/weewx.conf
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO main: User module: /etc/weewx/bin/user
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO main: Debug: 0
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.engine: Loading station type Simulator (weewx.drivers.simulator)
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.engine: StdConvert target unit is 0x1
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.wxservices: StdWXCalculate will use data binding wx_binding
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.engine: Archive will use data binding wx_binding
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.engine: Record generation will be attempted in 'hardware'
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.engine: Using archive interval of 300 seconds (specified in weewx configuration)
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.restx: StationRegistry: Station will be registered.
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.restx: Wunderground: Posting not enabled.
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.restx: PWSweather: Posting not enabled.
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.restx: CWOP: Posting not enabled.
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.restx: WOW: Posting not enabled.
Jun 7 04:41:23 raspberrypi weewxd[17162]: INFO weewx.restx: AWEKAS: Posting not enabled.
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: Caught unrecoverable exception:
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** init() takes 1 positional argument but 3 were given
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** Traceback (most recent call last):
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** File "/usr/share/weewx/weewxd.py", line 160, in main
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** engine = weewx.engine.StdEngine(config_dict)
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** File "/usr/share/weewx/weewx/engine.py", line 89, in init
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** self.loadServices(config_dict)
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** File "/usr/share/weewx/weewx/engine.py", line 157, in loadServices
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** obj = weeutil.weeutil.get_object(svc)(self, config_dict)
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** TypeError: init() takes 1 positional argument but 3 were given
Jun 7 04:41:24 raspberrypi weewxd[17162]: CRITICAL main: **** Exiting.
Jun 7 04:41:24 raspberrypi weewxd[17162]: Traceback (most recent call last):
Jun 7 04:41:24 raspberrypi weewxd[17162]: File "/usr/share/weewx/weewxd.py", line 265, in
Jun 7 04:41:24 raspberrypi weewxd[17162]: main()
Jun 7 04:41:24 raspberrypi weewxd[17162]: File "/usr/share/weewx/weewxd.py", line 160, in main
Jun 7 04:41:24 raspberrypi weewxd[17162]: engine = weewx.engine.StdEngine(config_dict)
Jun 7 04:41:24 raspberrypi weewxd[17162]: File "/usr/share/weewx/weewx/engine.py", line 89, in init
Jun 7 04:41:24 raspberrypi weewxd[17162]: self.loadServices(config_dict)
Jun 7 04:41:24 raspberrypi weewxd[17162]: File "/usr/share/weewx/weewx/engine.py", line 157, in loadServices
Jun 7 04:41:24 raspberrypi weewxd[17162]: obj = weeutil.weeutil.get_object(svc)(self, config_dict)
Jun 7 04:41:24 raspberrypi weewxd[17162]: TypeError: init() takes 1 positional argument but 3 were given
Jun 7 04:41:25 raspberrypi systemd[1]: weewx.service: Main process exited, code=exited, status=1/FAILURE
Jun 7 04:41:25 raspberrypi systemd[1]: weewx.service: Failed with result 'exit-code'.
Jun 7 04:41:25 raspberrypi systemd[1]: weewx.service: Consumed 3.927s CPU time.

tkeffer, 2024-06-08:
Please run the utility `weectl debug --output=/var/tmp/weewx.txt`. Look over the file `/var/tmp/weewx.tmp` to make sure that all of your passwords have been obfuscated, then post the file. 

It will contain a copy of your configuration file `weewx.conf`. I suspect one of the options got malformed.

mjm455, 2024-06-08:
Thanks for your help on this.  I was about to refresh the whole OS and start over but I haven't really done much to it.  I'm installing this on a Raspberry Pi Zero W.  Here is the output:

pi@raspberrypi:/etc/weewx $ more /var/tmp/weewx.txt

System info
  Platform:       Linux-6.1.21+-armv6l-with-glibc2.31
  Python Version: 3.9.2

Load Information
   1 minute load average:  0.08
   5 minute load average:  0.03
  15 minute load average:  0.01

General Weewx info
  Weewx version 5.0.2 detected.

Currently installed extensions
Extension Name    Version   Description
ambientweatherapi 0.0.11    WeeWx AmbientWeather API Driver.
interceptor       0.60      Capture weather data from HTTP requests

Archive info
  Database name:        weewx.sdb
  Table name:           archive
  Version               4.0
  Unit system:          None (Unknown unit constant)
  First good timestamp: ******* N/A *******     (    N/A   )
  Last good timestamp:  ******* N/A *******     (    N/A   )
  (no archive records found)
                        It is likely that the database (weewx.sdb) archive table (archive)
                        exists but contains no data.
  weewx (weewx.conf) is set to use an archive interval of 300 seconds.
  The station hardware was not interrogated to determine the archive interval.

Supported SQL keys
  dateTime                  usUnits                   interval
  altimeter                 appTemp                   appTemp1
  barometer                 batteryStatus1            batteryStatus2
  batteryStatus3            batteryStatus4            batteryStatus5
  batteryStatus6            batteryStatus7            batteryStatus8
  cloudbase                 co                        co2
  consBatteryVoltage        dewpoint                  dewpoint1
  ET                        extraHumid1               extraHumid2
  extraHumid3               extraHumid4               extraHumid5
  extraHumid6               extraHumid7               extraHumid8
  extraTemp1                extraTemp2                extraTemp3
  extraTemp4                extraTemp5                extraTemp6
  extraTemp7                extraTemp8                forecast
  hail                      hailBatteryStatus         hailRate
  heatindex                 heatindex1                heatingTemp
  heatingVoltage            humidex                   humidex1
  inDewpoint                inHumidity                inTemp
  inTempBatteryStatus       leafTemp1                 leafTemp2
  leafWet1                  leafWet2                  lightning_distance
  lightning_disturber_count lightning_energy          lightning_noise_count
  lightning_strike_count    luminosity                maxSolarRad
  nh3                       no2                       noise
  o3                        outHumidity               outTemp
  outTempBatteryStatus      pb                        pm10_0
  pm1_0                     pm2_5                     pressure
  radiation                 rain                      rainBatteryStatus
  rainRate                  referenceVoltage          rxCheckPercent
  signal1                   signal2                   signal3
  signal4                   signal5                   signal6
  signal7                   signal8                   snow
  snowBatteryStatus         snowDepth                 snowMoisture
  snowRate                  so2                       soilMoist1
  soilMoist2                soilMoist3                soilMoist4
  soilTemp1                 soilTemp2                 soilTemp3
  soilTemp4                 supplyVoltage             txBatteryStatus
[... 574 lines cut ...]
    [[MySQL]]
        driver = weedb.mysql
        # The host where the database is located.
        host = localhost
        # The user name for logging in to the host.
        user = XXXXXX
        # Use quotes around the password to guard against parsing errors.
        password = XXXXXX

##############################################################################

#   This section configures the internal weewx engine.

[Engine]

    # This section specifies which services should be run and in what order.
    [[Services]]
        prep_services = weewx.engine.StdTimeSynch
        data_services = ,
        process_services = weewx.engine.StdConvert, weewx.engine.StdCalibrate, weewx.engine.StdQC, weewx.wxservices.StdWXCalculate
        xtype_services = weewx.wxxtypes.StdWXXTypes, weewx.wxxtypes.StdPressureCooker, weewx.wxxtypes.StdRainRater, weewx.wxxtypes.StdDelta
        archive_services = weewx.engine.StdArchive
        restful_services = weewx.restx.StdStationRegistry, weewx.restx.StdWunderground, weewx.restx.StdPWSweather, weewx.restx.StdCWOP, weewx.restx.StdWOW, weewx.restx
.StdAWEKAS, user.ambientweatherapi.AmbientWeatherAPI
        report_services = weewx.engine.StdPrint, weewx.engine.StdReport

tkeffer, 2024-06-09:
Best I can tell, the extension `ambientweatherapi` is not initializing the driver correctly. It is doing

```
def loader(config_dict, engine):
    station = AmbientWeatherAPI(**config_dict[DRIVER_NAME])
    return station
```

but it should be 

```
def loader(config_dict, engine):
    station = AmbientWeatherAPI(engine, config_dict))
    return station
```

See [line 26](https://github.com/themoosman/weewx-ambientweatherapi-json/blob/e4637ddac9a968170a032a2ec76bae2a85570e53/bin/user/ambientweatherapi.py#L24C1-L27C19).

mjm455, 2024-06-16:
But the base install is not working for me so it cannot be the extension.  Do I need to refresh the OS and start from scratch?  Not sure what else to try.

tkeffer, 2024-06-16:
Not sure what you mean by "the base install is not working." Do you mean the installation of WeeWX? From your first post, it looks like WeeWX installed just fine, but the extension is not working. 

If WeeWX installed properly, it should not be necessary to reinstall the OS or WeeWX. Just fix the installation and reinstall it.

Or, am I misinterpreting what I'm seeing?

For the moment, I'll reopen the issue.

tkeffer, 2024-06-16:
I'm afraid that I may have led you slightly astray. 

I tried installing the ambient weather extension and got the same error as you. The problem is that it tries to install itself as a WeeWX service. That is, it includes itself among the RESTful services in option `restful_services`. See [line 17](https://github.com/themoosman/weewx-ambientweatherapi-json/blob/master/install.py#L17) of the installer.

Yet the actual code is not set up as a WeeWX service. It's set up as a driver.

I don't see how the extension could have ever worked as written. You'll have to contact the extension author. 

Sorry.

closed 2024-06-16 by tkeffer

mjm455, 2024-06-16:
I tried installing it and running it as a simulator but that wasn't working either when I tried running the service.  But now I see your latest post here and you're saying that it needs to run as a driver.  Is that right?  I can run /etc/weewx/weewxd then in the background.  

I posted the error I had on the extension page but he was pointing out that I need to get the base install working first.  Maybe that is the issue with the ambient weather api is that it is looking to work as a service.  

I'm new to the weather station and running this software.  I just really want to keep my data local and have many years stored.  Thanks!