weewx/weewx #965: weewx import brought incorrect data for rain

mjm455, 2024-06-27:
I have weewx running with the ambientweather api and everything is working great.  I've had my WS-2902 for about 3 years so I wanted to import my data from Weather Underground.  It looked somewhat easy and it appeared to have worked.  However when I looked at my record for daily rainfall, it showed it at 405" on January 28th.  I then went to Weather Underground and checked that date and it showed a total of 2.5".  

It seemed like a few of the dates in that month were giving different numbers.  I also noticed that there were a bunch of days that month and in the previous few years that did not show any data in Weewx despite there being data on WU.com.  

What did I do wrong and/or how do I fix this?  I tried looking around for a fix but didn't see anyone else with this issue.  Thanks!

gjr80, 2024-06-27:
The issues list is for tracking bugs in WeeWX, general support is handled through the [weewx-user Google group](https://groups.google.com/g/weewx-user). But since you've started the issue here we might as well deal with it. I'm going to need more information, as a start:

1. you say '_when I looked at my record for daily rainfall, it showed it at 405" on January 28th_', where did you see that? NOAA format reports? the archive table? or daily summaries?

2. what WeeWX version are you using?

3. in any case let's see what is in your archive for 28 January, assuming you are using a SQLite database and your database is `/home/weewx/archive/weewx.sdb` try running the following query and posting the output:
```
echo "SELECT min(usUnits),max(usUnits),min(rain),max(rain),sum(rain) FROM archive WHERE dateTime>unixepoch('2024-01-28 00:00:00', 'utc') and dateTime<=unixepoch('2024-01-29 00:00:00', 'utc');" | sqlite3 /home/weewx/archive/weewx.sdb
```
Note: the above assumes the sqlite3 package is installed, if it is not installed install it using your systems package manager, also if your database is named/located elsewhere, change `/home/weewx/archive/weewx.sdb` to suit 

4. exactly what command did you use to import the data? did the command execute without error or warning? do you have a copy of the console output when you ran the command?

5. what was in the WeeWX log due to the import, where there any errors?

6. please attach a copy of your import config file, you might want to ensure it does not contain any API keys before posting

7. what is your WU station ID, email to me at gjroderick (at) gmail dot com if you don't wish to post here

mjm455, 2024-06-28:
Thank you so much for your help on this.  I've been to the Google forum but wasn't sure which place to post.  I'll send any future issues to that site instead.

As for your questions:

1) I'm seeing this data from the Belchertown skin that I installed and when I looked at Reports at the top.  So these are from NOAA.

2) 5.0.2

3) After I installed Weewx, I noticed that there wasn't any database software.  So I installed apache2.  Is that ok?  Or should I install sqlite3?  So my sdb file is located at /var/lib/weewx/.
[wu.conf.txt](https://github.com/user-attachments/files/16023368/wu.conf.txt)

4)  weectl import --import-config=/var/tmp/wu.conf --from=2024-01-01T22:30 --to=2024-02-01T06:00

5) I reran the import command I posted above and I'm seeing this error (multiple times in a row):

ERROR weewx.manager: Replace failed for database weewx.sdb: database is locked

The permissions on the file are:

-rw-rw-r--  1 weewx weewx 68222976 Jun 27 22:22 weewx.sdb

6) I orginially had interval=derive but I changed it to conf hoping to fix my issue.  See attachment for my wu.conf file (I added txt so I could upload it here).

7) I'll send that soon.  Guessing this is a database issue.

gjr80, 2024-06-29:
Thank you, a few comments/suggestions:

If you notice an issue in the NOAA format reports it is quite possible the NOAA format reports contain old data; running `weectl import` causes the archive table and daily summaries to be updated, but it does not in itself cause any NOAA format reports to be rebuilt. I suggest you delete all of the NOAA format reports on your WeeWX machine (they should be somewhere under `/var/www/html/weewx`) and WeeWX will recreate them all on the next report cycle (depending on how much data is in your database this could take a few minutes). Once the reports are recreated have a look at them and see if the issue is resolved, bear in mind if you are looking at the reports via a hosted web page make sure the recreated reports have been uploaded to the remote server and also ensure any web browser caches have been cleared. 

Apache is a web server and sqlite3 is a SQLite database utility, they perform entirely different functions. If you have `.sdb` files in `/var/lib/weewx` then you almost certainly are using SQLite as your database. If you were using MySQL you would have had to have setup permissions etc in MySQL; you would certainly know that you are using MySQL.

The `[Databases]` and `[Database Types]` stanzas in `weewx.conf` define the SQLite and MySQL database parameters, by default both type of databases are defined. The actual database used is specified by the `[StdArchive]` `data_binding` setting and the respective `[DataBindings]` `[[]]` definition. By default the `wx_binding` database binding is used. WeeWX logs the database being used during startup, so if you restart WeeWX and monitor the log you will know for sure what database (and type) is being used. 

The sqlite3 package provides a command line utility allowing you to execute SQL commands on an SQLite database, WeeWX does not require it so you will see no  mention of it within WeeWX. However, humans require it if they wish to execute SQL commands on an SQLite database from the command line. I suggest you install the sqlite3 package and try again.

Running `weectl import` again is unlikely to achieve anything unless you removed the previously imported data as `weectl import` does not (at the moment) overwrite existing data.

In terms of permissions which user were you logged in as when you ran `weectl import`? Given those permissions only users that are either `weewx` or included in the `weewx` group will be able to write to the database; any other user will encounter a 'read only' error.

The interval setting should not have a significant effect on imported rain. 

I suggest you install the sqlite3 utility and query the database as suggested in my previous post. Until we know what is in the database we can only guess at what happened and how to fix it.

mjm455, 2024-06-29:
Thanks for the response.  I tried running that command above and I get this error:

Error: near line 1: no such function: unixepoch

I'm running sqlite3 v3.34.1 and I'm on Debian Bullseye.  Is there a different command I can run?

Also I removed the NOAA directories (there is one under /var/www/html/weewx and another under /var/www/html/weewx/belchertown).  I then ran sudo weectl report run and it recreated them.  However I'm noticing huge numbers again in older months and the current one.

gjr80, 2024-06-29:
Yes, you are correct, `unixepoch` was not added until sqlite3 v3.38. I included the `unixepoch` calls to make entering the start and stop timestamps of the query a little easier, as the start and stop timestamps will depend on your systems timezone. No matter, we will handle it slightly differently.

What you need to find out is the unix epoch timestamp for your systems location for midnight at the start of 28 January and again for midnight at the start of 29 January. To do this on any PC goto the page https://www.epochconverter.com. Fill out the Yr/Mon/Day/Hr/Min/Sec boxes with 2024/1/28/0/0/0, select 'AM' from 'AM/PM', select 'Local time' from 'GMT/Local time'. Click the 'Human date to Timestamp' button and four lines should be displayed underneath on the page. On the last of the four lines check the GMT value matches your systems time zone. Assuming it is correct, take note of the 'Epoch timestamp' number at the first of the four lines, that is the start timestamp you will use. Now change the Day box to 29. Click 'Human date to Timestamp' again and the four lines should refresh with a different Epoch timestamp (most likely 86400 greater than the previous). Take note of this number, it is your stop timestamp.

Now try the following command replacing `start_ts` with the first timestamp you noted and `stop_ts` with the second timestamp you noted:
```
echo "SELECT min(usUnits),max(usUnits),min(rain),max(rain),sum(rain) FROM archive WHERE dateTime>start_ts and dateTime<=stop_ts;" | sqlite3 /home/weewx/archive/weewx.sdb
```
Post the output of that command, if it returns anything other than five values post the command you entered (exactly as entered) as well.

mjm455, 2024-06-29:
Maybe I should just upgrade at some point soon.  But I got the numbers for 6/6/24 where NOAA report is showing a massive 45.61" for that day when it should 0.59.  Here are the results from the command:

pi@raspberrypi:~ $ echo "SELECT min(usUnits),max(usUnits),min(rain),max(rain),sum(rain) FROM archive WHERE dateTime>1717632000 and dateTime<=1717718399;" | sqlite3 /var/lib/weewx/weewx.sdb
1|1|0.0|0.59|19.69

and here is the NOAA txt file for that day:
[NOAA-2024-06.txt](https://github.com/user-attachments/files/16040767/NOAA-2024-06.txt)

gjr80, 2024-07-01:
Just to confirm, your timezone is GMT? (1717632000  = 00:00 6 June 2024 GMT)

```
pi@raspberrypi:~ $ echo "SELECT min(usUnits),max(usUnits),min(rain),max(rain),sum(rain) FROM archive WHERE dateTime>1717632000 and dateTime<=1717718399;" | sqlite3 /var/lib/weewx/weewx.sdb
1|1|0.0|0.59|19.69
```
The above tells me your WeeWX database uses US customary units so the database rain values are in inches. On 6 June the maximum rainfall that fell in any one archive interval was 0.59 inches, if your archive interval is 5 minutes that is a lot, but not impossible. The total rainfall for 6 June according to your archive table was 19.69 inches, sounds excessive to me. 

So I suspect the issue is not one of reports not being generated correctly or daily summaries being out of whack it is a fundamental problem with the data in you database. The easiest thing to do is to delete the imported data and re-import it. I will put together some instructions for this and post them here. In the meantime could you let me know you WU station ID, it would help to see what data actually should be coming from WU.

The permissions error is a separate issue that is not the cause of the excessive rain problem.

mjm455, 2024-07-02:
My timezone is EST.  So I put in that website GMT -4hrs.

I just sent you an email with my station ID.

gjr80, 2024-07-02:
Thanks. Having looked at your WU data for 6 June I can see how you ended up with the imported rain data you have, but I cannot see why. On 6 Jun you had a total of 0.59 inches of rain from roughly three periods of rain. WU records the cumulative daily total. WeeWX on the other hand stores rain as a per-archive interval value, but after the data was imported WeeWX appears to have stored the daily cumulative value at the time. For example, each archive record after it stopped raining for the day contains 0.59inches, these records should contain 0 inches and of course will result in a daily rain value that is too high.

The `weectl import` utility knows that WU provides cumulative rain data and consequently when importing WU data the imported rain value for each archive record is calculated by taking the difference between the current WU cumulative rain value and the previous cumulative rain value. This is not something that can be easily changed by the user (eg via config file setting), so I am at a bit of a loss as to how this could occur. I'd like to do a few tests, could you please post a copy of the import config file you used when you imported the 6 June data? Your WU import config file will contain your WU station ID and key, please make sure you remove those before posting, otherwise please don't change anything. If GitHub won't allow posting of the file you can email it to me using the same address as you sent me your station ID.

mjm455, 2024-07-03:
Sounds complicated but glad you know what you are doing!

Here is my wu.conf file (added txt extension just for this attachment)
[wu.conf.txt](https://github.com/user-attachments/files/16075493/wu.conf.txt)

gjr80, 2024-07-03:
Thanks. I think we have a bug in the v5.x.y `weectl import`; I've taken your import config file and used it verbatim, but with my WU station ID and key to import a few rainy days of data from WU. Under WeeWX v5.0.2 I see the same problem you experienced; archive period rain values track the WU cumulative rain value resulting in inflated daily rainfall values. Did the same under WeeWX v4.10.2 and it imported just fine.

I'm certain it is not your import config file, the only out of the ordinary setting is `interval = conf`. Changed this to use a fixed value with `interval = 5` and same result.

Let me look at this further.

gjr80, 2024-07-03:
OK. Found the problem, a setting was omitted from the field map in the example import config file. You need to edit your import config file, locate the `[[FieldMap]]` stanza near the end of the file and insert `is_cumulative = True` under the `[[[rain]]]` stanza as follows:
```
        [[[rain]]]
            source_field = precipTotal
            is_cumulative = True
```

Save the file.

You will have to re-import the previously imported WU data, but before you can do this you will need to delete the previously imported WU data. Once the data is re-imported you will need to re-calculate the daily summaries for the import period and delete affected NOAA format reports and then all should be good. The following instructions should help:

1. Stop WeeWX and make a backup copy of your database file `/var/lib/weewx/weewx.sdb`

2. You now need to determine the timestamps of the first and last records you imported from WU. Again the easiest approach is to use https://www.epochconverter.com. Enter the year, month, day, hour, minute, second of the first record in the Yr, Mon, Day, Hr, Min, Sec fields. Set the AM/PM (if visible) as required and set GMT/Localtime to Localtime. Click 'Human date to Timestamp', read off the Epoch timestamp immediately below ensuring the 'Date and time (your time zone)' is correct. This is important, if you get the wrong timestamps you will delete the wrong data. Now do the same for the date-time of the last record imported from WU.

3. Delete the imported records from your database using the following command replacing `first_ts` and `last_ts` with the timestamps obtained at 2. for your first and last imported records:
```
echo "DELETE FROM archive WHERE dateTime>first_ts and dateTime<=last_ts;" | sqlite3 /home/weewx/archive/weewx.sdb
```

4. Re-import your WU data using the previously modified import config file

5. Rebuild your daily summaries using the following command:
```
weectl database rebuild-daily --from=YYYY-mm-dd --to=YYYY-mm-dd
```
setting the `--from` and `--to` dates to match the date of the first and last records imported accordingly.

6. Find the NOAA format reports on your WeeWX machine, these will likely be in `/var/www/html/weewx/NOAA`. You need to delete all month and year NOAA format files from the date of the first record imported through until the present. For example, if you imported WU data from March 2023 - June 2023 you need to delete (1) all NOAA month files from `NOAA-2023-03.txt` through until the current file (`NOAA-2024-07.txt`) and (2) all NOAA year files from `NOAA-2023.txt` until the current file (`NOAA-2024.txt`). I am not sure if the Belchertown skin generates its own NOAA format files or not, if it does you need to delete them as well. The NOAA files will be re-generated on the next WeeWX report cycle.

7. Restart WeeWX

You should now see the correct rainfall data in WeeWX/Belchertown and the NOAA format reports.

The permission problem you highlighted is common under WeeWX v5. I suggest you have a look at the [Wiki page on permissions](https://github.com/weewx/weewx/wiki/Understanding-permissions), in particular the 'The WeeWX user' and 'Writing reports' sections.

mjm455, 2024-07-04:
That did the trick.  Great work.  
Here is my latest NOAA file for last month:
[NOAA-2024-06.txt](https://github.com/user-attachments/files/16092089/NOAA-2024-06.txt)

I only did the commands above for last month so now I need to repeat these steps but start from 2022 until today.

Thanks again!

mjm455, 2024-07-24:
Figured I would post in this thread rather than starting a new one because I noticed something off about my Belchertown skin page.  All of the charts are dated July 7th.  Even if I change them to show the whole month.  However when I look at reports, they are showing each day.  The coincidence with 7/7/24 is that was the last time I ran the command above to rebuild database.  I'm guessing that has something to do with it.

How do I fix this?

gjr80, 2024-08-18:
The original import problem with the missing `is_cumulative = True` was fixed at commit [a5ac2fd](https://github.com/weewx/weewx/commit/a5ac2fd77378425a583c83a84cf51ff6f3d1ba84). If the Belchertown issue persists it should be raised in the [weewx-user Google group](https://groups.google.com/g/weewx-user) or an issue raised against the Belchertown GitHub repo.

Closing this issue.

closed 2024-08-18 by gjr80