weewx/weewx #969: Permission error BYOWS

groetg, 2024-07-09:
I installed Weewx 5.1 on a RPI 2 and made a BYOWS station. It worked, but I screwed up installing updates and shutdown the RPI, so my SD was skrewed. I made a new installation, but no having trouble to get Weewx with BYOWS driver working. BYOWS scripts work, Weewx installing works (works fine with simulator. But when I install BYOWS driver and set the BYOWS as default in weewx.conf I get this error:
---------------------------------------------------
groetg@byows-jim:~ $ sudo systemctl status weewx
× weewx.service - WeeWX
     Loaded: loaded (/lib/systemd/system/weewx.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2024-07-09 17:38:35 CEST; 5min ago
   Duration: 1.661s
       Docs: https://weewx.com/docs
    Process: 17613 ExecStart=weewxd /etc/weewx/weewx.conf (code=exited, status=4)
   Main PID: 17613 (code=exited, status=4)
        CPU: 1.651s

Jul 09 17:38:35 byows-jim weewxd[17613]: CRITICAL weewx.engine:     ****    File "/usr/local/lib/python3.11/dist-packages/gpiozero/pins/pi.py", line 639, in _set_when_changed
Jul 09 17:38:35 byows-jim weewxd[17613]: CRITICAL weewx.engine:     ****      self._enable_event_detect()
Jul 09 17:38:35 byows-jim weewxd[17613]: CRITICAL weewx.engine:     ****    File "/usr/local/lib/python3.11/dist-packages/gpiozero/pins/rpigpio.py", line 220, in _enable_event_detect
Jul 09 17:38:35 byows-jim weewxd[17613]: CRITICAL weewx.engine:     ****      GPIO.add_event_detect(
Jul 09 17:38:35 byows-jim weewxd[17613]: CRITICAL weewx.engine:     ****  RuntimeError: Failed to add edge detection
Jul 09 17:38:35 byows-jim weewxd[17613]: CRITICAL __main__: Unable to load driver: Failed to add edge detection
Jul 09 17:38:35 byows-jim weewxd[17613]: CRITICAL __main__:     ****  Exiting...
Jul 09 17:38:35 byows-jim systemd[1]: weewx.service: Main process exited, code=exited, status=4/NOPERMISSION
Jul 09 17:38:35 byows-jim systemd[1]: weewx.service: Failed with result 'exit-code'.
Jul 09 17:38:35 byows-jim systemd[1]: weewx.service: Consumed 1.651s CPU time.

---------------------------------------------------

DEBUG info:
Using configuration file /etc/weewx/weewx.conf

System info
  Platform:       Linux-6.6.31+rpt-rpi-v7-armv7l-with-glibc2.36
  Python Version: 3.11.2

Load Information
   1 minute load average:  0.14    
   5 minute load average:  0.37    
  15 minute load average:  0.35    

General Weewx info
  Weewx version 5.1.0 detected.

Currently installed extensions
Extension Name    Version   Description
byows_rpi         0.51      Bring Your Own Weather Station (BYOWS) Driver for the Raspberry Pi

Archive info
  Database name:        weewx.sdb
  Table name:           archive
  Version               4.0
  Unit system:          1 (US)
  First good timestamp: 2024-07-09 15:50:00 CEST (1720533000)
  Last good timestamp:  2024-07-09 17:25:00 CEST (1720538700)
  Number of records:    20
  weewx (weewx.conf) is set to use an archive interval of 300 seconds.
  The station hardware was not interrogated to determine the archive interval.

Supported SQL keys
  dateTime                  usUnits                   interval                 
  altimeter                 appTemp                   appTemp1                 
  barometer                 batteryStatus1            batteryStatus2           
  batteryStatus3            batteryStatus4            batteryStatus5           
  batteryStatus6            batteryStatus7            batteryStatus8           
  cloudbase                 co                        co2                      
  consBatteryVoltage        dewpoint                  dewpoint1                
  ET                        extraHumid1               extraHumid2              
  extraHumid3               extraHumid4               extraHumid5              
  extraHumid6               extraHumid7               extraHumid8              
  extraTemp1                extraTemp2                extraTemp3               
  extraTemp4                extraTemp5                extraTemp6               
  extraTemp7                extraTemp8                forecast                 
  hail                      hailBatteryStatus         hailRate                 
[... 560 lines cut ...]
    
    # [OPTIONAL]
    # Pin to which rain bucket is connected, the DEFAULT is pin 6.
    rain_bucket_pin = 6
    
    # [OPTIONAL]
    # Port and address for sensor bme280, the DEFAULT are port=1 address=0x77
    bme280_port = 1
    bme280_address = 0x76
    
    # [OPTIONAL]
    # Channel to which wind vane is connected to on MCP3008, The DEFAULT is channel 0
    mcp3008_channel = 0
    
    # [OPTIONAL]
    # Anemometer adjustment value, the DEFAULT is 1.18
    anemometer_adjustment = 1.18
    
    # [OPTIONAL]
    # Bucket Size in mm, the DEFAULT is 0.2794 mm.
    bucket_size = 0.2794
    
    # [OPTIONAL]
    # Anemometer radious in cm, the DEFAULT is 9.0 cm.    
    anemometer_radius_cm = 9.0

tkeffer, 2024-07-10:
The issues list is for bug reporting and tracking. For general support question, please direct your questions to [weewx-user](https://groups.google.com/g/weewx-user).

However, this is most likely an issue with the byows driver. You should contact its author.

closed 2024-07-10 by tkeffer

groetg, 2024-07-10:
Thank you, I have found a workaround in editting /etc/systemd/system/weewx.service

And changed:

User and group into root:

# systemd service configuration file for WeeWX

[Unit]
Description=WeeWX weather system
Documentation=https://weewx.com/docs
Requires=time-sync.target
After=time-sync.target
Wants=network-online.target
After=network-online.target

[Service]
ExecStart=/usr/bin/python3 /usr/share/weewx/weewxd.py /etc/weewx/weewx.conf
StandardOutput=null
StandardError=journal+console
User=groetg
Group=groetg

[Install]
WantedBy=multi-user.target