weewx/weewx #971: incorrect instructions in quickstart and/or code for macos setup as a daemon

vinceskahan, 2024-08-17:
After installing 5.1.latest via pip on macos, I see....

You can now set up a daemon by running an appropriate script
in the directory /Users/vince/weewx-data/scripts
For example: 'sudo /Users/vince/weewx-data/scripts/setup-daemon.mac'

The quickstart says:
sudo sh ~/weewx-data/scripts/setup-daemon.sh

But the actual command for macos appears to be 'sudo /Users/vince/weewx-data/scripts/setup-daemon.mac' but the quickstart command for running via launchctl doesn't work either with a variety of errors that I have no idea how to get past.

Short description is the procedure for getting setup-daemon.sh through done and running as a service needs another looking at from somebody who knows how the (blasted) mac does things.

vinceskahan, 2024-09-01:
Thanks.   The generated plist file in weewx 5.1.0 was wrong for me, but this file below worked ok FWIW.

I found my setup also got a bit confused as well with which version(s) of python I had installed via homebrew so I needed to hand-edit that in.

I did not try a clean install to see if your commit worked end-to-end and hopefully generated the same plist file as below.   The instructions in the commentary worked fine.

```
<?xml version="1.0" encoding="UTF-8"?>
<!-- property list file for weewx daemon -->
<!-- put this file in /Library/LaunchDaemons -->
<!-- to start weewx: -->
<!--   sudo launchctl load -w /Library/LaunchDaemons/com.weewx.weewxd.plist -->
<!-- to stop: -->
<!--   sudo launchctl unload /Library/LaunchDaemons/com.weewx.weewxd.plist -->
<!DOCTYPE plist PUBLIC "-//Apple Computer//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Label</key>
    <string>com.weewx.weewxd</string>
    <key>Disabled</key>
    <false/>
    <key>RunAtLoad</key>
    <true/>
    <key>ProgramArguments</key>
    <array>
        <string>/opt/homebrew/opt/python@3.12/bin/python3.12</string>
        <string>/Users/vince/weewx-venv/lib/python3.12/site-packages/weewxd.py</string>
        <string>/Users/vince/weewx-data/weewx.conf</string>
    </array>
    <key>StandardErrorPath</key>
    <string>/var/log/weewx_err.log</string>
</dict>
</plist>
```

vinceskahan, 2024-09-03:
I might add the commit worked fine although the docs need to catch up

tkeffer, 2024-12-23:
When I do a pip install on my Mac, I see `~/weewx-data/scripts/setup-daemon.sh`, not `~/weewx-data/scripts/setup-daemon.mac`.

Where did you `setup-daemon.mac` file come from?

vinceskahan, 2024-12-23:
It comes from activating the initial station.  Give that a try.

Full reinstall here with transcript.   I can't follow the code well enough to tell where this one is coming from under the hood, but a 5.1.0 release today seems to still have the issue.

I get lost re: where commits are made and merged (or not) back and forth, so I can't speculate on what a hypothetical 5.1.1 might look like, but this one seems still valid.

````
vince@mini ~]$ python3 -m venv ~/weewx-venv

[vince@mini ~]$ source weewx-venv/bin/activate

(weewx-venv) [vince@mini ~]$ pip3 install weewx --no-cache
Collecting weewx
  Downloading weewx-5.1.0-py3-none-any.whl.metadata (5.5 kB)
Collecting CT3<4.0,>=3.1 (from weewx)
  Downloading CT3-3.4.0-cp313-cp313-macosx_11_0_arm64.whl.metadata (3.4 kB)
Collecting Pillow>=5.2 (from weewx)
  Downloading pillow-11.0.0-cp313-cp313-macosx_11_0_arm64.whl.metadata (9.1 kB)
Collecting PyMySQL<2.0,>=1.0 (from weewx)
  Downloading PyMySQL-1.1.1-py3-none-any.whl.metadata (4.4 kB)
Collecting configobj<6.0,>=5.0 (from weewx)
  Downloading configobj-5.0.9-py2.py3-none-any.whl.metadata (3.2 kB)
Collecting ephem<5.0,>=4.1 (from weewx)
  Downloading ephem-4.1.6-cp313-cp313-macosx_11_0_arm64.whl.metadata (6.0 kB)
Collecting pyserial<4.0,>=3.4 (from weewx)
  Downloading pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting pyusb<2.0.0,>=1.0.2 (from weewx)
  Downloading pyusb-1.2.1-py3-none-any.whl.metadata (2.2 kB)
Downloading weewx-5.1.0-py3-none-any.whl (1.5 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.5/1.5 MB 15.8 MB/s eta 0:00:00
Downloading configobj-5.0.9-py2.py3-none-any.whl (35 kB)
Downloading CT3-3.4.0-cp313-cp313-macosx_11_0_arm64.whl (873 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 873.2/873.2 kB 18.6 MB/s eta 0:00:00
Downloading ephem-4.1.6-cp313-cp313-macosx_11_0_arm64.whl (1.4 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.4/1.4 MB 20.2 MB/s eta 0:00:00
Downloading pillow-11.0.0-cp313-cp313-macosx_11_0_arm64.whl (3.0 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.0/3.0 MB 16.6 MB/s eta 0:00:00
Downloading PyMySQL-1.1.1-py3-none-any.whl (44 kB)
Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
Downloading pyusb-1.2.1-py3-none-any.whl (58 kB)
Installing collected packages: pyserial, ephem, pyusb, PyMySQL, Pillow, CT3, configobj, weewx
Successfully installed CT3-3.4.0 Pillow-11.0.0 PyMySQL-1.1.1 configobj-5.0.9 ephem-4.1.6 pyserial-3.5 pyusb-1.2.1 weewx-5.1.0

[notice] A new release of pip is available: 24.2 -> 24.3.1
[notice] To update, run: pip install --upgrade pip

(weewx-venv) [vince@mini ~]$ weectl station create --no-prompt
Creating configuration file /Users/vince/weewx-data/weewx.conf
Processing configuration file /Users/vince/weewx-data/weewx.conf
Creating directory /Users/vince/weewx-data/skins
Copying new skin Ftp into /Users/vince/weewx-data/skins/Ftp
Copying new skin Smartphone into /Users/vince/weewx-data/skins/Smartphone
Copying new skin Mobile into /Users/vince/weewx-data/skins/Mobile
Copying new skin Seasons into /Users/vince/weewx-data/skins/Seasons
Copying new skin Standard into /Users/vince/weewx-data/skins/Standard
Copying new skin Rsync into /Users/vince/weewx-data/skins/Rsync
Copying examples into /Users/vince/weewx-data/examples
Creating a new 'user' directory at /Users/vince/weewx-data/bin/user
Copying utility files into /Users/vince/weewx-data/util
Copying script files into /Users/vince/weewx-data/scripts
Saving configuration file /Users/vince/weewx-data/weewx.conf

You can now set up a daemon by running an appropriate script
in the directory /Users/vince/weewx-data/scripts
For example: 'sudo /Users/vince/weewx-data/scripts/setup-daemon.mac'

```

tkeffer, 2024-12-23:
Here's what I get:

```
~ python3 -m venv ~/weewx-venv
~ source weewx-venv/bin/activate
(weewx-venv) ~ pip3 install weewx --no-cache
Collecting weewx
  Downloading weewx-5.1.0-py3-none-any.whl (1.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.5/1.5 MB 1.8 MB/s eta 0:00:00
Collecting CT3<4.0,>=3.1
  Downloading CT3-3.4.0-cp38-cp38-macosx_11_0_arm64.whl (668 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 668.5/668.5 kB 1.8 MB/s eta 0:00:00
Collecting configobj<6.0,>=5.0
  Downloading configobj-5.0.9-py2.py3-none-any.whl (35 kB)
Collecting pyusb<2.0.0,>=1.0.2
  Downloading pyusb-1.2.1-py3-none-any.whl (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.4/58.4 kB 2.1 MB/s eta 0:00:00
Collecting Pillow>=5.2
  Downloading pillow-10.4.0-cp38-cp38-macosx_11_0_arm64.whl (3.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.4/3.4 MB 1.8 MB/s eta 0:00:00
Collecting pyserial<4.0,>=3.4
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 2.1 MB/s eta 0:00:00
Collecting PyMySQL<2.0,>=1.0
  Downloading PyMySQL-1.1.1-py3-none-any.whl (44 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.0/45.0 kB 1.5 MB/s eta 0:00:00
Collecting ephem<5.0,>=4.1
  Downloading ephem-4.1.6-cp38-cp38-macosx_11_0_arm64.whl (1.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.4/1.4 MB 1.8 MB/s eta 0:00:00
Installing collected packages: pyserial, ephem, pyusb, PyMySQL, Pillow, CT3, configobj, weewx
Successfully installed CT3-3.4.0 Pillow-10.4.0 PyMySQL-1.1.1 configobj-5.0.9 ephem-4.1.6 pyserial-3.5 pyusb-1.2.1 weewx-5.1.0

[notice] A new release of pip is available: 23.0.1 -> 24.3.1
[notice] To update, run: pip install --upgrade pip
(weewx-venv) ~ weectl station create --no-prompt
Creating configuration file /Users/tkeffer/weewx-data/weewx.conf
Processing configuration file /Users/tkeffer/weewx-data/weewx.conf
Creating directory /Users/tkeffer/weewx-data/skins
Copying new skin Seasons into /Users/tkeffer/weewx-data/skins/Seasons
Copying new skin Standard into /Users/tkeffer/weewx-data/skins/Standard
Copying new skin Smartphone into /Users/tkeffer/weewx-data/skins/Smartphone
Copying new skin Mobile into /Users/tkeffer/weewx-data/skins/Mobile
Copying new skin Rsync into /Users/tkeffer/weewx-data/skins/Rsync
Copying new skin Ftp into /Users/tkeffer/weewx-data/skins/Ftp
Copying examples into /Users/tkeffer/weewx-data/examples
Creating a new 'user' directory at /Users/tkeffer/weewx-data/bin/user
Copying utility files into /Users/tkeffer/weewx-data/util
Copying script files into /Users/tkeffer/weewx-data/scripts
Saving configuration file /Users/tkeffer/weewx-data/weewx.conf
```

That's it. No epilogue of, "You can now set up a daemon...". There used to be such a phrase in V5.0, but it was eliminated in V5.1. See commit https://github.com/weewx/weewx/commit/d63b8002b232780749c39e08c914916658f73056 Even then, it said `setup-daemon.sh`, not `setup-daemon.mac`. 

I'm at a loss..

vinceskahan, 2024-12-24:
> d63b800

Setting up a test user ran fine, so it was my $HOME or my $PATH.  It was
the latter.

Found it.  Actually them.   I somehow have 5.0.0rc1 buried in /opt/homebrew
so it must be coming from there.

What's interesting is in the venv $PATH has the venv first, but weectl from
the 5.1.0 venv has to be picking up the weectllib from /opt/homebrew.  Is
that a bug ?   Any way to instrument my system to see which weectllib it's
using, just to verify it ?

vinceskahan, 2024-12-24:
update - yup the venv is putting the venv site-packages 'last' rather than
'first' in sys.path, so weectl is getting its libs from an old 5.0.0rc1 in
/opt/homebrew.  Is that a bug ?

````
(weewx-venv) ***@***.*** ~]$ which python3
/Users/vince/weewx-venv/bin/python3
(weewx-venv) ***@***.*** ~]$ python3 --version
Python 3.13.0
(weewx-venv) ***@***.*** ~]$ python3
Python 3.13.0 (main, Oct  7 2024, 05:02:14) [Clang 15.0.0
(clang-1500.3.9.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path
['', ***@***.***/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python313.zip',
***@***.***/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13',
***@***.***/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/lib-dynload',
'/Users/vince/weewx-venv/lib/python3.13/site-packages']
>>>
````

tkeffer, 2024-12-24:
How did V5.0.0rc1 ever get in `/opt/homebrew` in the first place?

`weectl` does do dynamic loading, so it will be searching `sys.path`, but I'm not seeing how it would find something in `/opt/homebrew`.

vinceskahan, 2024-12-24:
Guessing I must have sudo installed via pep and forced it.  Uncertain.  It was a year ago.   What's even odder is that while 3.13 site-packages are in sys.path, the 3.11 (!!!) site-packages with the weectllib is 'not'.  Crazy.

Want to just punt on this one and call it a python oddity, or would you like to dig further?

tkeffer, 2024-12-24:
Your's is the only instance that I am aware of and you are not, ahem, a typical user. 

I'm closing the issue.

closed 2024-12-24 by tkeffer

vinceskahan, 2024-12-24:
Ouch.  I think :-)

No problem.  FWIW when I deleted the beta from the other python version the issue went away.