weewx/weewx #978: Vantage Pro 2 Driver - Ethernet - Please add a possibility to disconnect from the logger at interval

Hyrules, 2024-09-20:
This is a enhancement request : 

I'm using the latest version of Weewx 5 and I'm trying to juggle around Weewx and using the WifiLogger2 to get the data uploaded to wl.com. Since Weewx does not support pushing data weatherlink.com It would help if there could be a disconnection time from the logger to allow the logger to do it's job.

The way WifiLogger2 works is it pushes data to services when it's not busy feeding the data to the pc software connected to it. In this case Weewx is chocking the logger when polling every 2.5 seconds. The logger simply cannot process the other exports.

What I would like to see is Weewx implement a new property/function in the driver, something like a polling break of N seconds (configurable / default 15 secs) that happens every minutes so that it actually close the connection / socket to the logger so the logger can exports it's data. The polling could resume normally after. This is the way CumulusMX handle the connection to the logger. It's really the same as a Vantage Pro 2 USB or WeatherlinkIP logger but wifi and the ethernet polling normally works fine. At the moment the only way I can accomplish this is to stop weewx for a minute or so and restart it. 

Thanks,

tkeffer, 2024-09-20:
You're thinking of something like `time.sleep(x)` before sending the `LOOP` command, where `x` is a few seconds?

Hyrules, 2024-09-20:
it would be something like this :

Let say I set my disconnect time to 15 seconds

Pseudo code would be something like :

Get the data loops for 1 minute minus 15 seconds so 45 seconds
Close the connection / socket
Wait 15 seconds
Reconnect the socket and repeat ( you might want to fetch the loop you missed during that 15 seconds period)

Thing is you have to close the socket to disconnect from the logger It cannot just be a waiting period.

tkeffer, 2024-09-20:
I understand what you're suggesting. I'm not keen on the way Cumulus does it, which seems to involve explicitly closing the connection, waiting a bit, then reopening it, all in the main loop.

I'll have to think about this a bit. I'd rather see an implementation where the connection closes implicitly when the driver goes out of scope. Then you reconnect. Pseudo-code

```
while true:
    with vantage-loop(100) as connect:
        process loop data
    sleep a bit
```

Hyrules, 2024-09-21:
If you find a better way I'm all for it.

The only requirement for the logger is that it closes the socket so the logger is freed for a certain period of time so it can do it's thing. The WifiLogger2 dev which I talk with suggest optimally 15 seconds. It might not be needed every minutes. You could also implement it to happen on archive. Say if the station settings is every 5 minutes you could disconnect for n amount of seconds. That would cause an issue if you have set a export in the logger to anything not a multiple of 5 though.

On the other hand you i'm not sure if it's actually possible but add a way to support weatherlink.com. If a 3rd party logger can do it I imaging that weewx could support it. It's the only cloud/online popular service not support at the moment. I know that the API v2 does not allow write. That said I have no clue on how the logger does it. Does it use the V1 api that will stop working someday of is it a special protocol or a websocket I would be curious to sniff the packet send by the logger to wl.com.

Let met know if you want to test a driver. I'll be happy to help.

Hyrules, 2024-09-21:
Basically meteobridge is the better alternative. You can get it as a vm or a device that does the same thing as WifiLogger but it compatible with even more services as well as weatherlink.com. Plugger my usb logger back in, connect it to my meteobridge vm and the connect all the service. Then I connected weewx as a davis station with ethernet and here we go. Everything works as expected. 

That said You can close the issue if you want but It would be nice seeing weeatherlink.com implemented as a supported service in the near future. I wouldn't need meteobridge if weewx could support it. From my limited investigation the device writes at wl6.weatherlink.com/bin.php. Seems to be a binary stream. I`ve captured some packet with my router and analyse them in wireshark. Might just be pushing the loop it gets from the station. I will have to investigate more. 

Thanks again for your time and keep up the good work with Weewx.

tkeffer, 2024-09-21:
Being able to upload directly to weatherlink.com would definitely be the preferred option. Not only does it fit the existing architecture, but it would also allow other hardware to publish (although, I'm not sure how Davis would feel about that!)

Any insights into the protocol would be welcome. How does the WifiLogger2 do it?

Hyrules, 2024-09-21:
For my understanding it writes a binary stream using a http request more precisely a PUT to wl6.weatherlink.com/bin.php. I tried looking into how meteobridge does it which should basically the same as the WifiLogger2 but it's not a script it's a linux executatble "compiled" file so there's no easy way to see it. I'm trying to find out what's being sent. I would assume it's the observations but it seems that it's not documented anywhere and least no easy to find. Davis does not seem to mind right now because I believe all those other tools are spoofing a Weatherlink IP so there is no way for them to know 

[image]

PUT /bin.php HTTP/1.1
Host: wl6.weatherlink.com
User-Agent: WLIP/115 HW/111
Content-Type: application/octet-stream
Content-Length: 116

[image]

Currently topying with wireshark to see if I cannot find more.

tkeffer, 2024-09-21:
Well, if someone could reverse engineer it, I'd gladly put it in WeeWX.

I've always liked Davis equipment because it's reliable, and because they've published the serial protocol required to access it. This is in contrast to other vendors who regard my data as their proprietary information.

Unfortunately, with the emphasis towards the Weatherlink Console, they seem to be following industry trends. It would be nice to find alternative ways of getting access to our data.

Hyrules, 2024-09-22:
I'm trying to decode the octet stream. I'm wondering if the application just shoot the loop or the archive directly from the station to the bin.php file. Logically it would be the simplest way to do it since it's already encoded into a binary stream. I don't know if you are interested for some wireshark packet capture but can tell everytime it upload data to wl.com it makes 3 POST for meteobridge PUT for WifiLogger2 to w6.weatherlink.com. I'm not an expert of the serial communication protocol of the vantage stations but it might be faster for you to see if there is a correlation between the loops and the packet. If there were a way for me to visually see a loop in a hex editor I could try to match the data. Let me know I can send you via email. Not to keen on posting raw data in case my keys and station ID are in there somewhere.

Hyrules, 2024-09-22:
Here some interesting stuff :) I suggest you open in a code editor with c# color coding. The 3 POST / PUT are the actual IP Loop, HiLow and Clear Hi Low Packet. I can tell by the bytes sent in the packets. 02 , 04 and 05 which are first bytes of the stream.

[VantageCS.txt](https://github.com/user-attachments/files/17091053/VantageCS.txt)

Hyrules, 2024-09-27:
Currently reverse engineering the protocol. It's not simple but it's doable. I tried reaching to the devs (for Davis) but they seem to ignore my requests. If anybody has interest in reverse engineering the protocol I would welcome any help.

Jterrettaz, 2024-10-01:
Il look like part of the block contains the loop packet as described in p.22 and 23 of the Vantage Serial Communication Reference Manual. ( https://support.davisinstruments.com/article/rbzgl0rh6k-vantage-pro-pro-2-and-vue-communications-reference-2-6-1-any-os ).
It starts at byte 00d8 with 14 as "Bar Trend " value and ends at byte 133 with the last byte of "Time of Sunset".

Several years ago, I was using a Davis Datalogger IP, and I observed that there was two different set of data sent to weatherlink.com : loop data every minute for the weatherlink bulletin , and archive data each hour.

Hyrules, 2024-10-01:
Thanks for the info this should help alot. I have indeed match the loop packet with the structure sent and my initial findings seem to corroborate that. I had already guessed the value from the bulletin and from the serial protocol strucure it does match.

Bug-Storm, 2024-12-27:
Hey guys , any news ?? :)

Hyrules, 2024-12-27:
Basically I have abandoned the idea of reverse engineering the protocol. It's rather time consuming and not easy plus mostly the fact that I don't have the will to do it at the moment. There are other alternatives that exists that allow this in combination with weewx. You can take a look at Meteobridge if you want. It's not free but you can get it for raspberry pi (which I use) or a vm (which I used before going for a pi). It's basically a logger that you can add and connect it to some networks. I have also connected weewx to it as it work like a weatherlink live device.

closed 2026-01-21 by github-actions[bot]