weewx/weewx #980: "Exception ignore in PyMapping_HasKeyString()" and "string indices must be integers" in weewx 5.1.0+Fedora 41

cmadamsgit, 2024-11-04:
Since upgrading to Fedora 41 and updating weewx to 5.1.0, I'm seeing this in my log every loop:

`Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Traceback (most recent call last):
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]:   File "<string>", line 1, in <module>
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: TypeError: string indices must be integers, not 'str'
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Traceback (most recent call last):
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]:   File "<string>", line 1, in <module>
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: TypeError: string indices must be integers, not 'str'
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Traceback (most recent call last):
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]:   File "<string>", line 1, in <module>
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: TypeError: string indices must be integers, not 'str'
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: Traceback (most recent call last):
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]:   File "<string>", line 1, in <module>
Nov 04 12:40:15 pi4.cmadams.net weewxd[25981]: TypeError: string indices must be integers, not 'str'
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]: Traceback (most recent call last):
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]:   File "<string>", line 1, in <module>
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]: TypeError: string indices must be integers, not 'str'
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]: Traceback (most recent call last):
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]:   File "<string>", line 1, in <module>
Nov 04 12:40:16 pi4.cmadams.net weewxd[25981]: TypeError: string indices must be integers, not 'str'`

tkeffer, 2024-11-04:
`PyMapping_HasKeyString()` is not used in WeeWX, so it must be getting called in some underlying library or extension.

- Are you using any extensions?
- What version of Python?
- Some more of the log leading up to the exception would be useful. Incidentally, it is really tough to read the segment that you did most because of the lack of line breaks.

cmadamsgit, 2024-11-05:
No extensions installed. It's python3-3.13.0-1.fc41.aarch64. Sorry about the formatting, thought "code" would handle it. The only other log entries are the usual getting data and generating output.

```Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: INFO weewx.manager: Added record 2024-11-04 18:15:00 CST (1730765700) to database 'weewx.sdb'
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: INFO weewx.manager: Added record 2024-11-04 18:15:00 CST (1730765700) to daily summary in 'weewx.sdb'
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Traceback (most recent call last):
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]:   File "<string>", line 1, in <module>
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: TypeError: string indices must be integers, not 'str'
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Traceback (most recent call last):
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]:   File "<string>", line 1, in <module>
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: TypeError: string indices must be integers, not 'str'
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Traceback (most recent call last):
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]:   File "<string>", line 1, in <module>
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: TypeError: string indices must be integers, not 'str'
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: Traceback (most recent call last):
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]:   File "<string>", line 1, in <module>
Nov 04 18:15:16 pi4.cmadams.net weewxd[28238]: TypeError: string indices must be integers, not 'str'
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]: Traceback (most recent call last):
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]:   File "<string>", line 1, in <module>
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]: TypeError: string indices must be integers, not 'str'
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]: Exception ignored in PyMapping_HasKeyString(); consider using PyMapping_HasKeyStringWithError(), PyMapping_GetOptionalItemString() or PyMapping_GetItemString():
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]: Traceback (most recent call last):
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]:   File "<string>", line 1, in <module>
Nov 04 18:15:17 pi4.cmadams.net weewxd[28238]: TypeError: string indices must be integers, not 'str'
Nov 04 18:15:18 pi4.cmadams.net weewxd[28238]: INFO weewx.cheetahgenerator: Generated 8 files for report SeasonsReport in 1.77 seconds
Nov 04 18:15:18 pi4.cmadams.net weewxd[28238]: INFO weewx.imagegenerator: Generated 14 images for report SeasonsReport in 0.65 seconds
Nov 04 18:15:18 pi4.cmadams.net weewxd[28238]: INFO weewx.reportengine: Copied 0 files to /var/lib/weewx/html
Nov 04 18:15:22 pi4.cmadams.net weewxd[28238]: INFO weewx.reportengine: ftpgenerator: Ftp'd 20 files in 3.81 seconds```

gjr80, 2024-11-05:
And the driver being used is ....?

cmadamsgit, 2024-11-05:
Vantage. It's connected via USB to a Raspberri Pi 4B. I've been using it for a while, just upgraded from Fedora 39 (and weewx 4.10.2) last week (clean install on a new card).

tkeffer, 2024-11-05:
I was able to replicate this. Appears to be a problem with the Cheetah Template engine and the new Python v3.13. See https://github.com/CheetahTemplate3/cheetah3/issues/64

You have two solutions:
1. Use the latest version of Cheetah from their repository as outlined in their issues list.
2. Downgrade to an earlier version of Python.

-tk

closed 2024-11-05 by cmadamsgit