weewx/weewx #981: Pyephem is deprecated

roe-dl, 2024-11-17:
*PyEphem* is deprecated as described on the [*PyEphem* website](https://rhodesmill.org/pyephem/). The author replaced it by the [*Skyfield* module](https://rhodesmill.org/skyfield/).

I looked into the source code of Pyephem's underlying C library and found out that it contains a hard coded table of base values that ends in the year of 2018. After that the values are extrapolated. So there is an accuracy issue there. In contrary the *Skyfield* module downloads that table from the NASA website to always get an actual version of it.

There is also another Python astronomy project called *Astropy*. A first glance I did not see which one is better.

tkeffer, 2024-11-17:
This is something I looked into a few years ago, and reached a similar conclusion. We will need to move to something else.

At the time, I didn't pursue Skyfield because it depends on NumPy, which would add a lot of overhead to WeeWX.

First I've heard of _Astropy_. Looks like it would also requires NumPy.

Because the extended almanac is an optional feature, maybe it's not so bad if it pulls in NumPy.

roe-dl, 2024-12-11:
* [Skyfield Elegant Astronomy for Python](https://rhodesmill.org/skyfield/)
* [The Astropy Project](https://www.astropy.org)

As you say, **Skyfield** depends on **NumPy**, but **AstroPy** is not better. It reads

> The first step to installing astropy is to ensure that you have a Python environment which is **isolated** from your system Python installation. This is important because astropy has many dependencies, and you do not want to accidentally break your system by installing incompatible versions of these dependencies.

According to the documentation, **AstroPy** depends on **NumPy**, too, and additionally on [PyERFA](http://pyerfa.readthedocs.org/), [PyYAML](https://pyyaml.org/), and [packaging](https://packaging.pypa.io/). And there are a lot more optional dependencies.

roe-dl, 2024-12-31:
I guess I will try to do some coding. I would prefer Skyfield, as it has one dependency only, compared to AstroPy which has many.

I would check for the presence of Skyfield and NumPy by trying to import them. If they are there Skyfield is used. If not, the program falls back to the current behavior which is to check for PyEphem etc.

Unlike PyEphem, Skyfield downloads actual timescale and ephemerides data from IERS and USNO. So if they are updated, Skyfield can use them. On the other hand, Skyfield needs a directory to store files. I would use `SQLITE_ROOT` for that, as it points to a directory of continuously updated data.

The download takes time, as the files are very large. So it cannot be done during initializiation of an instance of class `Almanac`. Instead, the download needs to take place during WeeWX startup. So I would define a separate class for that.

tkeffer, 2024-12-31:
That sounds very intrusive. 

There is another approach: update PyEphem.

roe-dl, 2024-12-31:
> There is another approach: update PyEphem.

What do you mean by that?

PyEphem is based on an outdated C library. The library itself is based on the formulae of Jean Meeus, which are outdated, too. The PyEphem documentation reads:

> PyEphem will continue to receive critical bugfixes and be ported to each new version of Python. But be warned that it has some rough edges!

Brandon Rhodes does not want to update PyEphem any more.

tkeffer, 2024-12-31:
I mean that if Rhodes decides to completely abandon PyEphem, we can maintain it ourselves. But, it doesn't seem like he's going to do that anytime soon.

I don't think accuracy will be a problem. The documentation says it is limited to one arcsecond, which should be plenty good enough for our purposes.

The alternative is to use Skyfield, which will require an internet connection, a long download, a place to store the download, additional dependencies, and new support issues. Residence size will probably increase as well. It just doesn't seem worth it.

But, I encourage you to give Skyfield a try. Maybe it's not as bad as it seems.

roe-dl, 2024-12-31:
> I mean that if Rhodes decides to completely abandon PyEphem, we can maintain it ourselves. 

Actually, the astronomical data, PyEphem is based, end in 2018. So all results we get for today are extrapolated. Updating the base data would involve changing the C library. 

> I don't think accuracy will be a problem. The documentation says it is limited to one arcsecond, which should be plenty good enough for our purposes.

Today PyEphem reports a sunrise time of 08:10:57 for my location. Skyfield calculates 08:11:41. USNO, which I guess could be a reference, says 08:12. So Skyfield agrees with USNO, but PyEphem does not for that example. The difference is not always that much. The transit is 12:11:01 with PyEphem and 12:11:02 with Skyfield. The sunset is 16:11:16 with PyEphem and 16:10:33 with Skyfield. While the difference is 3/4 of a minute, both agree with the USNO value of 16:11.

> The alternative is to use Skyfield, which will require an internet connection, a long download, a place to store the download, additional dependencies, and new support issues. 

The internet connection is required for the upload, too. The long download is at most once a week, and it is no problem to do it less often. But I agree with you regarding the dependencies and the support issues.

I guess I will look a little bit further into the issue and report.

tkeffer, 2024-12-31:
Many users do not do any uploads. Their reports run strictly locally.

We do not want to require an internet connection. If the ephemeral data does not change too much, we could include it in the install package. It would get freshened with a WeeWX upgrade.

matthewwall, 2025-01-01:
we use astropy and numpy all the time.  they are rather large, and the changes in numpy releases often break things, typically in the other packages that depend on numpy.  if numpy is pulled in *only* for astropy/skyfield for use in weewx, it won't be a big deal - venv installations will be fine. it will probably be ok for deb/rpm-packaged python too.  it could be problematic if you use pip to install weewx without a venv on a system that uses other software that depends on numpy, but that is probably a corner case for a weewx installation.

roe-dl, 2025-01-02:
I thought about how to address all those things about support and compatibility. I guess it could be best to make the almanac a little bit extensible and put the Skyfield wrapper into a WeeWX extension. That would include the following advantages:

* As long as nobody complains about the almanac, nothing has to be done. But if someone comes and says he cannot use PyEphem (because their Linux distribution does not contain PyEphem any more or Brandon Rohdes does not adapt it to a new Python version or there are accuracy issues or something), then we can simply say to the user: "install that almanac extension and see if it solves your problem."
* It is not necessary to import NumPy to core WeeWX.
* The initialization (including that download) can simply be done within the extension by the well-known way of defining a WeeWX service. No new functionality is required to core WeeWX.
* To add extensibility to the almanac, some little lines of codes are required only. No big thing, that could be difficult to debug.
* In case of support issues, the almanac extension can be simply de-activated by commenting out the service, that initializes it.

> Many users do not do any uploads. Their reports run strictly locally.

I would add the actual versions of the files to the installation package. So the user does not need to download them, but they can update them if they see a reason to do so.

I still have to look for side effects, but roughly the changes to `almanac.py` could be that small:

at line 26:
```Python
almanac_extensions = []
```

in function `__getattr__` at line 321:
```Python
        for extension in almanac_extensions:
            try:
                return extension(self, attr)
            except weewx.UnknownType
                pass
```

The extension developer has to provide a function `get_almanac_data(almanac_obj, attr)`, which returns a `ValueTuple` or an instance of a binder class, if the attribute is supported by the extension, or raises an `weewx.UnknownType` exception otherwise. This is like XType extensions do.

I would provide the Skyfield wrapper as an WeeWX extension then. The users can then install it or not, just as they like. And we keep core WeeWX small, as you often require.

What about this approach?

tkeffer, 2025-01-02:
I think it's a brilliant approach, and very much in the WeeWX spirit!

I'm eager to see what you come up with.

chaunceygardiner, 2025-01-04:
For what it’s worth, I just updated my weewx-celestial extension to use Skyfield.  It was straightforward.There is no issue with an internet connection.  The data files DO NOT change and one can use load_file (rather than load—which needs an internet connection).I used the de421.bsp file which goes to 2050 and contains everything needed:-rw-r--r--  1 root root 16788480 Jan  3 20:21 de421.bspI install it in USER_DIR and the extension loads it from there.  If there is ever a new file (it will have a new name), it can simply be packaged with a new version.IMHO, it would be nice if we updated WeeWX to use it.  I was motivated when roe-dl pointed out how far off PyEphem is at this point.John

roe-dl, 2025-01-04:
@tkeffer See pull request #988

gdt, 2025-02-08:
From a packaging viewpoint, needing numpy in core weewx does not seem reasonable.   numpy is huge, and it drags in a fortran compiler.   Having an extension that works if skyfield/numpy are present and fails if not seems ok.    It also is not reasonable of programs to auto-update data files.   Having a script that the user can invoke to update is ok.   I strongly agree with Tom's view that weewx should work 100% ok without using the Internet.   I read the ticket and it sounds like things are heading in a sensible direction.

(FWIW, personally I decided to just remove almanac info from skin output.)

roe-dl, 2025-02-08:
Please look at the PR. There are no plans to include NumPy in core WeeWX. Instead, we make the almanac extensible. Generally you need neither PyEphem nor Skyfield. You *can* use them, but you are not required. 

Time is not steady. Astronomers measure the earth rotation and record their measurements. If you want correct almanac values (sunrise, sunset etc.) you have to adjust the base data from time to time. This is what was meant by Internet access. You do *NOT* need Internet access all the time for that.

roe-dl, 2025-02-11:
You can now see how the [Skyfield almanac extension](https://github.com/roe-dl/weewx-skyfield-almanac) is about to look like. There is already some documentation. 

I am not sure whether it would be allowed to provide the actual ephemeris and timescale files from JPL and IERS together with the extension in order to make it unnecessary to download them separately. But it is possible to copy those files there and configure the extension to not download anything.

roe-dl, 2025-02-13:
Extensible almanac and Skyfield support will appear in WeeWX 5.2.