weewx/weewx #982: Reference / Units Page

David-Enst, 2024-11-21:
The units table is messed up on this page (in all browsers):

https://weewx.com/docs/5.1/reference/units/

FYI,
Dave E

tkeffer, 2024-11-21:
Thanks for your report.

What do you mean by "messed up?" It looks fine to me.

David-Enst, 2024-11-22:
Here is what shows up for me, all browsers; the table is truncated on the right:

 

 

Cheers,

David Enstrom

David-Enst, 2024-11-22:
> Your post did not include a graphic.
>
> Did you try scrolling the table to the right?

Curious … image attached this time.

 

No, it cannot scroll!

 

 

Cheers,

David Enstrom

tkeffer, 2024-11-22:
Still not seeing an image. Try posting directly to the issues list. 

https://github.com/weewx/weewx/issues/982

tkeffer, 2024-11-22:
Yes, that's what it looks like until you scroll to the right. Look at the bottom of the table for the scroll bar. Is it there?

Or, if you're using a touchpad, two-finger scroll to the right.

David-Enst, 2024-11-22:
OH! Just saw the scroll bar ... rather annoying, but sort of works. But you would have to scroll back and forth to actually see the whole row. :-\

tkeffer, 2024-11-22:
Yes, you do. That's how MkDocs, the documentation software that we use, fits a wide table in a narrow screen. 

Frankly, I don't know how else it could be done.

Closing issue.

closed 2024-11-22 by tkeffer