weewx/weewx #983: Detail page of smartphone skin will not scroll after every even number of activations

aslak47, 2024-11-27:
I have noticed a small issue in the **smartphone skin**.
After first (and every odd number of  activations of any detail page containing the graphs) the detail page will scroll as expected, but after the second (and every even number of activations of any detail page) the detail page will not scroll.
I have tested from my phone and also with more browsers from my laptop after reducing the window size in order to force scrolling, and the behaviour is as described in all environments tested.
The skin is installed with my weather station no changes of templates, css- or js-files.
I have a WS1080 compatible weather station, and I run WeeWX version 5.1.0 (but I saw the same problem with earlier versions).

roe-dl, 2024-11-27:
Are you sure it is about odd and even numbers of activations? Not about activations near a 5-minute boundary?

tkeffer, 2024-11-27:
Hello. I'm not able to reproduce this on my laptop. It always scrolls (I'm assuming you mean vertically).

A few questions:
1. What browser?
2. What do you mean by an "activation"? A refresh?
3. How about a website that exhibits this behavior?

aslak47, 2024-11-27:
Some details:

1. I have tested with Chrome, Firefox and Microsoft Egde on my Android thwo phones, and Chrome and Microsoft Edge on thwo different PCs both running Windows 11 23H2.
2. Some examples:
Front page where I may expose details pressing "Lufttrykk", "Temperatur ute", "Fuktighet ute", "Regnintensitet" or "Vind"
[image]
I press "Lufttrykk", and I get details for that item that is vertically scollable.
[image]
I press "Back" (sorry for missing translation) and I am brought back to the front page
[image]
I then make a new choice of details to look at
[image]
and this time the scrollbar has disappeared.
I go back til front page, and make same choise (or another) and then the vertical scrollbar has reappeared on my third activation of details.
[image]
This pattern repeats so that the vertical scrollbar is present on every odd number of activation of details, and not present on every even number of activation of any detail page.
3. My website is: https://pavwx.ddns.net/weewx/smartphone/humidity.html
The weather station is a WS1080 based variant, and I use version 5.1.0 og WeeWX

tkeffer, 2024-11-27:
OK, now I understand and, yes, I can replicate it.

I have no idea why it is doing this. I suspect it's a JQuery Mobile issue, but that library was deprecated years ago.

The whole Smartphone skin should be modernized.

This is a minor problem, so I don't think it's worth chasing down. Sorry, but I don't think I c an help you.

-tk

aslak47, 2024-11-28:
I agree, this is really a minor problem. But I am pleased that you was able to repicate it.
If you choose to close this issue, its ok with me.

closed 2024-11-28 by tkeffer