weewx/weewx #984: Feature Request: Add config option to hard-set station model

staze, 2024-11-30:
For those of us with Vantage Vue's, the Vantage driver just identifies as a "Vantage Pro 2". And while we can hard set it in the skin, that then breaks easy upgrades of the skin. 

I have a Vantage Vue and a Weather Envoy with a WeatherLink IP Ethernet Data Logger. 

Would love to just see a way to hardcode the Station model in weewx.conf. Heck, maybe that's already possible and I just don't know how. If it's not, however, would be great to see this ability added. 

Thanks!

tkeffer, 2024-12-01:
There is already code to detect a Vantage Vue. See [Line 1413](https://github.com/weewx/weewx/blob/master/src/weewx/drivers/vantage.py#L1413) of the driver. Is it not working for you?

staze, 2024-12-01:
Hey @tkeffer , 

It doesn't. I assume that hardware code is returned by the console rather than the station so the fact I'm using a Weather Envoy rather than the Vue console is why it doesn't work?

Happy to help figure out if we can get auto-detection to work some other way (or just show some output from whatever). 

Here's output from weectl device --info

`weectl device --info
Using configuration file /etc/weewx/weewx.conf
Using driver weewx.drivers.vantage.
Using Vantage driver version 3.6.2 (weewx.drivers.vantage)
Querying...
Davis Vantage EEPROM settings:
    
    CONSOLE TYPE:                   Vantage Pro2
    
    CONSOLE FIRMWARE:
      Date:                         Nov 12 2019
      Version:                      3.88
    
    CONSOLE SETTINGS:
      Archive interval:             300 (seconds)
      Altitude:                     463 (foot)
      Wind cup type:                small
      Rain bucket type:             0.01 inches
      Rain year start:              10
      Onboard time:                 2024-11-30 16:24:26
      
    CONSOLE DISPLAY UNITS:
      Barometer:                    inHg
      Temperature:                  degree_F
      Rain:                         inch
      Wind:                         mile_per_hour
      
    CONSOLE STATION INFO:
      Latitude (onboard):           +44.1
      Longitude (onboard):          -123.0
      Use manual or auto DST?       AUTO
      DST setting:                  N/A
      Use GMT offset or zone code?  ZONE_CODE
      Time zone code:               4
      GMT offset:                   N/A
      Temperature logging:          LAST
        
    TRANSMITTERS: 
      Channel   Receive   Retransmit  Repeater    Type
         1      active        N         NONE      iss 
         2      inactive      N         NONE      (N/A) 
         3      inactive      N         NONE      (N/A) 
         4      inactive      N         NONE      (N/A) 
         5      inactive      N         NONE      (N/A) 
         6      inactive      N         NONE      (N/A) 
         7      inactive      N         NONE      (N/A) 
         8      inactive      N         NONE      (N/A) 

    RECEPTION STATS:
      Total packets received:       22829
      Total packets missed:         212
      Number of resynchronizations: 1
      Longest good stretch:         304
      Number of CRC errors:         5
      
    BAROMETER CALIBRATION DATA:
      Current barometer reading:    30.279 inHg
      Altitude:                     463 feet
      Dew point:                    33 F
      Virtual temperature:          35 F
      Humidity correction factor:   1.4
      Correction ratio:             1.018
      Correction constant:          +0.084 inHg
      Gain:                         0.000
      Offset:                       -19.000
      
    OFFSETS:
      Wind direction:               +0 deg
      Inside Temperature:           -6.0 F
      Inside Humidity:              +0 %
      Outside Temperature:          +0.0 F
      Outside Humidity:             +0 %
`

tkeffer, 2024-12-01:
Yes. The code is returned by the console. 

You didn't say, but I assume the `weectl` excerpt is from your Envoy? The Envoy returns the code for a VantagePro 2. Unfortunately, there is no way to tell the difference between the two.

You can always "hard code" anything you want using the [`[Extras]` section](https://www.weewx.com/docs/5.1/reference/skin-options/extras/) in either `skin.conf` or `weewx.conf`. That's what it's there for.

I'm reluctant to add an option specifically for the Vantage driver because that would be two different ways to do the same thing.

staze, 2024-12-01:
Yeah, I figured. The Envoy seems to just assume Vantage Pro 2 (I haven't found any way to set that to be different). Yes, that output is from the Envoy. Sadly don't seem to be a lot of Envoy users of Weewx to see if there's options. =/ 

How would I hardcode that in weewx.conf? I don't see an Extras section in weewx.conf. =/

tkeffer, 2024-12-01:
I own an Envoy. If it were possible to differentiate, I would have long ago done it!

To use `[Extras]` in `weewx.conf`, see the documentation on [_processing order_](https://www.weewx.com/docs/5.1/custom/custom-reports/#processing-order). It will look something like this (NOT TESTED):

``` ini
[StdReport]
  [[SeasonsReport]]
    [[[Extras]]]
      station_type = "Envoy"
```

Then you can use it in a skin as `$Extras.station_type`.

closed 2024-12-01 by tkeffer

staze, 2024-12-01:
Ah, gotcha. So was hoping to not modify the skin (since it's otherwise default). So there's way to just hardset in weewx.conf and have it override auto-detection? 

Looks like it's getting set in skin (Seasons) as 

`about.inc:      <td class="data">$station.hardware</td>`

Thanks!