weewx/weewx #985: weectl device --info      goes into an error

Psycho85stgt, 2024-12-21:
Hi together,
its a fresh setup with an Pi4 + WS1080

but I receive data from the station (every 5 minutes)

I would just like to be able to call up the other functions such as timedrift, live etc. as well.
It worked once (about 2-3 weeks ago)

However, the SD card of the PI broke. I had written down the steps at the time and fortunately saved the weewx.conf separately.
Raspberry PiOS 19.11.2024 (64-bit)
so I only had to install the OS, update, replace weewx.conf
run sudo systemctl restart weewx and nothing more.
But this time the function **weectl device** does not work.

```
weectl device --set-interval=30
Using configuration file /etc/weewx/weewx.conf
Using driver weewx.drivers.fousb.
Using FineOffsetUSB driver version 1.3 (weewx.drivers.fousb)
Traceback (most recent call last):
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 1036, in openPort
    self.devh.claimInterface(self.usb_interface)
  File "/usr/lib/python3/dist-packages/usb/legacy.py", line 232, in claimInterface
    util.claim_interface(self.dev, interface)
  File "/usr/lib/python3/dist-packages/usb/util.py", line 207, in claim_interface
    device._ctx.managed_claim_interface(device, interface)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/usb/core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 829, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/weewx/weectl.py", line 75, in <module>
    main()
  File "/usr/share/weewx/weectl.py", line 42, in main
    weectllib.device_actions.device()
  File "/usr/share/weewx/weectllib/device_actions.py", line 91, in device
    configurator.configure(config_dict)
  File "/usr/share/weewx/weewx/drivers/__init__.py", line 71, in configure
    self.do_options(options, parser, config_dict, not options.noprompt)
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 381, in do_options
    self.station = FineOffsetUSB(**config_dict[DRIVER_NAME])
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 967, in __init__
    self.openPort()
  File "/usr/share/weewx/weewx/drivers/fousb.py", line 1040, in openPort
    raise weewx.WeeWxIOError(e)
weewx.WeeWxIOError: [Errno 16] Resource busy
```

Psycho85stgt, 2024-12-21:
realy confusing, now i setup the raspy new, and now i get
`raspberrypi weewxd[993]: ERROR weewx.manager: Unable to add record 2024-12-21 14:57:54 CET (1734789474) to database 'weewx.sdb': attempt to write a readonly database`

with the same code and the same steps

install raspy with pi imager (fixed setup with default userconfig)
sudo apt update
sudo apt upgrade
sudo reboot
sudo raspi-config  >activate vnc
sudo reboot

do steps: https://weewx.com/docs/5.1/quickstarts/debian/
reboot

database looked.
ok i know whats to do  sudo chown weewx:weewx /var/lib/weewx/weewx.sdb

but why? several installation processes with different errors....

tkeffer, 2024-12-21:
The issues list is for reporting bugs and tracking them. For general support questions, please use the [weewx-user](https://groups.google.com/g/weewx-user) mailing list.

Having said that, see the Wiki section on [_Reading/writing to a database_](https://github.com/weewx/weewx/wiki/Understanding-permissions#readingwriting-to-a-database).

closed 2024-12-21 by tkeffer