weewx/weewx #986: User device drivers (like wxt5x0) not visibile in station reconfiguration device list

OH2LAK, 2024-12-28:
When using a user driver for a weather station like the wxt5x0 driver which installs to /etc/weewx/bin/user/wxt5x0.py, the driver is not visible in the station reconfiguration device list.

A workaround for this is to link the user driver to the built-in driver directory (ln -s /etc/weewx/bin/user/wxt5x0.py /usr/share/weewx/weewx/drivers/wxt5x0.py) but I believe this is maybe a rights issue?

tkeffer, 2024-12-28:
The reconfiguration script is supposed to search `/etc/weewx/bin/user`. 

I just tried this on a VM using WeeWX V5.1 and it worked fine. 

Try it again, except this time set `debug=` in `weewx.conf`. Stop `weewxd` (if it's running), then do `weectl station reconfigure`. Post the resultant log.

tkeffer, 2025-02-23:
@OH2LAK 

Erik?

tkeffer, 2025-03-29:
Unresponsive.

closed 2025-03-29 by tkeffer