weewx/weewx #989: WOW only allows 5 minute data -> dirty fix

BramBaert, 2025-01-12:
I did a local change on my restx.py version because as I kept getting retries trying to push 1 minute data to WOW. Below my changes. I am not doing it via a pull request as I ain't happy with how I did it but I currently don't know how to do it properly

Added  `from types import SimpleNamespace`

AmbientThread.check_response()
    put inside a try -> except

```
        except Exception as e:
            log.debug("Ignoring exception: {} -> allowing WOWThread to simply respond with OK".format(e))
```
            
WOWThread
   * Added the method extract_datetime_from_url    
```
        def extract_datetime_from_url(self, url):
            
            # Decode the URL
            decoded_url = urllib.parse.unquote(url)
            
            # Regex pattern to extract the dateutc value
            match = re.search(r"dateutc=([\d\-]+)\+([\d%:]+)", decoded_url)
            if match:
                date_part = match.group(1)  # Extract the date part (e.g., 2025-01-12)
                time_part = match.group(2)  # Extract the time part (e.g., 17:36:00)

                # Replace %3A with ":" in the time part
                time_part = time_part.replace("%3A", ":")
                
                # Combine date and time
                datetime_str = f"{date_part} {time_part}"
                
                # Parse the datetime string into a datetime object
                parsed_datetime = datetime.datetime.strptime(datetime_str, "%Y-%m-%d %H:%M:%S")
                
                return parsed_datetime
            else:
                raise ValueError("No valid datetime found in the URL")
```            
  *  Added the following at the start of the post_request
    
```
        try:
            if hasattr(self, 'time_last_update'):
                newLogTime = self.extract_datetime_from_url(request.full_url)
                if((newLogTime < self.time_last_update +  datetime.timedelta(minutes=5)) and
                    (newLogTime != self.time_last_update)):
                    log.debug("WOW: skipping upload to stay under 5 min limit. {} < {}+5min".format(newLogTime,self.time_last_update))
                    response = SimpleNamespace(**{"code":200})
                    return response
                else:
                    self.time_last_update = self.extract_datetime_from_url(request.full_url)
            else:
                self.time_last_update = self.extract_datetime_from_url(request.full_url)
        except:
            pass
```

tkeffer, 2025-01-12:
Why not just use option [`post_interval`](https://www.weewx.com/docs/5.1/reference/weewx-options/stdrestful/#post_interval_3)?

BramBaert, 2025-01-15:
Thanks Tom,

That is indeed what I needed.