weewx/weewx #991: inconsistency between wview_extended.py and units.py on luminosity vs illuminance

toast14, 2025-01-29:
In src/schemas/wview_extended.py, a schema field is defined for "luminosity".  In src/weewx/units.py, the value "illuminance" is in "group_illuminance".  Ideally, these two should match so the value is properly enumerated with units.

tkeffer, 2025-01-30:
Thank you for bringing this up! There have been several email and issue discussions about this in the past, going back to issue #685 and beyond. Unfortunately, none of them have been driven to conclusion.

First, note that despite their similar names, luminosity and illuminance are not the same. 

__Luminosity__ is the amount of _perceived_ light emitted by a source, typically measured in _lumens_. It is distinct from radiative power (measured in watts) in that it is weighted by the sensitivity of the human eye. For example, visible light is weighted more than UV or infrared.

__Illuminance__ is the amount of perceived light that falls on a surface area, typically measured in `lux`, or 1 lumen / m&sup2;.

In WeeWX, we have a unit for `illuminance` (unit `lux`), but no unit for `luminosity`.

I see that you have submitted PR #992 as an attempt to fix this. Unfortunately, I think it would make things worse because a unit intended for illuminance, would be used for luminosity.

Three things need to be done to set this all right:

1. We need to add a column `illuminance` to the `wview_extended` schema.
2. We need to add a new unit group `group_luminosity` to the set of unit groups in `units.py`.
3. The default unit of measure for `group_luminosity` should be set to `lumen`. 

The downside of this approach is that several extensions have been using `luminosity`, when they really should have been using `illuminance` (which didn't exist). It's unclear to me whether these extensions have been assuming that type `luminosity` is measured in `lux`. If so, these changes won't break anything, but extension authors should really fix things to use the right unit.

If you want, you could submit a PR that does these 3 things. Or modify the one you did submit.

Thanks again!

toast14, 2025-01-31:
I have modified the code as above (and on the pull request).

gdt, 2025-02-01:
As I said in the PR, I suspect all uses of luminosity are likely erroneous.   Thus I think the best course is to add illuminance and lux, and not to change luminosity right now.   Then, to address each use of luminosity, decide if it is correct (my guess is 99% not correct) and fix it.

I suspect we are where we are because various weather station manuals use terms incorrectly, and they've been adopted.

There is another term, "luminance", which is photometric rather than radiometric.  It is about the perceived light relative to a solid angle.   It is used to measure the brightness of computer displays.   But, I also don't see how it would be correctly used in a weather station.   The sensible measures are irradiance in W/m^2 and illuminance in lux (lm/m^2).

tkeffer, 2025-02-07:
Fixed in commit https://github.com/weewx/weewx/commit/3ebb32f00688d9143a1a0adf554469ff06382645

gjr80, 2025-02-07:
Not wanting to flog a dead horse but it seems this commit only fixed part of the problem; `luminosity` remains in the default schema as an orphan (ie no assigned unit group or units). I would expect that anything we put in the schema should belong to a unit group. I realise the comments in PR #992 mention simply adding `illuminance` and then investigating and then dealing with the use of `luminosity`. I think we know this will likely see `luminosity` left as an orphan for some time to come.

We know `luminosity` is a measure of power with the unit of `lumen`, why not deal with the orphan issue completely and assign `luminosity` to a unit group? I don't see such a change negatively affecting any existing `luminosity` users; if they care about units and formatting they will be manually assigning `luminosity` to a unit group and if they haven't they are probably not interested in unit labelling and formatting anyway.

gdt, 2025-02-08:
Do we have a precedent for removing luminosity from the schema?  There are only a few uses of the term in the source tree, basically:
  - extended schema def
  - markdown comments about fineoffset
  - translations for Seasons, but apparently not actually in the skin.

We could reword the markdown, drop the translations, and adjust the schema to remove the column so new installs won't have it.  Then any out-of-tree drivers/services/extensions that reference it need fixing.   I tend to be a "rip the bandaid off" type, but obviously my view about that should for very little.

Doing a database column rename is a much bigger deal, and as I understand weewx doctrine that doesn't happen on upgrade.

tkeffer, 2025-02-08:
There is no reason to remove `luminosity` from the database. On its own, it is not incorrect, or wrong, or a bug. 

It may have been interpreted incorrectly, but that is a supposition. We should leave it alone.

gjr80, 2025-02-09:
So will it continue to be the only obs type in the schema that is not assigned to unit group?

tkeffer, 2025-02-09:
The problem is that we can't come up with a good definition of it. Is it the ["radiant power of a light emitting object"](https://en.wikipedia.org/wiki/Luminosity), typically measured in watts, as defined by Wikipedia? Or, is it the [perceived power](https://en.wikipedia.org/wiki/Luminous_flux), typically measured in lumens, as also defined by Wikipedia?

To make an informed choice, we would have to see how it is used in the user base, but I suspect that to the extent it's used as all, it's used as a measure of what is actually illuminance.

Has anyone looked through the universe of extensions to see how it is actually used?

gjr80, 2025-02-10:
Spent some time this morning looking through the drivers and services listed in the wiki. From my brief search (went to the repo and looked at the driver/service `.py` file in `bin/user`) the only drivers/services that populate `luminosity` are the [Ecowitt gateway driver](https://github.com/gjr80/weewx-gw1000), the [OWFS driver](https://github.com/matthewwall/weewx-owfs) and the [WH23xx driver](https://github.com/matthewwall/weewx-wh23xx). From my brief look all populate `luminosity` from an Ecowitt type 'light sensor' with what we would now call `illuminance` ie a value in Lux. I didn't dig to see if any would be adversely affected by a change in schema.

FWIW I would go with luminosity being a measure of radiant power.

tkeffer, 2025-02-10:
So, measured in Watts?

If it were possible, my first choice would be to remove it from the schema. It seems that no one is using it correctly.

gjr80, 2025-02-10:
> It seems that no one is using it correctly.

I guess that is the one thing that is clear!

gdt, 2025-02-10:
I lean to saying luminosity is Watts, the astronomy definition.  I don't see the word being used in photometry (by those with an underestanding of the field).

So, sounds like 3 drivers should get updated to store illuminance.   Would be awesome if they can store to illuminance if it exists, otherwise keep going with luminosity.

I agreed with Tom that it would be best to remove luminosity.    2nd best is to put it in a new unit group deprecated which is a clue.