weewx/weewx #993: building and installing wheel into the system doesn't really follow module norms

gdt, 2025-02-01:
[If you'd rather I bring this up on weewx-devel just tell me.]

(I'm getting to updating to 5 from 4.x; no good reason for the delay.   The new install system with separation of code and data is a vast step forward and I'm glad to see it.)

I created a py-weewx in pkgsrc-wip, which basically declares the dependencies, asks for poetry as a build-time tool, and builds a wheel in a staging destdir from which a package is tarred up.  This is how pretty much every python package is built.    The point of doing it this way is that the code and dependencies are managed by the OS packaging system.

I see a few things which surprise me:

  - LICENSE.txt and README.md are in `/usr/pkg/lib/python3.12/site-packages`.   I realize you want to install them somewhere, but they are at top-level,  rather than in a weewx place.    In pkgsrc they belong in `/usr/pkg/share/doc/weewx`, and I suspect on various GNU/Linux distributions there is a similar place they belong.  It looks like they are `include`d in `pyproject.toml`, so I don't think this is a pkgsrc issue, but just poetry doing what it was asked to do.
  - there is `weectl.py` and `weewxd.py` at top level in `site-packages`.   Scanning my installed site dir, I see only a tiny number of such packages, compared to many instances of a pair of module dir and info dir.
  - there are a lot of modules that seem like they should be under `weewx` rather than also at top-level.   Many of them start with `wee` so that seems ok.   But `schemas` seems likely to collide with other packages.
  - I see `weewx_data`, which is perhaps the skeleton used in configuration.  That's more or less namespaced, but some of it belongs in `share/examples/weewx`, like the webserver config files, which aren't really about python at all.

To resolve the parts that I think will cause problems, it seems we could
  - put LICENSE/README elsewhere (and it's easy for me to just do that in packaging)
  - move schemas into weewx, and import weewx.schemas everywhere it imports schemas now.
but I am quite possibly failing to understand something.

Then,  a lower value, harder step would be making more things submodules of weewx, and putting the bits that aren't python module code into share someplace.

(I realize that the recommendation is to make a venv, in which case if the venv is just for weewx then this doesn't really cause problems, even if things are installed in a way that doesn't quite meet wheel norms.)

gdt, 2025-02-01:
Also, it seems slightly older and really recent poetry differ in terms of whether they actually install these files or they do not.

tkeffer, 2025-02-07:
Thanks for your comments.

While the packages and utilities that start with `wee` are a reasonable claim to a namespace, you're right about `schema`. We should probably move it to under `weewx`. This will require patching the import so that configuration files that reference `schemas.something` try `weewx.schemas.something`. If that doesn't work, then it tries `schemas.something`.

`LICENSE.txt` and `README.md` are just stupid oversights. They can be tucked under `weewx`.

Didn't understand your 2nd post.

Care to hack up a PR?

gdt, 2025-02-07:
In packaging, I found that poetry-core 1.9.1 installed README/LICENSE and poetry-core 2.0.1 did not.   Thus, it seems that newer poetry has a defense against "random files in site-packages".

Sounds like you would like PRs to:
  - put LICENSE.txt and README.md in weewx/ (but still in site-packages) instead of at top level
  - install schema under weewx and change the imports to match

I don't follow having fallback code.   This is a python module and I would expect it to be one version or the other.   I don't see how the new code that imports from weewx.schema could be installed (validly, without someone doing something crazy) without the new schema code location.   But I see you said "configuration files", and that I get.  I'll have to read the code more.

Longer term, there's a larger issue, which is that from at least pkgsrc norms (and I'd expect this is true also e.g. in Debian modulo paths), the documentation, skeletons, etc. belong in $prefix/share/doc/weewx  and $prefix/share/examples/weewx.   But let's keep that out of this issue and resulting PRs.

gdt, 2025-02-07:
Regarding schemas, the weatherlink live driver does
```
from schemas import wview_extended
```
and I don't see how to deal with that other than expecting extensions that import schemas to wrap the import in try, use weewx.schemas first, and then fallback to schemas.
Well, we could have schemas be near empty and import weewx.schemas as a transition, but that feels like more trouble than it's worth.

If you don't want to deal with this now I can create a fresh issue just about schemas and close this one.

tkeffer, 2025-02-07:
Now it's coming back to me why we left `schema` alone. 

I suppose we could put a `try` block around the instantiation of drivers, services, report engines, and search list extensions. 

```
try:
    instantiate_something()
except Import Error:
    import weewx.schemas
    sys.modules['schemas'] = sys.modules['weewx.schemas']
    instantiate_something()
```

But, man, that's ugly. 

Why does the weatherlink live driver import the schema anyway? Why doesn't it use the provided API? Perhaps it should just fail hard.

gdt, 2025-02-08:
I asked about why the WLL driver is how it is and if it is reasonable to change it.
https://github.com/michael-slx/weewx-weatherlink-live/discussions/50

bellrichm, 2025-02-08:
> Now it's coming back to me why we left `schema` alone.
>
> I suppose we could put a `try` block around the instantiation of drivers, services, report engines, and search list extensions.
>
> ```
> try: [...]

Not my extension, but as an extension writer I'd be happy if you took the approach you took with V4 logging. Essentially telling developers that it is going to change and write a wiki article explaining how to deal with it. Something like the following.

In V6 (I picked V6 because it is breaking change....), schema definitions are going to move from the `schemas` namespace to the `weewx.schemas` namespace. If one is importing `schemas.something`, they should update the code to the following (note, I have not tested this).
```
try:
    import weewx.schemas.something
except ImportError:
    import schemas.something
```

tkeffer, 2025-02-25:
Fixed in commit https://github.com/weewx/weewx/commit/c18af7e8f181ed8bea2e18e3e81f683c0864073f

A shim has been included for old configuration files. If importing `schema.schema_name` does not work, then `weewx.schema.schema_name` is tried.

closed 2025-02-25 by tkeffer

tkeffer, 2025-02-25:
That's what it does. If you try to import `weewx.schemas.wview_extended` in a V5.2 installation, it will succeed. 

If you try to import `schemas.wview_extended` it will fail and try `weewx.schemas.wview_extended`.

Or, am I missing something?

gdt, 2025-02-26:
I get it now, from your comment and reading weeutil.get_object.  This is a helper that should be called with weewx.schemas.foo, and not particularly related to schemas -- it's just that's the best place to accomodate pre-move schema refs.  Thanks for explaining and the namespace purity improvement.