weewx/weewx #995: Moon almanac error

roe-dl, 2025-02-18:
The Seasons skin (and other skins, too) state moon rise at 00:20:58 CET on 2025-02-18 (today). But there is no moonrise on 2025-02-18 at 00:20:58 CET. There is no moonrise at all in timezone CET on 2025-02-18.

According to [mooncalc.org](https://mooncalc.org) (or [mondverlauf.de](https://mondverlauf.de) for german readers) the moon is visible:

Date | From (rise) | To (set)
-----|-----|------
2025-02-17 | previous day 21:58:17 CET | 08:41:52 CET
2025-02-18 | previous day 23:09:03 CET | 08:53:48 CET
2025-02-19 | 00:21:18 CET | 09:08:38 CET
2025-02-20 | 01:35:01 CET | 09:28:23 CET

Obviously, what is displayed as moonrise of 2025-02-18, is moonrise of the next day (2025-02-19).

[image]

The Seasons skin does not display from when to when the moon is visible, but it displays those events that fall into the given day, even if they belong to different periods of visibility. Because of that behavior `N/A` would be the correct value for 2025-02-18 in timezone CET.

tkeffer, 2025-02-22:
To implement "rising" time, the almanac code uses the pyephem function `next_rising()`. It effectively becomes:

```
observer = ephem.Observer()
observer.lat = ...something...
observer.lon = ...something...
observer.date = ... the time ...
moon = ephem.Moon()
moon.compute(my_location)

observer.next_rising(moon)
```

So, the code is not wrong: in your example, the "next rising" is the following day, 2025-02-19.

Perhaps rather than supply code to calculate moonrise on the given (or previous) day, we should simply change the label from `Rise` to `Next rising`? Alternatively supply the date as well as the time.

roe-dl, 2025-02-22:
> Perhaps rather than supply code to calculate moonrise on the given (or previous) day, we should simply change the label from `Rise` to `Next rising`? Alternatively supply the date as well as the time.

Yes. I guess, it is a question of the point of view. Because there was only a time, I thought the value was related to the actual day. But that was not the case in the example above. 

I thought, I should not only report a problem, but also suggest a possible solution. No problem if you do not use it.

tkeffer, 2025-03-04:
Another thought: we could add a `+1` to the time if the time is in the next day (similar to what airlines do when your arrival time is in the next day). This would require passing along a reference time to the `ValueHelper` --- simple enough to do.

roe-dl, 2025-03-04:
I guess you mean class `Formatter`, and there something like this?

```
            if reference_time:
                day_timespan = weeutil.weeutil.archiveDaySpan(reference_time)
                if t>day_timespan[1]:
                    val_str += ' +1'
                if t<=day_timespan[0]:
                    val_str += ' -1'
```

May be, we should additionally check for the absence of `%d` in the format string.

tkeffer, 2025-03-05:
Something like that.

roe-dl, 2025-03-05:
Do you want me to add it to the PR or to prepare a new one?

(Of course, that code snippet is not the whole thing. There are changes required to pass that reference time down to the place where it is used.)

roe-dl, 2025-03-13:
And just another thought: The lines of new and full moon are sorted in chronological order. The same would be possible for the lines of rising, transit, and setting.

closed 2025-10-08 by github-actions[bot]