weewx/weewx #996: Temperature & dew point jumps after restarting WeeWx

OH2LAK, 2025-02-18:
I have been wondering this issue for a while but have not come to any conclusion of what is the root cause for this.

When restarting WeeWx after shorter or longer period of running, the temperature & dew point values suddenly jump either up or down as seen in the picture. I restarted Weewx service at 1940hrs and 0300hrs, and first the temperature went down a few degrees, and then up.

[image]

The weather station is a Vaisala WXT510 and it sends temperature data once a second to WeeWx. I've understood that WeeWx then puts an average of the 60 seconds to the archive database which is also visualized. The same glitch is visible also in the WeeWx report, the picture above is from Grafana where data is also sent with a WeeWx plugin.

I will most likely modify the wxt5x0 driver to write the raw weather station data received to a file or a database to compare the numbers to further understand where the jump maybe originates. Could it be some long-period median counter resetting with WeeWx restart?

..
Erik

OH2LAK, 2025-02-25:
I'll check this and get back to you. Sorry I missed your reply at first!

..
Erik

OH2LAK, 2025-02-27:
Good day.

Last night the nightly restart of WeeWx did not give a huge deviation but something still.

[image]

Just before the cron job kicked in, the last temperature was parsed and mapped as +1.1C, humidity as 89.1% and wind direction 10min average as 236 degrees
```
helmi 27 03:00:00 DEBUG weewx.drivers.wxt5x0: parsed: {'wind_dir_avg': 236.0, 'wind_speed_avg': 2.6, 'temperature': 1.1, 'humidity': 89.1, 'pressure': 1015.5, 'rain': 104.87, 'heating_temperature': 7.9, 'heating_voltage':>
helmi 27 03:00:00 DEBUG weewx.drivers.wxt5x0: mapped: {'windDir': 236.0, 'windSpeed': 2.6, 'outTemp': 1.1, 'outHumidity': 89.1, 'pressure': 1015.5, 'rain_total': 104.87, 'heatingTemp': 7.9, 'heatingVoltage': 23.7, 'dateTi>
helmi 27 03:00:01 Stopping weewx.service - WeeWX...
```

After WeeWx got back on its feet, the first measurement parsed and mapped was temperature +1.7C, humidity 91.9% and wind direction average at 141 degrees.
```
helmi 27 03:00:01 DEBUG weewx.drivers.wxt5x0: parsed: {'wind_dir_avg': 141.0, 'wind_speed_avg': 1.7, 'temperature': 0.7, 'humidity': 91.9, 'pressure': 1015.6, 'rain': 106.95, 'heating_temperature': 8.0,...
helmi 27 03:00:01 INFO weewx.drivers.wxt5x0: skipping rain measurement: rain=106.95 last_rain=None
helmi 27 03:00:01 DEBUG weewx.drivers.wxt5x0: mapped: {'windDir': 141.0, 'windSpeed': 1.7, 'outTemp': 0.7, 'outHumidity': 91.9, 'pressure': 1015.6, 'rain_total': 106.95, 'heatingTemp': 8.0, 'heatingVoltage': 23.7,...
```

What in this time is significant change is the 10 minute wind direction average which changed from 236 degrees to 141 degrees within few seconds. This makes me think that restarting WeeWx resets/restarts the WXT when the driver initializes, and that seems to be the case:

wxt5x0.py driver has these lines for various reset functions of the WXT:

```
def reset(self):
        self.send_cmd(b'XZ')

def precip_counter_reset(self):
        self.send_cmd(b'XZRU')

def precip_intensity_reset(self):
        self.send_cmd(b'XZRI')

def measurement_reset(self):
        self.send_cmd(b'XZM')
```

I'm not exactly sure which of these are used in which phase but certainly this is the reason why the numbers jump as WXT basic function is based on averaging, but to use with WeeWx the averaging time needs to be put close to zero or disabled completely.

This is how the WXT can be configured, so as WeeWx writes the database once a minute maybe temperature reporting should be just once a minute, and same for the wind?

[image]

..
Erik

tkeffer, 2025-02-27:
If I understand you correctly, this is a problem in the wxt5x0 driver, and not in WeeWX. I would suggest that you take this up with the author.

Closing, but we can reopen if necessary.

closed 2025-02-27 by tkeffer