weewx/weewx #997: Solar Time

roe-dl, 2025-02-22:
@tkeffer I have a question.

The daily course of the weather is heavily influenced by the position of the sun in the sky, which is expressed by solar time. Clock time can be far away from that because of time zones and daylight saving switching. I therefore find it interesting to also display the solar time. 

Solar time can be calculated by the almanac, but actually it cannot be displayed in an easy way, because it is already local time and therefore cannot be converted by the `time.localtime()` statement in the formatter. To add support for displaying the solar time or other local time I would like to add some little 10 lines to the formatter in `units.py`. 

Would you consider accepting a PR for that?

tkeffer, 2025-02-23:
Sounds useful, but I'm not sure what you have in mind. Tags such as `$almanac.sun.rise.solar`?

roe-dl, 2025-02-24:
I thought about something like `$almanac.solar_datetime.format("%H:%M:%S")`. Your example sounds interesting, too.

roe-dl, 2025-03-04:
As this is continued in the PR #998, I guess, this issue can be closed now.

closed 2025-03-04 by roe-dl