weewx/weewx #1001: $system_locale_js - should be handled dfferently with locale C.utf8

LMSSonos, 2025-04-06:
I'm running weewx 5.1 in the docker-container <https://github.com/felddy/weewx-docker> on kubernetes. The locale output is as follows

```
# locale -a
C
C.utf8
POSIX
```

When I generate a report, in my case `belchertown` there is a the following variable in the code `var moment_locale = "$system_locale_js";` 

When I run the report `weectl report run Belchertown --config=/data/weewx.conf` it will generated this output:  `var moment_locale = "C";`

In the browser the script will fail, because `C` is not a supported locale for the JS code. In my case, I simply replaced the `$system_locale_js` with `en-US`. I tried to find the code for the generator and this variable to make a PR, so maybe somebody could direct me to the right file.

Thanks

tkeffer, 2025-04-06:
Thanks for your report, but this is a Belchertown issue. I would suggest filing it there.

https://github.com/poblabs/weewx-belchertown/issues

closed 2025-04-06 by tkeffer

LMSSonos, 2025-04-07:
@tkeffer thanks, found the correct spot in these templates and was already fixed in the `main` branch, sorry I thought logic happens in weewx.