weewx/weewx #1005: Ecowitt Gateway Driver AWOL

pashdown, 2025-06-19:
There are repeated references to the [Ecowitt Gateway Driver](https://github.com/gjr80/weewx-gw1000/).  Unfortunately, it appears that the owner has removed it as it ends in a 404.

tkeffer, 2025-06-19:
I feel your pain, but the driver is not actually part of WeeWX. For reasons known only to him, the author has chosen to disappear. We can only hope that someone else steps up and takes his place.

closed 2025-06-19 by tkeffer

pashdown, 2025-06-20:
Are there any forks that should be pointed to instead?  I ended up [using this one](https://github.com/Millardiang/weewx-gw1000).  No idea if it was forked from the original, but it works.

tkeffer, 2025-06-21:
I don't know --- I'm not familiar with the Ecowitt driver.

I'm hoping that someone with more knowledge than me steps up, sorts through the various versions, then comes up with something definitive.