weewx/weewx #1010: Critical error using weewx 5.1 vs mariadb 12

themoosman, 2025-08-14:
When running weewx 5.1 vs mariadb `12.0.2`, the following critical error is thrown, and the application will not start.
```
weewxd[1] INFO weewx.wxservices: StdWXCalculate will use data binding wx_binding
weewxd[1] CRITICAL weewxd: Caught unrecoverable exception:
weewxd[1] CRITICAL weewxd:     ****  (1193, "Unknown system variable 'storage_engine'")
weewxd[1] CRITICAL weewxd:     ****  Traceback (most recent call last):

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 53, in guarded_fn

weewxd[1] CRITICAL weewxd:     ****      return fn(*args, **kwargs)

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 132, in __init__

weewxd[1] CRITICAL weewxd:     ****      set_engine(self.connection, engine)

weewxd[1] CRITICAL weewxd:     ****      ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 316, in set_engine

weewxd[1] CRITICAL weewxd:     ****      connect.query("SET storage_engine=%s;" % engine)

weewxd[1] CRITICAL weewxd:     ****      ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 563, in query

weewxd[1] CRITICAL weewxd:     ****      self._affected_rows = self._read_query_result(unbuffered=unbuffered)

weewxd[1] CRITICAL weewxd:     ****                            ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 825, in _read_query_result

weewxd[1] CRITICAL weewxd:     ****      result.read()

weewxd[1] CRITICAL weewxd:     ****      ~~~~~~~~~~~^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 1199, in read

weewxd[1] CRITICAL weewxd:     ****      first_packet = self.connection._read_packet()

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 775, in _read_packet

weewxd[1] CRITICAL weewxd:     ****      packet.raise_for_error()

weewxd[1] CRITICAL weewxd:     ****      ~~~~~~~~~~~~~~~~~~~~~~^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/pymysql/protocol.py", line 219, in raise_for_error

weewxd[1] CRITICAL weewxd:     ****      err.raise_mysql_exception(self._data)

weewxd[1] CRITICAL weewxd:     ****      ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/pymysql/err.py", line 150, in raise_mysql_exception

weewxd[1] CRITICAL weewxd:     ****      raise errorclass(errno, errval)

weewxd[1] CRITICAL weewxd:     ****  pymysql.err.OperationalError: (1193, "Unknown system variable 'storage_engine'")

weewxd[1] CRITICAL weewxd:     ****  

weewxd[1] CRITICAL weewxd:     ****  During handling of the above exception, another exception occurred:

weewxd[1] CRITICAL weewxd:     ****  

weewxd[1] CRITICAL weewxd:     ****  Traceback (most recent call last):

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weewxd.py", line 121, in main

weewxd[1] CRITICAL weewxd:     ****      engine = weewx.engine.StdEngine(config_dict)

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weewx/engine.py", line 89, in __init__

weewxd[1] CRITICAL weewxd:     ****      self.loadServices(config_dict)

weewxd[1] CRITICAL weewxd:     ****      ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weewx/engine.py", line 157, in loadServices

weewxd[1] CRITICAL weewxd:     ****      obj = weeutil.weeutil.get_object(svc)(self, config_dict)

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weewx/wxservices.py", line 103, in __init__

weewxd[1] CRITICAL weewxd:     ****      self.db_manager = engine.db_binder.get_manager(data_binding=data_binding,

weewxd[1] CRITICAL weewxd:     ****                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****                                                     initialize=True)

weewxd[1] CRITICAL weewxd:     ****                                                     ^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weewx/manager.py", line 775, in get_manager

weewxd[1] CRITICAL weewxd:     ****      self.manager_cache[data_binding] = open_manager(manager_dict, initialize)

weewxd[1] CRITICAL weewxd:     ****                                         ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weewx/manager.py", line 924, in open_manager

weewxd[1] CRITICAL weewxd:     ****      return manager_cls.open_with_create(manager_dict['database_dict'],

weewxd[1] CRITICAL weewxd:     ****             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****                                          manager_dict['table_name'],

weewxd[1] CRITICAL weewxd:     ****                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****                                          manager_dict['schema'])

weewxd[1] CRITICAL weewxd:     ****                                          ^^^^^^^^^^^^^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weewx/manager.py", line 231, in open_with_create

weewxd[1] CRITICAL weewxd:     ****      connection = weedb.connect(database_dict)

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weedb/__init__.py", line 105, in connect

weewxd[1] CRITICAL weewxd:     ****      return driver_mod.connect(**db_dict)

weewxd[1] CRITICAL weewxd:     ****             ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 70, in connect

weewxd[1] CRITICAL weewxd:     ****      return Connection(host=host, port=int(port), user=user, password=password,

weewxd[1] CRITICAL weewxd:     ****                        database_name=database_name, engine=engine, autocommit=autocommit, **kwargs)

weewxd[1] CRITICAL weewxd:     ****    File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 62, in guarded_fn

weewxd[1] CRITICAL weewxd:     ****      raise klass(e)

weewxd[1] CRITICAL weewxd:     ****  weedb.DatabaseError: (1193, "Unknown system variable 'storage_engine'")

weewxd[1] CRITICAL weewxd:     ****  Exiting.
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 53, in guarded_fn
    return fn(*args, **kwargs)
  File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 132, in __init__
    set_engine(self.connection, engine)
    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/weedb/mysql.py", line 316, in set_engine
    connect.query("SET storage_engine=%s;" % engine)
    ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 563, in query
    self._affected_rows = self._read_query_result(unbuffered=unbuffered)
                          ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 825, in _read_query_result
    result.read()
    ~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 1199, in read
    first_packet = self.connection._read_packet()
  File "/usr/local/lib/python3.13/site-packages/pymysql/connections.py", line 775, in _read_packet
    packet.raise_for_error()
    ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/pymysql/protocol.py", line 219, in raise_for_error
    err.raise_mysql_exception(self._data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymysql/err.py", line 150, in raise_mysql_exception
    raise errorclass(errno, errval)
pymysql.err.OperationalError: (1193, "Unknown system variable 'storage_engine'")

```

Something must have broken in this code:  https://github.com/weewx/weewx/blob/master/src/weedb/mysql.py#L313-L316

themoosman, 2025-08-14:
For those looking for a temporary workaround, update the `mysql.py` file; the path will be in the error message.  

Change `set_engine` function to look like the following:

```
def set_engine(connect, engine):
    """Set the default MySQL storage engine."""
    try:
        server_version = connect._server_version
    except AttributeError:
        server_version = connect.server_version
    # Some servers return lists of ints, some lists of strings, some a string.
    # Try to normalize:
    if isinstance(server_version, (tuple, list)):
        server_version = '%s.%s' % server_version[:2]
    connect.query("SET default_storage_engine=%s;" % engine)
```

tkeffer, 2025-08-14:
I think your analysis is correct. For some reason, the `mysql.py` driver is not recognizing the server version.

Could you do me a favor and go back to the original version of `set_engine()` and instrument it to print out the variable `server_version`?

When done, it would look something like this:

```python
def set_engine(connect, engine):
    """Set the default MySQL storage engine."""
    try:
        server_version = connect._server_version
    except AttributeError:
        server_version = connect.server_version
    # Some servers return lists of ints, some lists of strings, some a string.
    # Try to normalize:
    if isinstance(server_version, (tuple, list)):
        server_version = '%s.%s' % server_version[:2]

    # Add the following line:
    print(f"server version: {server_version}")

    if server_version >= '5.5':
        connect.query("SET default_storage_engine=%s" % engine)
    else:
        connect.query("SET storage_engine=%s;" % engine)
```

Then run from the command line.

themoosman, 2025-08-14:
Mine prints out:

```
server version: 12.0.2-MariaDB-ubu2404
```

tkeffer, 2025-08-15:
Ah, the old natural sort problem. The string `'12.0.2'` sorts to before `'5.5'` because of the leading `1`. Change the line

        if server_version >= '5.5':

to

        if server_version >= '05.5':

and see what happens.

themoosman, 2025-08-15:
That works.

tkeffer, 2025-08-15:
Fixed in commit https://github.com/weewx/weewx/commit/17c71427ecad7ab480e93c04b001e5e1b997ac8e

Thanks for your help!

closed 2025-08-15 by tkeffer